wlcrs/huawei_solar #262: [Bug]: not working with 1.2.0

mariusnauj, 2023-01-19:
### Describe the issue

After update from 1.1.6 to 1.2.0 integration do not working. HA version 2023.1.5.
Useing wireless method of connection with inverter. Back to 1.1.6 all working fine.

### Bescribe your Huawei Solar Setup

Inverter Type: 
Inverter Firmware version:
SDongle present: yes / no
Power meter present: three phase / single phase / no
Battery: LG RESU xxkWh / LUNA2000 xxkWh / None
Battery Firmware version:

### How do you connect to the inverter?

Via the `SUN2000-<serial_no>` WiFi

### Upload your Diagnostics File

Drag & Drop your Diagnostics File here.

### Upload your relevant debug logs

```text
Logger: huawei_solar.huawei_solar
Source: /usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py:163
First occurred: 19:13:14 (1 occurrences)
Last logged: 19:13:14

Aborting client creation due to error.
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 157, in create
    await huawei_solar._initialize()
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 120, in _initialize
    self.time_zone = (await self.get(rn.TIME_ZONE)).value
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 254, in get
    return (await self.get_multiple([name], slave))[0]
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 296, in get_multiple
    response = await self._read_registers(
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 389, in _read_registers
    result = await _do_read()
  File "/usr/local/lib/python3.10/site-packages/backoff/_async.py", line 151, in retry
    ret = await target(*args, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 354, in _do_read
    raise ConnectionException(message)
huawei_solar.exceptions.ConnectionException: Modbus client is not connected to the inverter.
```

### Please confirm the following:

- [ ] The problem is still present in the latest release of this integration.
- [ ] I did not find an existing issue describing this problem.
- [ ] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [ ] I increased the logging configuration, restarted HA, and have copied all relevant log lines from the 'Full Logs' into the textarea.

gieljnssns, 2023-01-19:
When I did the update my huawei_solar integration also stoped working with a lot of errors
I have a dongle but I used the connection to the internal wifi of the inverter, I do not have optimisers or battery
```
Deze fout is ontstaan door een aangepaste integratie.

Logger: custom_components.huawei_solar
Source: helpers/update_coordinator.py:168
Integration: Huawei Solar
First occurred: 20:03:32 (5 occurrences)
Last logged: 20:15:19

Timeout fetching HV20C0092844_data_update_coordinator data
```

```
Deze fout is ontstaan door een aangepaste integratie.

Logger: backoff
Source: custom_components/huawei_solar/__init__.py:87
Integration: Huawei Solar
First occurred: 20:17:56 (1 occurrences)
Last logged: 20:17:56

Giving up _do_read(...) after 6 tries (asyncio.exceptions.TimeoutError)
```

```
Logger: huawei_solar.huawei_solar
Source: /usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py:99
First occurred: 20:17:56 (1 occurrences)
Last logged: 20:17:56

Got error Failed to read register 47000 after 6 tries while trying to determine battery.
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/asyncio/tasks.py", line 456, in wait_for
    return fut.result()
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/backoff/_async.py", line 151, in retry
    ret = await target(*args, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 282, in _do_read
    response = await self._client.protocol.read_holding_registers(
  File "/usr/local/lib/python3.10/site-packages/pymodbus/client/base.py", line 303, in execute
    resp = await asyncio.wait_for(req, timeout=self.params.timeout)
  File "/usr/local/lib/python3.10/asyncio/tasks.py", line 458, in wait_for
    raise exceptions.TimeoutError() from exc
asyncio.exceptions.TimeoutError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 89, in _initialize
    self.battery_type = (await self.get(rn.STORAGE_UNIT_1_PRODUCT_MODEL)).value
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 200, in get
    return (await self.get_multiple([name], slave))[0]
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 235, in get_multiple
    response = await self._read_registers(registers[0].register, total_length, slave)
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 314, in _read_registers
    result = await _do_read()
  File "/usr/local/lib/python3.10/site-packages/backoff/_async.py", line 159, in retry
    await _call_handlers(on_giveup, **details, exception=e)
  File "/usr/local/lib/python3.10/site-packages/backoff/_async.py", line 37, in _call_handlers
    await handler(details)
  File "/usr/local/lib/python3.10/site-packages/backoff/_async.py", line 16, in f
    return coro_or_func(*args, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 261, in backoff_giveup
    raise ReadException(f"Failed to read register {register} after {details['tries']} tries")
[... 163 lines cut ...]
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 200, in get
    return (await self.get_multiple([name], slave))[0]
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 235, in get_multiple
    response = await self._read_registers(registers[0].register, total_length, slave)
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 314, in _read_registers
    result = await _do_read()
  File "/usr/local/lib/python3.10/site-packages/backoff/_async.py", line 151, in retry
    ret = await target(*args, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 280, in _do_read
    raise ConnectionException(message)
huawei_solar.exceptions.ConnectionException: Modbus client is not connected to the inverter.

```

```
Logger: pymodbus.client.tcp
Source: runner.py:120
First occurred: 20:19:02 (7822 occurrences)
Last logged: 22:43:13

Failed to connect:
Failed to connect: [Errno 113] Connect call failed ('192.168.200.1', 6607)
Failed to connect: [Errno 104] Connect call failed ('192.168.200.1', 6607)
Failed to connect: [Errno 111] Connect call failed ('192.168.200.1', 6607)
```

mariusnauj, 2023-01-20:
Yes for me the same.  I used the connection to the internal wifi of the inverter, I do not have optimisers or battery. After upgrade integration stop working.

pomah86, 2023-01-20:
I had issue over serial connection, had to downgrade to get it back up.

“Failed to read register 43006 after 6 tries”

timconinx, 2023-01-21:
Same for me (using wifi connection), _but_ it seems that I can't get it back to work after downgrading to 1.1.6... The `Failed to connect` errors keep popping up

wlcrs, 2023-01-21:
Incomplete bugreports make it very hard for me to diagnose the problem. Please do the effort of uploading the diagnostics file, describing your installation etc.

Can you also confirm that the problem is still present when you do the upgrade when the inverter is online (eg: the sun is up and the inverter is generating power?)

pomah86, 2023-01-21:
> Incomplete bugreports make it very hard for me to diagnose the problem. Please do the effort of uploading the diagnostics file, describing your installation etc.
>
> Can you also confirm that the [...]

I just upgraded from 1.1.6 to 1.2.0 and the connection failed with following error:

“Failed to read register 43006 after 6 tries”

After restoring HA from an earlier backup I got the plugin back to version 1.1.6 and connection was back up. Before restoring backup I tried to reload the integration and reboot HA but both did not give any positive out come.

timconinx, 2023-01-22:
For me, using modbus over wifi, the exact error both with 1.1.6 and 1.2.0 is

```Logger: pymodbus.client.tcp
Source: runner.py:120
First occurred: 10:29:30 PM (7 occurrences)
Last logged: 10:33:43 PM

Failed to connect: [Errno 111] Connect call failed ('192.168.200.1', 502)
```

This message pops up independent from whether the sun is up: I can confirm that the inverter is always working. Green leds are popping and the fusionsolar app is confirming everything is okay.

rubenbrieva, 2023-01-24:
In my case  I used the connection to the home wifi of the inverter. After upgrade integration stop working. But reviewing the data from the inverter I discovered that the problem is in the firmware update that blocks the modbus over home wifi.
You need to use the inverter AP for connect to modbus.
In my case, the update of the home assistant module with the OTA update of the inverters was a coincidence.

The easy way to test is download a modbus app in the mobile phone and check to connect using the inverter AP.

Depends the firmware version you have the option to enter in the inverter config as a installer and change the security options for the modbus service. The problem is that you need the installer password for that. The default is "00000a" but normally was replaced by the installer company.

Using inverter AP the integration has worked perfectly again

timconinx, 2023-01-25:
Thank you, I was already using the inverter AP, but it seemed the firmware update was indeed to blame: it changed the port from 502 to 6607 :/
Incidentally, 1.2.0 also works flawless

mariusnauj, 2023-01-27:
Does anyone have this version 1.2.0 working when using the inverter's internal wi-fi ?

timconinx, 2023-01-27:
For me its working:
```
{
  "home_assistant": {
    "installation_type": "Home Assistant Core",
    "version": "2023.1.7",
    "dev": false,
    "hassio": false,
    "virtualenv": true,
    "python_version": "3.10.9",
    "docker": false,
    "arch": "x86_64",
    "timezone": "Europe/Brussels",
    "os_name": "Linux",
    "os_version": "6.1.0-1-amd64",
    "run_as_root": false
  },
...
    "huawei_solar": {
      "version": "1.2.0",
      "requirements": [
        "huawei-solar==2.2.0"
      ]
    }
  },
  "integration_manifest": {
    "domain": "huawei_solar",
    "name": "Huawei Solar",
    "config_flow": true,
    "documentation": "https://www.home-assistant.io/integrations/huawei_solar",
    "issue_tracker": "https://github.com/wlcrs/huawei_solar/issues",
    "requirements": [
      "huawei-solar==2.2.0"
    ],
    "codeowners": [
      "@wlcrs"
    ],
    "iot_class": "local_polling",
    "version": "1.2.0",
    "loggers": [
      "huawei_solar",
      "pymodbus"
    ],
    "is_built_in": false
  },
  "data": {
    "config_entry_data": {
      "host": "192.168.200.1",
      "port": 6607,
      "slave_ids": [
        0
      ],
      "enable_parameter_configuration": false,
      "username": null,
      "password": null
    },
```

mariusnauj, 2023-01-27:
Hello @timconinx 

Which inverter model you use and what inverter firmware?

mariusnauj, 2023-01-27:
Yes for me now it's working with 1.2.1 version. But after I upgrade to 1.2.1 my other modbus integration stopped. Whats the problem?

pomah86, 2023-01-27:
After an upgrade to 1.2.1 all is now working, (modbus over usb-modbus adapter) 1.2.0 did not work before

gieljnssns, 2023-02-01:
For me it's OK now

closed 2023-02-02 by wlcrs