wlcrs/huawei_solar #264: [Bug]: pymodbus version mismatch

luzik, 2023-01-19:
### Describe the issue

After upgrading huawei_solar my other Modbus clients stopped working.

more info there:
https://github.com/home-assistant/core/issues/85869

Is there any workaround ?

### Bescribe your Huawei Solar Setup

-

### How do you connect to the inverter?

Over serial, with a serial-to-USB stick

### Upload your Diagnostics File

-

### Upload your relevant debug logs

```text
-
```

### Please confirm the following:

- [ ] The problem is still present in the latest release of this integration.
- [ ] I did not find an existing issue describing this problem.
- [ ] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [ ] I increased the logging configuration, restarted HA, and have copied all relevant log lines from the 'Full Logs' into the textarea.

wlcrs, 2023-01-19:
Wait with the upgrade until the release of HA 2023.2 . The pyModbus library was upgraded in https://github.com/home-assistant/core/pull/85961 , and that will resolve the version conflict.

luzik, 2023-01-19:
My installer upgraded my devices firmware, so this was the only way to get optimizers working again. Now I have to choice between optimizers and rest of my Modbus devices ?
Is huawei_solar safe downgradable ?

wlcrs, 2023-01-19:
Yes

luzik, 2023-01-19:
Ok, downgrade helped. It would be awesome to have branch with old pymodbus and new firmware support.
Please leave this issue open until home assistant upgrade or some other fix

steinaaj, 2023-01-22:
Agreed - the 1.2 upgrade seems amazing. Can't wait for a Home Assistant-fix!

martinarva, 2023-01-25:
Upgraded to 1.2 and i'm on Home Assistant 2023.1.7, Frontend 20230110.0 - latest.

After restart there is an error:
#######
Logger: homeassistant.setup
Source: setup.py:210
First occurred: 2:14:18 PM (1 occurrences)
Last logged: 2:14:18 PM

Setup failed for custom integration huawei_solar: Unable to import component: cannot import name 'AsyncModbusSerialClient' from 'pymodbus.client' (/usr/local/lib/python3.10/site-packages/pymodbus/client/__init__.py)
########

Tried to downgrade to 1.16, but error remains.

Is it related to same issue here?

twarnick, 2023-01-25:
> Ok, downgrade helped. It would be awesome to have branch with old pymodbus and new firmware support. Please leave this issue open until home assistant upgrade or some other fix

Hi, can you give me a hint, how to downgrade? I have the same problem.

martinarva, 2023-01-25:
> > Ok, downgrade helped. It would be awesome to have branch with old pymodbus and new firmware support. Please leave this issue open until home assistant upgrade or some other fix
> 
> 
> 
> Hi, can you give me a hint, how to downgrade? I have the same problem.

[image]

[image]

mariusnauj, 2023-01-27:
Hello,

After I upgrade Huawei solar integration to 1.2.1 my other modus integration stopped. When this will be solved?

tomderidder, 2023-01-27:
Normally with HA release 2023.2
https://github.com/wlcrs/huawei_solar/issues/264#issuecomment-1397525530

mariusnauj, 2023-02-02:
Hello,

Maybe someone tried the new version of this integration and HA v2013.2 and can confirm that after upgrade huawei integration to v1.2.3 and other modbus integration working without problems?

guevara777, 2023-02-02:
I can confirm it Runs without problems on HA2023.2.0 and Version 1.2.3

closed 2023-02-02 by luzik

scooterama, 2023-06-26:
same Problem - no config running - installation not possible
"message":"Invalid handler specified"

Home Assistant 2023.6.3
Supervisor 2023.06.2

I've tried to downgrade / upgrade to
1.2.3
1.1.6
1.2.5
1.2.6b7