wlcrs/huawei_solar #265: [Bug]: removing and re-adding Integration does not allow switching to evelvated permissions

r13ssh, 2023-01-20:
### Describe the issue

Hi,

I ran into a similar issue as discussed in https://github.com/wlcrs/huawei_solar/discussions/190, where originally I had added my inverter(s) without elevated permissions, but would now like to enable them.

Following the suggestions in https://github.com/wlcrs/huawei_solar/discussions/190#discussioncomment-4241862 I tried to remove and reinstall the ingetration, but - while I could re-add the inverters no problem, with the history retained - I was not prompted for the installer password, and I do not seem to have elevated permissions.

Did I do something wrong?

### Bescribe your Huawei Solar Setup

Inverter Type: SUN2000-5KTL-L1 (master), SUN2000-4KTL-L1 (slave)
Inverter Firmware version:
SDongle present: yes
Power meter present: single phase
Battery: None
Battery Firmware version: n/a

### How do you connect to the inverter?

Via the SDongle, wireless connection

### Upload your Diagnostics File

Diagnostics File (before reinstall): 

[config_entry-huawei_solar-6bc38b13c42dd33e898c51bf2e025c08.json.txt](https://github.com/wlcrs/huawei_solar/files/10470728/config_entry-huawei_solar-6bc38b13c42dd33e898c51bf2e025c08.json.txt)

Diagnostics File (after reinstall):
[config_entry-huawei_solar-28a773df6e4f2ca6279ac53f82103d62.json.txt](https://github.com/wlcrs/huawei_solar/files/10470732/config_entry-huawei_solar-28a773df6e4f2ca6279ac53f82103d62.json.txt)

### Upload your relevant debug logs

```text
2023-01-20 22:39:03.338 WARNING (SyncWorker_2) [homeassistant.loader] We found a custom integration wundergroundpws which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2023-01-20 22:39:03.338 WARNING (SyncWorker_2) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2023-01-20 22:39:03.339 WARNING (SyncWorker_2) [homeassistant.loader] We found a custom integration battery_sim which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2023-01-20 22:39:03.339 WARNING (SyncWorker_2) [homeassistant.loader] We found a custom integration huawei_solar which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2023-01-20 22:39:07.871 WARNING (MainThread) [homeassistant.helpers.frame] Detected integration that accesses the `is_metric` property of the unit system. This is deprecated and will stop working in Home Assistant 2023.1. Please adjust to use instance check instead.. Please report issue to the custom integration author for wundergroundpws using this method at custom_components/wundergroundpws/sensor.py, line 353: if hass.config.units.is_metric:
2023-01-20 22:39:19.134 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'homeassistant.util.read_only_dict.ReadOnlyDict object' has no attribute 'current_power_w' when rendering '{{ states.switch.fritz_dect_200_6.attributes.current_power_w }}'
2023-01-20 22:39:19.136 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'homeassistant.util.read_only_dict.ReadOnlyDict object' has no attribute 'current_power_w' when rendering '{{ states.switch.fritz_dect_200_6.attributes.current_power_w }}'
2023-01-20 22:39:19.137 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'homeassistant.util.read_only_dict.ReadOnlyDict object' has no attribute 'total_consumption' when rendering '{{ states.switch.fritz_dect_200_6.attributes.total_consumption }}'
2023-01-20 22:39:19.138 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'homeassistant.util.read_only_dict.ReadOnlyDict object' has no attribute 'total_consumption' when rendering '{{ states.switch.fritz_dect_200_6.attributes.total_consumption }}'
2023-01-20 22:39:19.138 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'homeassistant.util.read_only_dict.ReadOnlyDict object' has no attribute 'current_power_w' when rendering '{{ states.switch.termo.attributes.current_power_w }}'
2023-01-20 22:39:19.139 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'homeassistant.util.read_only_dict.ReadOnlyDict object' has no attribute 'current_power_w' when rendering '{{ states.switch.termo.attributes.current_power_w }}'
2023-01-20 22:39:19.140 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'homeassistant.util.read_only_dict.ReadOnlyDict object' has no attribute 'total_consumption' when rendering '{{ states.switch.termo.attributes.total_consumption }}'
2023-01-20 22:39:19.141 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'homeassistant.util.read_only_dict.ReadOnlyDict object' has no attribute 'total_consumption' when rendering '{{ states.switch.termo.attributes.total_consumption }}'
2023-01-20 22:39:19.142 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'homeassistant.util.read_only_dict.ReadOnlyDict object' has no attribute 'current_power_w' when rendering '{{ states.switch.fritz200.attributes.current_power_w }}'
2023-01-20 22:39:19.142 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'homeassistant.util.read_only_dict.ReadOnlyDict object' has no attribute 'current_power_w' when rendering '{{ states.switch.fritz200.attributes.current_power_w }}'
2023-01-20 22:39:19.143 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'homeassistant.util.read_only_dict.ReadOnlyDict object' has no attribute 'total_consumption' when rendering '{{ states.switch.fritz200.attributes.total_consumption }}'
2023-01-20 22:39:19.144 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'homeassistant.util.read_only_dict.ReadOnlyDict object' has no attribute 'total_consumption' when rendering '{{ states.switch.fritz200.attributes.total_consumption }}'
2023-01-20 22:39:19.144 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'homeassistant.util.read_only_dict.ReadOnlyDict object' has no attribute 'current_power_w' when rendering '{{ states.switch.recirculation_pump.attributes.current_power_w }}'
2023-01-20 22:39:19.145 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'homeassistant.util.read_only_dict.ReadOnlyDict object' has no attribute 'current_power_w' when rendering '{{ states.switch.recirculation_pump.attributes.current_power_w }}'
2023-01-20 22:39:19.145 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'homeassistant.util.read_only_dict.ReadOnlyDict object' has no attribute 'total_consumption' when rendering '{{ states.switch.recirculation_pump.attributes.total_consumption }}'
2023-01-20 22:39:19.146 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'homeassistant.util.read_only_dict.ReadOnlyDict object' has no attribute 'total_consumption' when rendering '{{ states.switch.recirculation_pump.attributes.total_consumption }}'
2023-01-20 22:39:19.147 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'homeassistant.util.read_only_dict.ReadOnlyDict object' has no attribute 'current_power_w' when rendering '{{ states.switch.fritz_dect_210_7.attributes.current_power_w }}'
2023-01-20 22:39:19.147 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'homeassistant.util.read_only_dict.ReadOnlyDict object' has no attribute 'current_power_w' when rendering '{{ states.switch.fritz_dect_210_7.attributes.current_power_w }}'
2023-01-20 22:39:19.148 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'homeassistant.util.read_only_dict.ReadOnlyDict object' has no attribute 'total_consumption' when rendering '{{ states.switch.fritz_dect_210_7.attributes.total_consumption }}'
2023-01-20 22:39:19.148 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'homeassistant.util.read_only_dict.ReadOnlyDict object' has no attribute 'total_consumption' when rendering '{{ states.switch.fritz_dect_210_7.attributes.total_consumption }}'
[... cut ...]
2023-01-20 22:44:12.886 WARNING (MainThread) [zigpy_deconz.api] No response to 'Command.aps_data_indication' command with seq id '0x2c'
```

### Please confirm the following:

- [X] The problem is still present in the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [X] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [ ] I increased the logging configuration, restarted HA, and have copied all relevant log lines from the 'Full Logs' into the textarea.

wlcrs, 2023-01-21:
Can you enable extensive logging as described on the bottom of the README, restart HA and then re-add the integration? Finally download the "Full logs" and upload them here (feel free to filter out any lines unrelated to the integration). This will generate a log showing me all the messages sent/received to/from the inverter and will allow me to diagnose the problem.

steve-hawkins, 2023-01-23:
@wlcrs I am also experiencing issues with elevated permission operations, my use case is changing battery configuration

I didn't want to raise a new issue - bug as it is more than likely the same as @r13ssh is running into, but let me know if you want me to raise a new one instead

here are the details after enabling extensive logging as you have suggested: -

### Describe your Huawei Solar Setup
Inverter Type: SUN2000-6KTL-L1
Inverter Firmware version: V200R001C00SPC124
SDongle present: no
Power meter present: single phase
Battery: LUNA2000 10kWh
Battery Firmware version: V100R002C00SPC116

### How do you connect to the inverter?
Via SUN2000-<serial_no> WiFi

### Upload your Diagnostics File
Diagnostics File:

[config_entry-huawei_solar-e698316461e8c0ed05ffbd592187e6c1.json.txt](https://github.com/wlcrs/huawei_solar/files/10482435/config_entry-huawei_solar-e698316461e8c0ed05ffbd592187e6c1.json.txt)

### Upload your relevant debug logs
[home-assistant_2023-01-23T17-59-18.918Z.log](https://github.com/wlcrs/huawei_solar/files/10482438/home-assistant_2023-01-23T17-59-18.918Z.log)

TriploidTree, 2023-01-28:
Same as above, I started with just the regular permissions and tried to get the elevated permissions, but no dice.

Things I have tried:
- Delete the device, re-add - doesn't help
- Delete the device, powercycle the system - doesn't help
- Delete the device, delete the integration from HACS, powercycle the system - doesn't help

*BUT* the weird thing is, when I first tried with "elevated permissions" and wasn't offered the password, I toggled "charge from grid" to off to see if I had the permissions. 
HA blinked the toggle to off and back to on, and "you don't have permissions for that".
Then the following day, I noticed the battery didn't charge from grid overnight (It's winter and dull here).
So  I tried toggling to "on", and same again, HA said no, but later the switch was indeed toggled to "on".

So the short term workaround, while this bug gets fixed, that I'm going to try is:
- Set schedule for charging from grid through the FusionSolar web interface
- See if I can toggle charging from grid on/off from HA based on the forecast (automation)

(Didn't grab log files yet, running v1.2.1 of the integration, HA 2023.1.7 on a pi. smart dongle running V100R001C00SPC125, 

  | SUN2000-6KTL-L1 | V200R001C00SPC125 )

wlcrs, 2023-01-28:
I can only diagnose the core issue if I receive the full logs that I asked for in the first reply. I need to understand what your inverter responds while the integration tries to detect if it has write permission on the inverter during the initial setup.

TriploidTree, 2023-01-28:
[home-assistant_2023-01-28T15-07-45.369Z.log](https://github.com/wlcrs/huawei_solar/files/10527544/home-assistant_2023-01-28T15-07-45.369Z.log)
Here are the logs.

wlcrs, 2023-01-28:
Thank you. I see that the inverter gives a valid reply to write requests, but then decides to close the connection. This leads to all kinds of erratic behavior 😕

Do you see the same issue with v1.1.6 of this integration? Can you generate the same logs while testing? This will allow me to compare the messages being sent and the replies received to find what has changed and is causing the issues.

TriploidTree, 2023-01-28:
[home-assistant_2023-01-28T16-06-53.897Z.log](https://github.com/wlcrs/huawei_solar/files/10527654/home-assistant_2023-01-28T16-06-53.897Z.log)

HACS doesn't let you just downgrade handily, so I removed from HACS and readded 1.1.6

Same thing, elevated permissions doesn't ask you for a password.
*and* it still lets you change the "charge from grid" setting w/o the password.

wlcrs, 2023-01-30:
Thank you for the extra logs. I can confirm that it's normal that you don't have to provide login details: the inverter only requires this when using the Modbus TCP interfaces exposed on the WLAN it's connected to, or when using the inverter WiFi. The integration detects this by doing a "test-write" to the time-zone register.

I've implemented some changes and released them in v1.2.2 . I really hope they improve robustness and correctness for your case (eg. when the write-command reports a success, but the device immediately terminates the Modbus-TCP connection after it).

TriploidTree, 2023-02-07:
So the "charge from grid" toggle won't stay toggled in the short term, but later when you go back it's set how you intended. I was trawling the home assistant forums looking for something else and saw you had a comment about backup boxes interfering with the modbus. It may be worth noting that I also have a backup box.

wlcrs, 2023-02-08:
The delay in the "charge from grid" toggle displaying the changed state has been fixed in 1.2.4

thbiela, 2023-02-09:
I also have this issue. Already upgraded to 1.2.4, no luck. I had added my inverters without elevated permissions, but would now like to enable them to conditionally enable "charge from grid".

Removed and readded the integration (with elevate permission checked) without getting password prompt. I am connected to the Modbus TCP port exposed by my dongle on my home network.

When I try to toggle the battery switch I get this error:

`Error executing service: <ServiceCall switch.turn_off (c:01GRVVM429FJP2JH40PDYPTVJ2): entity_id=['switch.battery_charge_from_grid_2']>
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/core.py", line 1805, in catch_exceptions
    await coro_or_task
  File "/usr/src/homeassistant/homeassistant/core.py", line 1824, in _execute_service
    await cast(Callable[[ServiceCall], Awaitable[None]], handler.job.target)(
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 213, in handle_service
    await service.entity_service_call(
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 681, in entity_service_call
    future.result()  # pop exception if have
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 968, in async_request_call
    await coro
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 721, in _handle_entity_call
    await result
  File "/config/custom_components/huawei_solar/switch.py", line 185, in async_turn_off
    await self.coordinator.async_request_refresh()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 189, in async_request_refresh
    await self._debounced_refresh.async_call()
  File "/usr/src/homeassistant/homeassistant/helpers/debounce.py", line 82, in async_call
    await task
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 215, in async_refresh
    await self._async_refresh(log_failures=True)
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 330, in _async_refresh
    self.async_update_listeners()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 135, in async_update_listeners
    update_callback()
  File "/config/custom_components/huawei_solar/sensor.py", line 947, in _handle_coordinator_update
    self._attr_native_value = len(data)
TypeError: object of type 'DecodeError' has no len()
`

Any ideas? I'd appreciate any help!

wlcrs, 2023-02-09:
Can you please generate more extensive debug logs for that error with this method: https://www.home-assistant.io/docs/configuration/troubleshooting/#debug-logs-and-diagnostics . I want to understand what causes that DecodeError.

thbiela, 2023-02-09:
Hi, see attached. Please let me know if you need further information. Thanks!

[home-assistant_huawei_solar_2023-02-09T20-21-50.003Z.log](https://github.com/wlcrs/huawei_solar/files/10701221/home-assistant_huawei_solar_2023-02-09T20-21-50.003Z.log)

wlcrs, 2023-02-09:
I'll need to do some thorough analysis on those debug logs. Can you describe your system please? What are the firmware versions?

thbiela, 2023-02-09:
Sure!

Dongle:
- SDongleA-05
- Version V100R001C00SPC133
- Connected to 5KTL

Battery
- Luna2000 5kWh
- Version V100R002C00SPC116
- Connected to 10KTL

5KTL
- SUN2000-5KTL-M1
- Version V100R001C00SPC148
- Modbus ID address 1
- No battery

10KTL
- SUN2000-10KTL-M1
- Version V100R001C00SPC150
- Modbus ID address 2
- Battery

Everything seems to work except of changing battery settings and changing Inverter polling frequency.

[image]
[image]

wlcrs, 2023-02-09:
Have you set your battery to TOU working mode in the past? Can you tell me what the exact configuration is that is currently set for that working mode? 

In the FusionSolar app: go to Devices -> Connected devices -> Battery -> three dots in top right corner -> Parameter settings -> working mode.

Can you screenshot the content of that page? Change the working mode to TOU if necessary and screenshot that too. You can then click cancel to abort changing the working mode for real if necessary.

wlcrs, 2023-02-09:
For reference, it should look something like this: 
[image]

thbiela, 2023-02-09:
No, I haven't set that working mode. Current working mode is max self consumption. Here are the screenshots:

[image]

Changed to TOU, without saving:

[image]

Settings:
[image]

wlcrs, 2023-02-10:
Can you test my fix by changing the `"requirements"` in your `/config/custom_components/huawei_solar/manifest.json` to

```
  "requirements": ["huawei-solar==2.2.5b1"],
```

?

Restart HA and test again please. If you run into problems, please generate new debug logs and include your diagnostics file as well.

thbiela, 2023-02-10:
After changing the requirement and reboot the integration doesn't start up anymore. This is the error message: 

`Config entry 'SUN2000-5KTL-M1GIN300-5KTL-M1' for huawei_solar integration not ready yet: Could not update HV21<redacted> values: Invalid model to decode TOU Registers for: 0; Retrying in background`

Here are the logs:

[home-assistant_huawei_solar_2023-02-10T11-07-14.364Z.log](https://github.com/wlcrs/huawei_solar/files/10706700/home-assistant_huawei_solar_2023-02-10T11-07-14.364Z.log)

wlcrs, 2023-02-10:
Can you try again with 2.2.5b2? 

Please also include the **diagnostics file** if you are still having problems.

thbiela, 2023-02-10:
It works! Thank you very much for your effort. :-)

closed 2023-03-21 by wlcrs