wlcrs/huawei_solar #270: [Feature Request]: Faster switching of Grid Charge On/Off

CommanderROR9, 2023-01-22:
### Describe your feature request

Hi,

With my Setup (Sun2000 8TKL M1, Luna2000 10kWh) and connection via Modbus/TCP to the Sdongle I get a delay of pretty much exactly 10 Minutes when I use the "Charge from Grid" switch. This makes using it for automations pretty difficult, because I need to add a 10 Minute wait after using the "Charge from Grid" switch.

Can this be changed somehow?

### Proper usage

- [X] I confirm that this is not a bug report

wlcrs, 2023-01-22:
This is not a feature request. Please fill in the bug report template with steps to reproduce, including debug logs generated as described here: https://www.home-assistant.io/docs/configuration/troubleshooting/#debug-logs-and-diagnostics

Now I have no clue what version of the integration you are using, what you mean by the 10 minute delay, etc.

closed 2023-01-22 by wlcrs

CommanderROR9, 2023-01-22:
I am sorry, but this is not a bug AFAIK. There are no errors, nothing in
the logs, it simply takes exactly 10 Minutes for the switch "Charge from
Grid" to actually enable or disable the Charge from Grid function.
But if you insist I can try and go through the procedure of creating a bug
report if that helps...

commit 6be1255d (2023-01-30): Prevent outdated state in switch entities

Fixes #270