wlcrs/huawei_solar #272: Battery configuration issue

wlcrs, 2023-01-23:
@wlcrs I am also experiencing issues with elevated permission operations, my use case is changing battery configuration

I didn't want to raise a new issue - bug as it is more than likely the same as @r13ssh is running into, but let me know if you want me to raise a new one instead

here are the details after enabling extensive logging as you have suggested: -

### Describe your Huawei Solar Setup
Inverter Type: SUN2000-6KTL-L1
Inverter Firmware version: V200R001C00SPC124
SDongle present: no
Power meter present: single phase
Battery: LUNA2000 10kWh
Battery Firmware version: V100R002C00SPC116

### How do you connect to the inverter?
Via SUN2000-<serial_no> WiFi

### Upload your Diagnostics File
Diagnostics File:

[config_entry-huawei_solar-e698316461e8c0ed05ffbd592187e6c1.json.txt](https://github.com/wlcrs/huawei_solar/files/10482435/config_entry-huawei_solar-e698316461e8c0ed05ffbd592187e6c1.json.txt)

### Upload your relevant debug logs
[home-assistant_2023-01-23T17-59-18.918Z.log](https://github.com/wlcrs/huawei_solar/files/10482438/home-assistant_2023-01-23T17-59-18.918Z.log)

_Originally posted by @steve-hawkins in https://github.com/wlcrs/huawei_solar/issues/265#issuecomment-1400755606_

wlcrs, 2023-01-23:
@steve-hawkins Your diagnostics file shows that the configuration parameters are correctly populated. This is different from the original issue. Can you describe exactly what your problem is, preferably with screenshots illustrating your problem?

steve-hawkins, 2023-01-23:
none of the configuration changes that require elevated permissions appear to work, as an example if I set the switch.battery_charge_from_grid entity value to false / off it shows as off in Home Assistant yet it is not then reflected in the Huawei configuration: -

[image]

[image]

wlcrs, 2023-01-23:
Can you please generate debug logs via the method described here: https://www.home-assistant.io/docs/configuration/troubleshooting/#debug-logs-and-diagnostics

Please mention what changes to the configuration you tried to make while debug logging was enabled. Which one succeeded? Which didn't?

steve-hawkins, 2023-01-23:
Configuration changes whilst running debug logs: -

- Charge from grid (from on to off) = didn't work
- Grid charge cutoff SOC (from 100% to 80%) = didn't work
- Excess PV energy use in TOU (from "Charge" to "Fed to grid") = didn't work
- Working Mode (from "Time of Use" to "Maximum Self Consumption") = succeeded
- Working Mode (from "Maximum Self Consumption" to "Time of Use") = succeeded

[home-assistant_huawei_solar_2023-01-23T19-11-36.987Z.log](https://github.com/wlcrs/huawei_solar/files/10483005/home-assistant_huawei_solar_2023-01-23T19-11-36.987Z.log)

steve-hawkins, 2023-01-29:
Thanks @wlcrs I'll do some more testing as I starting looking into this when turning off charge from grid whilst away from the property for a couple of days. Doing this through home assistant showed it was off, but the overnight TOU charging period still pulled from the grid.

etamtlosz, 2023-02-02:
Someone in the HA forum mentioned that it takes 10 minutes for the "Charge from grid" to be activated/deactivated after he flips the "Battery Charge from Grid" switch.
I tried the replicate it with my setup and is doing the same.

Steve, did you try to wait for the 10 minutes to pass before checking in the fusion solar app/commisioning?

zbuh, 2023-03-27:
I'm also seeing this behaviour, even after like 30 minutes.
 Not sure if the changes are being applied or not.

closed 2023-06-08 by stale[bot]