wlcrs/huawei_solar #281: [Bug]: Display below zero optimiser temperature

Tuningmormota, 2023-01-31:
### Describe the issue

The optimiser temperature value in case of negativ temperatures are are shown incorrectly. -1°C shown as 6552.5°C. (looks like signed integer coneverted to unsigned one, or a revers convertion missing). 
[image]

### Bescribe your Huawei Solar Setup

Inverter Type: SUN2000-2KTL-L0
Inverter Firmware version: SPC124
SDongle present: no
Power meter present:  no
Battery:  None
Battery Firmware version: -

HA is connected to inverter via the built-in WLAN, but not through the comissoning SSID. (Inverter is on our home network, modbus poprt 6607) -> Selection option missing from the list below for this case.

### How do you connect to the inverter?

Via the `SUN2000-<serial_no>` WiFi

### Upload your Diagnostics File

Drag & Drop your Diagnostics File here.
[config_entry-huawei_solar-814235ca7abc6fa49382432899bdd713.json.txt](https://github.com/wlcrs/huawei_solar/files/10543830/config_entry-huawei_solar-814235ca7abc6fa49382432899bdd713.json.txt)

### Upload your relevant debug logs

```text
I do not have any bedug log for this issue yet due to weather conditions.
```

### Please confirm the following:

- [X] The problem is still present in the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [X] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [ ] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

wlcrs, 2023-01-31:
I checked the register definitions in https://github.com/wlcrs/huawei-solar-lib/blob/master/src/huawei_solar/registers.py, and all temperature fields are of the type I16.

I will need the debug logs to verify where it is going wrong. You can use the built-in functionality described here to easily generate them: https://www.home-assistant.io/docs/configuration/troubleshooting/#debug-logs-and-diagnostics . Please make sure to create them while the problem is manifesting itself.

Tuningmormota, 2023-01-31:
Thanks, I will prepare the logging in case any following morning would be cold enough

Tuningmormota, 2023-02-06:
[home-assistant_huawei_solar_2023-02-06T07-29-29.761Z.log](https://github.com/wlcrs/huawei_solar/files/10660665/home-assistant_huawei_solar_2023-02-06T07-29-29.761Z.log)
Finally I cloud make a log of it. Some of the optimisera have positive and some of the negatv temprature readings.
[config_entry-huawei_solar-814235ca7abc6fa49382432899bdd713.json (1).txt](https://github.com/wlcrs/huawei_solar/files/10660671/config_entry-huawei_solar-814235ca7abc6fa49382432899bdd713.json.1.txt)

wlcrs, 2023-02-06:
Aha, I didn't realize you were talking about the temperature fields from the optimizers. Huawei did not document whether these fields are signed or unsigned. I incorrectly assumed they were unsigned 😅

I'll update the parsing of the optimizer data to reflect your finding. Thank you!

closed 2023-02-06 by wlcrs

luzik, 2023-02-07:
Any hint how to clear database statistics for bad values ? Maybe it's time for
https://www.home-assistant.io/integrations/repairs/ ??

or maybe you can help with sql ?
update statistics update value=fix(value) where id in (x,y,x) and value>6550 and value < 6554