wlcrs/huawei_solar #286: [Bug]: Controls in battery (charge from grid) not working

bekovan, 2023-02-05:
### Describe the issue

Controls in battery not working / grayed out

charge from grid : toggle switch not working, will go back to original setting
capacity control mode : grayed out

Charge from grid toggle is working on 1.1.6

### Bescribe your Huawei Solar Setup

Inverter Type: SUN2000-3.68KTL-L1
Inverter Firmware version: V200R001C00SPC125
SDongle present: no
Power meter present: three phase 
Battery:  / LUNA2000 5kWh 
Battery Firmware version: V100R002C00SP116

### How do you connect to the inverter?

Over serial, with a serial-to-USB stick

### Upload your Diagnostics File

Drag & Drop your Diagnostics File here.
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.2.2",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.7",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Europe/Brussels",
    "os_name": "Linux",
    "os_version": "5.15.84-v8",
    "supervisor": "2023.01.1",
    "host_os": "Home Assistant OS 9.5",
    "docker_version": "20.10.22",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {
    "hacs": {
      "version": "1.30.1",
      "requirements": [
        "aiogithubapi>=22.10.1"
      ]
    },
    "nordpool": {
      "version": "0.0.11",
      "requirements": [
        "nordpool>=0.2"
      ]
    },
    "huawei_solar": {
      "version": "1.2.3",
      "requirements": [
        "huawei-solar==2.2.3"
      ]
    }
  },
  "integration_manifest": {
    "domain": "huawei_solar",
    "name": "Huawei Solar",
    "config_flow": true,
    "documentation": "https://www.home-assistant.io/integrations/huawei_solar",
    "issue_tracker": "https://github.com/wlcrs/huawei_solar/issues",
[... 442 lines cut ...]
2023-02-05 15:14:31.745 DEBUG (MainThread) [pymodbus.transaction] Getting transaction 1
2023-02-05 15:14:31.745 DEBUG (MainThread) [pymodbus.framer.rtu_framer] Frame - [b'\x01\x06\xb7\xef\x00\x01^K'] not ready
2023-02-05 15:14:35.003 DEBUG (MainThread) [pymodbus.payload] [1]
2023-02-05 15:14:35.004 DEBUG (MainThread) [pymodbus.payload] [1]
2023-02-05 15:14:35.005 DEBUG (MainThread) [huawei_solar.huawei_solar] Writing to 47087: [1]
2023-02-05 15:14:35.005 DEBUG (MainThread) [pymodbus.client.base] send: 0x1 0x6 0xb7 0xef 0x0 0x1 0x5e 0x4b
2023-02-05 15:14:35.007 DEBUG (MainThread) [pymodbus.transaction] Adding transaction 1
2023-02-05 15:14:35.047 DEBUG (MainThread) [pymodbus.client.base] recv: 0x1 0x6 0xb7 0xef 0x0 0x1 0x5e 0x4b
2023-02-05 15:14:35.048 DEBUG (MainThread) [pymodbus.framer.rtu_framer] Getting Frame - 0x6 0xb7 0xef 0x0 0x1
2023-02-05 15:14:35.048 DEBUG (MainThread) [pymodbus.factory] Factory Response[WriteSingleRegisterResponse': 6]
2023-02-05 15:14:35.049 DEBUG (MainThread) [pymodbus.framer.rtu_framer] Frame advanced, resetting header!!
2023-02-05 15:14:35.049 DEBUG (MainThread) [pymodbus.transaction] Getting transaction 1
2023-02-05 15:14:35.049 DEBUG (MainThread) [pymodbus.framer.rtu_framer] Frame - [b'\x01\x06\xb7\xef\x00\x01^K'] not ready
2023-02-05 15:15:00.440 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064
2023-02-05 15:15:00.441 DEBUG (MainThread) [pymodbus.client.base] send: 0x1 0x3 0x7d 0x40 0x0 0x34 0x5d 0xa5
2023-02-05 15:15:00.443 DEBUG (MainThread) [pymodbus.transaction] Adding transaction 1
2023-02-05 15:15:00.506 DEBUG (MainThread) [pymodbus.client.base] recv: 0x1 0x3 0x68 0x0 0x0 0x1 0x7e 0x9 0x47 0x0 0x0 0x0 0x0 0x4 0xa2 0x0 0x3 0x0 0x0 0x0 0x0 0x4 0xc 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0
```

### Please confirm the following:

- [X] The problem is still present in the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [X] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [X] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

wlcrs, 2023-02-08:
'Charge from grid' and 'capacity control' are interlocked: charge from grid must be enabled if you want to enable capacity control.

This means:
- capacity control will be unavailable when 'charge from grid' is disabled
- 'charge from grid' will appear unavailable when capacity control is enabled (HA incorrectly shows it as disabled when I report this entity as being unavailable). You need to disable capacity control before you can effectively disable 'charge from grid'.

bekovan, 2023-02-10:
Seems to be working now, after upgrading to 1.2.4. 
I will start "playing" with the capacity control options in the weekend
Thanks for solving the issue !

closed 2023-02-10 by wlcrs