wlcrs/huawei_solar #293: [Feature Request]: Phase power exposed as entity

Rvh91, 2023-02-16:
### Describe your feature request

Aside from having the current and voltage of each phase exposed, it would be great to also have the power per phase. This prevents the need from creating template entities in home assistant, resulting in a cleaner config. 

### Proper usage

- [X] I confirm that this is not a bug report

wlcrs, 2023-02-16:
These are available but not enabled by default. You can find and enable them in the settings:

[image]

closed 2023-02-16 by wlcrs

Rvh91, 2023-02-16:
Ok great to hear that this is already there. However, it appears that for me it is not listed. What could be the reason for this?

[image]

wlcrs, 2023-02-16:
I misread your initial request.

The goal of this integration is to make the information exposed by Huawei inverters via Modbus available in HA. As the inverters do not have a `power` register, it is not within the scope of this integration.

Using a template entity to compute the power on each phase using the voltage and current is the correct way to retrieve this information.

Of course, this gives you a result in VA (apparent power) instead of W (real power), so you need to include the power factor to be completely correct. (and this is just one of the examples that show how computing values myself in this integration is dangerous, and thus should not be a part of this integration).

Rvh91, 2023-02-16:
Thank you for the info! Makes sense not to include it in the integration indeed if it is not directly exposed by Huawei. 
This power meter that you showed a screen shot from, is that something that is attached to the inverter externally, and provides entities separate from your huawei solar integration?

The power factor is interesting to me, I wasn't aware of this. Is it related to voltage and current being out of phase? 

also to compute the real power would the following be correct?
voltage_<phase>*current_<phase>*powerfactor? where powerfactor appears is identical across al phases?