wlcrs/huawei_solar #307: [Bug]: Storage_total_discharge and storage_total_charge wrong values

kalika303, 2023-02-27:
### Describe the issue

Hello.

Everything was working as expected, but suddenly we got a strange value in the following fields:

storage_total_discharge
storage_total_charge

```json
"storage_total_charge": [
 34911.75,
  "kWh"
],
"storage_total_discharge": [
  4295259.47,
  "kWh"
],
```

Any idea what may cause this issue? Forcing a value change in the recorder at "Devloper/Statistics" section does not work.

Thanks!

### Bescribe your Huawei Solar Setup

Inverter Type: SUN2000-6KTL-L1
Inverter Firmware version: V200R001C00SPC127
SDongle present: yes
Power meter present: single phase
Battery: LUNA2000 30kWh
Battery Firmware version: V100R002C00SPC118

### How do you connect to the inverter?

Over serial, with a serial-to-USB stick

### Upload your Diagnostics File

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.2.5",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.7",
    "docker": true,
    "arch": "aarch64",
    "timezone": "",
    "os_name": "Linux",
    "os_version": "5.15.84-v8",
    "supervisor": "2023.01.1",
    "host_os": "Home Assistant OS 9.5",
    "docker_version": "20.10.22",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {
    "daikin_residential_altherma": {
      "version": "2.1.0",
      "requirements": [
        "oic==1.4.0"
      ]
    },
    "worlds_air_quality_index": {
      "version": "0.3.5",
      "requirements": []
    },
[... 387 lines cut ...]
      ],
      "storage_capacity_control_periods": [
        [],
        null
      ]
    },
    "slave_0_optimizer_data": {}
  }
}
```

### Upload your relevant debug logs

```text
Nothing special except for this:

2023-02-27 16:15:52.992 ERROR (MainThread) [custom_components.huawei_solar] Timeout fetching HV2220041572_configuration_update_coordinator data
```

### Please confirm the following:

- [X] The problem is still present in the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [X] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [X] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

wlcrs, 2023-02-27:
What do you see in FusionSolar (app or website)? 

Please generate debugging logs for a few minutes so that I can review the Modbus traffic

kalika303, 2023-02-27:
> What do you see in FusionSolar (app or website)?

[image]

> Please generate debugging logs for a few minutes so that I can review the Modbus traffic

Can you point me to the documentation on how to enable debugging logs?
Thanks.

wlcrs, 2023-03-01:
I need the pymodbus logs too. Please use the built in diagnostics tool: https://www.home-assistant.io/docs/configuration/troubleshooting/#debug-logs-and-diagnostics

kalika303, 2023-03-02:
> I need the pymodbus logs too. Please use the built in diagnostics tool: https://www.home-assistant.io/docs/configuration/troubleshooting/#debug-logs-and-diagnostics

I used the diagnostic tool and also tried to add the following (as stated here https://www.home-assistant.io/integrations/modbus/#opening-an-issue):

```yaml
logger:
  default: warning
  logs:
    homeassistant.components.modbus: debug
    pymodbus: debug
```

Still did not see different logs related to pymodbus in the output.
Any other idea?

kalika303, 2023-03-05:
@wlcrs is there anything I can force my manually trigger some actions to give you more interesting logs and feedback? If you have some idea, I can try to debug the issue and some tests, but I really need some direction...

wlcrs, 2023-03-05:
I need pymodbus log lines to further evaluate your problem. 

You can also contact Huawei support to ask them why your battery is reporting strange values

kalika303, 2023-03-06:
No way. I've tried to use the configuration indicated here: https://github.com/wlcrs/huawei_solar#faq---troubleshooting
After more than 100MB of log, no log concerning modbus has been recorded.

Is there any other place to check for log? Can I place a logger manually in the code somewhere?

kalika303, 2023-03-08:
> You can also contact Huawei support to ask them why your battery is reporting strange values

In the FusionSolar app the report is correct. So I feel like it is an error in my installation of HA.
What I can do is trying to read the modbus register manually from the cli and see if it reports the correct value, so we can narrow down between a modbus/record issue in HA.

Makes sense?

wlcrs, 2023-03-09:
I finally understand why you are not getting pymodbus logs. In v1.2.5 of this integration I have updated pymodbus to v3.1.3, which apparently includes an unexpected and breaking change in how logging is configured: https://github.com/pymodbus-dev/pymodbus/pull/1324

You should be able to get proper logs by downgrading to 1.2.4 (and thus to pymodbus 3.1.2).

turkano, 2023-04-24:
I've the same problem last month.

svencharleer, 2023-05-12:
Hope these help?

`{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.5.2",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.11",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Europe/Brussels",
    "os_name": "Linux",
    "os_version": "6.1.21-v8",
    "supervisor": "2023.04.1",
    "host_os": "Home Assistant OS 10.1",
    "docker_version": "23.0.3",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {
    "huawei_solar": {
      "version": "1.2.6b5",
      "requirements": [
        "huawei-solar==2.2.7b5"
      ]
    },
    "hacs": {
      "version": "1.32.1",
      "requirements": [
        "aiogithubapi>=22.10.1"
      ]
    },
    "audiconnect": {
      "version": "2021.1.1",
      "requirements": [
        "beautifulsoup4"
      ]
    }
  },
  "integration_manifest": {
    "domain": "huawei_solar",
    "name": "Huawei Solar",
    "config_flow": true,
    "documentation": "https://www.home-assistant.io/integrations/huawei_solar",
    "issue_tracker": "https://github.com/wlcrs/huawei_solar/issues",
    "requirements": [
      "huawei-solar==2.2.7b5"
    ],
    "codeowners": [
      "@wlcrs"
    ],
    "iot_class": "local_polling",
    "version": "1.2.6b5",
    "loggers": [
      "huawei_solar",
      "pymodbus"
    ],
    "is_built_in": false
  },
  "data": {
    "config_entry_data": {
      "host": "192.168.200.1",
      "port": 6607,
      "slave_ids": [
        0
      ],
      "enable_parameter_configuration": false,
      "username": null,
[... 474 lines cut ...]
2023-05-12 15:46:56.711 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x00', b'\x00\n']
2023-05-12 15:46:56.711 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x00', b'\x01h']
2023-05-12 15:46:56.712 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x00', b'\xc70']
2023-05-12 15:46:56.712 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x01', b'\xacI']
2023-05-12 15:46:56.712 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x00', b'\x00\x00']
2023-05-12 15:46:56.712 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x00', b'\x10[']
2023-05-12 15:46:56.712 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x00', b'\x10B']
2023-05-12 15:46:56.713 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x00', b'\x10H']
2023-05-12 15:46:56.713 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x00', b'\x00\xbd']
2023-05-12 15:46:56.713 DEBUG (MainThread) [pymodbus.logging] handle: [b'\xff\xff', b'\xffb']
2023-05-12 15:46:56.713 DEBUG (MainThread) [pymodbus.logging] handle: [b'\xff\xff', b'\xff\xec']
2023-05-12 15:46:56.765 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37760 with length 28 from slave 0
2023-05-12 15:46:56.766 DEBUG (MainThread) [pymodbus.logging] send: 0x0 0x2a 0x0 0x0 0x0 0x6 0x0 0x3 0x93 0x80 0x0 0x1c
2023-05-12 15:46:56.768 DEBUG (MainThread) [pymodbus.logging] Adding transaction 42
2023-05-12 15:46:56.779 DEBUG (MainThread) [pymodbus.logging] recv: 0x0 0x2a 0x0 0x0 0x0 0x3b 0x0 0x3 0x38 0x3 0x7a 0x0 0x0 0x0 0x2 0x11 0x2c 0x0 0x6b 0x0 0x0 0x12 0x6a 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x1 0xa7 0xed 0xaf 0x0 0x0 0x9c 0x1 0x0 0x0 0x3 0x39 0x0 0x0 0x1 0x2c
2023-05-12 15:46:56.780 DEBUG (MainThread) [pymodbus.logging] Processing: 0x0 0x2a 0x0 0x0 0x0 0x3b 0x0 0x3 0x38 0x3 0x7a 0x0 0x0 0x0 0x2 0x11 0x2c 0x0 0x6b 0x0 0x0 0x12 0x6a 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x1 0xa7 0xed 0xaf 0x0 0x0 0x9c 0x1 0x0 0x0 0x3 0x39 0x0 0x0 0x1 0x2c
2023-05-12 15:46:56.780 DEBUG (MainThread) [pymodbus.logging] Factory Response[ReadHoldingRegistersResponse': 3]
2023-05-12 15:46:56.780 DEBUG (MainThread) [pymodbus.logging] Getting transaction 42
2023-05-12 15:46:56.781 DEBUG (MainThread) [pymodbus.logging] [890, 0, 2, 4396, 107, 0, 4714, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 423, 60847, 0, 39937, 0, 825, 0, 300]
2023-05-12 15:46:56.781 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x00', b'\x12j']
2023-05-12 15:46:56.782 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x01\xa7', b'\xed\xaf']
2023-05-12 15:46:56.782 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x00', b'\x9c\x01']
2023-05-12 15:46:56.782 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x00', b'\x039']
2023-05-12 15:46:56.782 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x00', b'\x01,']
2023-05-12 15:46:56.783 DEBUG (MainThread) [custom_components.huawei_solar] Finished fetching HV2250368630_data_update_coordinator data in 0.328 seconds (success: True)`

svencharleer, 2023-05-12:
(if you'd prefer a file, let me know)

wlcrs, 2023-05-12:
The interesting stuff from your logs, where total charge is found in the register 7-8 and total discharge found in register 5-6 **when counting from the end**. I've highlighted those registers in the response of the inverter by annotating them with ***.

```
2023-05-12 15:46:26.735 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37760 with length 28 from slave 0
2023-05-12 15:46:26.736 DEBUG (MainThread) [pymodbus.logging] send: 0x0 0x26 0x0 0x0 0x0 0x6 0x0 0x3 0x93 0x80 0x0 0x1c
2023-05-12 15:46:26.738 DEBUG (MainThread) [pymodbus.logging] Adding transaction 38
2023-05-12 15:46:26.747 DEBUG (MainThread) [pymodbus.logging] recv: 0x0 0x26 0x0 0x0 0x0 0x3b 0x0 0x3 0x38 0x3 0x7a 0x0 0x0 0x0 0x2 0x11 0x2b 0x0 0x6d 0x0 0x0 0x12 0xd0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x1 0xa7 0xed 0xaf 0x0 0x0 0x9c 0x1 0x0 0x0 0x3 0x39 0x0 0x0 0x1 0x2c
2023-05-12 15:46:26.748 DEBUG (MainThread) [pymodbus.logging] Processing: 0x0 0x26 0x0 0x0 0x0 0x3b 0x0 0x3 0x38 0x3 0x7a 0x0 0x0 0x0 0x2 0x11 0x2b 0x0 0x6d 0x0 0x0 0x12 0xd0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 *** 0x1 0xa7 0xed 0xaf 0x0 0x0 0x9c 0x1 *** 0x0 0x0 0x3 0x39 0x0 0x0 0x1 0x2c
2023-05-12 15:46:26.748 DEBUG (MainThread) [pymodbus.logging] Factory Response[ReadHoldingRegistersResponse': 3]
2023-05-12 15:46:26.749 DEBUG (MainThread) [pymodbus.logging] Getting transaction 38
2023-05-12 15:46:26.749 DEBUG (MainThread) [pymodbus.logging] [890, 0, 2, 4395, 109, 0, 4816, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, ***423, 60847, 0, 39937***, 0, 825, 0, 300]
```

As you can see the inverter reports value 0x01a7edaf for the battery total charge register.

pyModbus correctly decodes the bytes received, as verified on https://rapidscada.net/modbus/
[image]

This 0x01a7edaf translates to 27782575 Wh, or 277825,75 kWh
[image]

So the integration receives incorrect data from the inverter. Please contact Huawei support and ask them why your inverter reports that big value when reading modbus register `37780`

svencharleer, 2023-05-12:
Oh great, thanks. Well, not great ;) But thanks for that, I'll try and contact them (I guess I'll still have to wait for the installer to give me an account before I can bug Huawei)

wlcrs, 2023-05-12:
Please report back with their answer, as I guess you won't be the last one
with this issue.

Good luck!

svencharleer, 2023-05-24:
Quick update. Been mailing back and forth with Huawei, just sent them the dongle logs, so hopefully news soon. Do realise now that I'm connecting to the inverter through _inverter wifi_ <-> pi, while I could also be connecting it through _dongle wifi_ <-> pi. Not sure if that'll make a difference.

(if anyone can explain me why there are 2 ways to connect to these inverters, would be great 😅)

wlcrs, 2023-05-24:
AFAIK the dongle has 2 use cases:
- you have multiple daisy-chained inverters that you want to monitor
- you need a wired connection to the internet because the inverter is too far away from your WiFi router.

If neither is the case, you're better of removing the dongle and complaining to your installer that they sold you something unnecessary.

svencharleer, 2023-05-31:
According to Huawei (and now that I have portal login info), the battery should be fine. So it's just an odd bus behaviour. Hope this info is useful for you in one way or the other!

wlcrs, 2023-05-31:
Closing this issue as there is no work to be done.

closed 2023-05-31 by wlcrs