wlcrs/huawei_solar #313: [Bug]: From version 1.2.5 the battery data sensor on slave inverter is unknown

lazzalf, 2023-03-05:
### Describe the issue

After upgrade from version 1.2.4 to 1.2.5 the sensor in homeassitant of battery LUNA2000 on slave inverter (SUN2000-5KTL-L1) is become all unknown.
I tried to revert the version to 1.2.4 and it returned to read the data correctly.

### Bescribe your Huawei Solar Setup

Inverter Master
Inverter Type: SUN2000-6KTL-M1 
Inverter Firmware version: V100R001C00SPC148
SDongle present: yes
Power meter present: three phase
Battery: LUNA2000 30kWh
Battery Firmware version: V100R002C00SPC116

Inverter slave
Inverter Type: SUN2000-5KTL-L1 
Inverter Firmware version: V200R001C00SPC125 
Battery: LUNA2000 15kWh
Battery Firmware version: V100R002C00SPC116

### How do you connect to the inverter?

Via the SDongle, wireless connection

### Upload your Diagnostics File

Drag & Drop your Diagnostics File here.
[config_entry-huawei_solar-7a73bb3bf67b6fc744c89c9fc13277f6_1.2.5.json.txt](https://github.com/wlcrs/huawei_solar/files/10892797/config_entry-huawei_solar-7a73bb3bf67b6fc744c89c9fc13277f6_1.2.5.json.txt)
[config_entry-huawei_solar-7a73bb3bf67b6fc744c89c9fc13277f6_1.2.4.json.txt](https://github.com/wlcrs/huawei_solar/files/10892798/config_entry-huawei_solar-7a73bb3bf67b6fc744c89c9fc13277f6_1.2.4.json.txt)

### Upload your relevant debug logs

```text
2023-03-05 20:45:08.557 WARNING (SyncWorker_2) [homeassistant.loader] We found a custom integration huawei_solar which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2023-03-05 20:45:08.570 WARNING (SyncWorker_2) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant

2023-03-05 20:46:25.985 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 43006
2023-03-05 20:46:27.399 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 47000
2023-03-05 20:46:28.975 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 30000
2023-03-05 20:46:31.318 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 30071
2023-03-05 20:46:33.523 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37200
2023-03-05 20:46:34.986 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 47000
2023-03-05 20:46:36.257 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 47089
2023-03-05 20:46:37.783 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 47954
2023-03-05 20:46:38.242 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37100
2023-03-05 20:46:39.472 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37125
2023-03-05 20:46:41.898 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 30000
2023-03-05 20:46:44.443 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 30071
2023-03-05 20:46:46.363 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37200
2023-03-05 20:46:47.549 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 47000
2023-03-05 20:46:48.929 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 47089
2023-03-05 20:46:49.782 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 47954
2023-03-05 20:46:52.172 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37100
2023-03-05 20:46:55.053 WARNING (MainThread) [homeassistant.bootstrap] Waiting on integrations to complete setup: huawei_solar
2023-03-05 20:46:56.695 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37125
2023-03-05 20:47:01.543 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064
2023-03-05 20:47:04.122 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32000
2023-03-05 20:47:05.995 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37100
2023-03-05 20:47:07.942 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37760
2023-03-05 20:47:09.034 DEBUG (MainThread) [custom_components.huawei_solar] Finished fetching TA2190034467_data_update_coordinator data in 7.492 seconds (success: True)
2023-03-05 20:47:09.036 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 47075
2023-03-05 20:47:11.547 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 47200
2023-03-05 20:47:13.702 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 47255
2023-03-05 20:47:16.190 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 47954
2023-03-05 20:47:18.750 DEBUG (MainThread) [custom_components.huawei_solar] Finished fetching TA2190034467_configuration_update_coordinator data in 9.714 seconds (success: True)
2023-03-05 20:47:18.753 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064
2023-03-05 20:47:20.455 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32000
2023-03-05 20:47:21.597 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37100
2023-03-05 20:47:23.570 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37760
[... 232 lines cut ...]
2023-03-05 20:50:48.840 DEBUG (MainThread) [custom_components.huawei_solar] Finished fetching TA2190034467_data_update_coordinator data in 6.557 seconds (success: True)
2023-03-05 20:50:48.856 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064
2023-03-05 20:50:48.865 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 182, in _handle_refresh_interval
    await self._async_refresh(log_failures=True, scheduled=True)
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 330, in _async_refresh
    self.async_update_listeners()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 135, in async_update_listeners
    update_callback()
  File "/config/custom_components/huawei_solar/sensor.py", line 851, in _handle_coordinator_update
    value = self.coordinator.data[self._register_key].value
KeyError: 'power_meter_active_power'
2023-03-05 20:50:50.349 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32000
2023-03-05 20:50:50.993 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37100
2023-03-05 20:50:52.827 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37760
2023-03-05 20:50:55.537 DEBUG (MainThread) [custom_components.huawei_solar] Finished fetching HV2180108754_data_update_coordinator data in 8.300 seconds (success: True)
```

### Please confirm the following:

- [X] The problem is still present in the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [X] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [X] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

lazzalf, 2023-03-06:
> So both inverters have batteries attached to them, 3 battery systems in total?
> 
> Your master inverter has slave-id 2, while your slave inverter has slave-id 1 ?

Yes to both, my system have 3 batteries:
- 2 LUNA2000 (15kWh + 15kWh) on master inverter SUN2000-6KTL-M1 (with modbus slave-id 2).
- 1 LUNA2000 (15kWh) on slave inverter SUN2000-5KTL-L1 (with modbus slave-id 1).

The Wifi-Dongle is physically connected to master inverter.

stones80, 2023-03-07:
I have the same problem

closed 2023-03-08 by wlcrs

wlcrs, 2023-03-10:
Can you please confirm that the issue is fixed in 1.2.6 beta 1 ?

lazzalf, 2023-03-10:
I updated to 1.2.6b1, it seems to work fine. Thank you!
If there will be problems in the next few days I will reopen the issue.