wlcrs/huawei_solar #318: [Bug]: Lots of TimeoutErrors

nicx, 2023-03-09:
### Describe the issue

I am using the integration for a few days now. Unfortunately the integration is not very reliable, in my case I get a lot or TimeoutErrors and the Sensors are unavailable.

It doesn't matter if I am using direct modbus connection of the integration to my inverter, or a modbusproxy. The error always exists in the integration. All sensors are flapping between a real state and unavailable state.

My Huawei dongle is connected via LAN and not WLAN. The ping is good and shows no abnormalities.

Why I think it is a bug in the integration: If I couple another software (eg EVCC) directly to the inverter, these problems do not exist here. All values can be read out correctly and steadily by modbus.

Any idea what could be done to stabilize the integration?

### Bescribe your Huawei Solar Setup

Inverter Type: SUN2000-10KTL-M1
Inverter Firmware version: V100R001C00SPC153
SDongle present: yes
Power meter present: three phase
Battery: LUNA2000 5kWh
Battery Firmware version: V100R002C00SPC118

### How do you connect to the inverter?

Over serial, with a serial-to-USB stick

### Upload your Diagnostics File

Drag & Drop your Diagnostics File here.

### Upload your relevant debug logs

```text
2023-03-09 11:25:41.078 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064
2023-03-09 11:25:51.080 INFO (MainThread) [backoff] Backing off _do_read(...) for 1.0s (asyncio.exceptions.TimeoutError)
2023-03-09 11:25:51.080 DEBUG (MainThread) [huawei_solar.huawei_solar] Backing off reading for 1.0 seconds after 1 tries
2023-03-09 11:25:56.947 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32000
2023-03-09 11:25:59.019 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37100
2023-03-09 11:26:31.078 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064
2023-03-09 11:26:41.080 INFO (MainThread) [backoff] Backing off _do_read(...) for 1.0s (asyncio.exceptions.TimeoutError)
2023-03-09 11:26:41.080 DEBUG (MainThread) [huawei_solar.huawei_solar] Backing off reading for 1.0 seconds after 1 tries
2023-03-09 11:26:43.232 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32000
2023-03-09 11:26:44.371 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37100
2023-03-09 11:26:44.985 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37760
2023-03-09 11:27:15.078 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064
2023-03-09 11:27:25.080 INFO (MainThread) [backoff] Backing off _do_read(...) for 1.0s (asyncio.exceptions.TimeoutError)
2023-03-09 11:27:25.081 DEBUG (MainThread) [huawei_solar.huawei_solar] Backing off reading for 1.0 seconds after 1 tries
2023-03-09 11:27:28.227 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32000
2023-03-09 11:27:30.976 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37100
2023-03-09 11:27:32.926 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37760
2023-03-09 11:28:04.079 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064
2023-03-09 11:28:14.081 INFO (MainThread) [backoff] Backing off _do_read(...) for 1.0s (asyncio.exceptions.TimeoutError)
2023-03-09 11:28:14.081 DEBUG (MainThread) [huawei_solar.huawei_solar] Backing off reading for 1.0 seconds after 1 tries
2023-03-09 11:28:15.970 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32000
2023-03-09 11:28:19.866 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37100
2023-03-09 11:28:22.404 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37760
2023-03-09 11:37:53.080 INFO (MainThread) [backoff] Backing off _do_read(...) for 1.0s (asyncio.exceptions.TimeoutError)
2023-03-09 11:37:53.080 DEBUG (MainThread) [huawei_solar.huawei_solar] Backing off reading for 1.0 seconds after 1 tries
2023-03-09 11:37:55.377 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32000
2023-03-09 11:37:57.559 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37100
2023-03-09 11:37:59.946 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37760
2023-03-09 11:38:32.079 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064
2023-03-09 11:38:42.080 INFO (MainThread) [backoff] Backing off _do_read(...) for 1.0s (asyncio.exceptions.TimeoutError)
2023-03-09 11:38:42.081 DEBUG (MainThread) [huawei_solar.huawei_solar] Backing off reading for 1.0 seconds after 1 tries
2023-03-09 11:38:45.172 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32000
2023-03-09 11:38:46.752 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37100
2023-03-09 11:38:49.223 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37760
2023-03-09 11:39:21.079 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064
2023-03-09 11:39:31.080 INFO (MainThread) [backoff] Backing off _do_read(...) for 1.0s (asyncio.exceptions.TimeoutError)
2023-03-09 11:39:31.080 DEBUG (MainThread) [huawei_solar.huawei_solar] Backing off reading for 1.0 seconds after 1 tries
2023-03-09 11:39:33.591 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32000
2023-03-09 11:39:34.628 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37100
2023-03-09 11:39:35.753 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37760
2023-03-09 11:40:09.078 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064
2023-03-09 11:40:09.078 ERROR (MainThread) [huawei_solar.huawei_solar] Modbus client is not connected to the inverter.
NoneType: None
2023-03-09 11:40:09.078 INFO (MainThread) [backoff] Backing off _do_read(...) for 1.0s (huawei_solar.exceptions.ConnectionInterruptedException: Modbus client is not connected to the inverter.)
2023-03-09 11:40:09.078 DEBUG (MainThread) [huawei_solar.huawei_solar] Backing off reading for 1.0 seconds after 1 tries
2023-03-09 11:40:10.748 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32000
2023-03-09 11:40:14.364 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37100
2023-03-09 11:40:17.355 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37760
```

### Please confirm the following:

- [X] The problem is still present in the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [ ] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [X] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

Soesjie78, 2023-03-09:
Same here. Original Huawei app is working, inverter is online. Everything is OK but no connection with the integration.

Inverter: SUN2000-3KTL-L1
Connection: WiFi

[image]

nicx, 2023-03-09:
@Soesjie78 the connection works in general in my case, I just have many timeouts and therefor flapping states of sensors. so you should have another issue ;)

Soesjie78, 2023-03-09:
Ah! yes, then the issue is not the same, my bad.

wlcrs, 2023-03-09:
So your inverter is frequently not responding within 10 seconds to queries of this integration. As my inverter does not show this behavior it is very difficult for me to diagnose the root issue.

Any chance that you can log the modbus traffic of both the integration and EVCC so that we can try to spot the differences? Maybe this is possible using that modbus proxy you're mentioning?

sportopeden, 2023-03-09:
Same here. Going offline for a minute about every hour. Since last update of Huawei integration.

wlcrs, 2023-03-09:
No

seeker1973, 2023-03-10:
I have the same problem since I installed the 153 on the inverter.

2023-03-10 07:51:51.215 ERROR (MainThread) [huawei_solar.huawei_solar] Could not read register value, has another device interrupted the connection?
2023-03-10 07:51:51.218 INFO (MainThread) [backoff] Backing off _do_read(...) for 1.0s (huawei_solar.exceptions.ConnectionInterruptedException: Could not read register value, has another device interrupted the connection?)
2023-03-10 07:51:52.219 ERROR (MainThread) [huawei_solar.huawei_solar] Modbus client is not connected to the inverter.
NoneType: None
2023-03-10 07:51:52.220 INFO (MainThread) [backoff] Backing off _do_read(...) for 2.0s (huawei_solar.exceptions.ConnectionInterruptedException: Modbus client is not connected to the inverter.)
2023-03-10 07:51:54.220 ERROR (MainThread) [huawei_solar.huawei_solar] Modbus client is not connected to the inverter.
NoneType: None
2023-03-10 07:51:54.221 INFO (MainThread) [backoff] Backing off _do_read(...) for 4.0s (huawei_solar.exceptions.ConnectionInterruptedException: Modbus client is not connected to the inverter.)
2023-03-10 07:51:58.223 ERROR (MainThread) [huawei_solar.huawei_solar] Modbus client is not connected to the inverter.
NoneType: None
2023-03-10 07:51:58.223 INFO (MainThread) [backoff] Backing off _do_read(...) for 8.0s (huawei_solar.exceptions.ConnectionInterruptedException: Modbus client is not connected to the inverter.)
2023-03-10 07:52:16.137 WARNING (MainThread) [homeassistant.components.sensor] Setup of sensor platform huawei_solar is taking over 10 seconds.
2023-03-10 09:09:48.424 INFO (MainThread) [backoff] Backing off _do_read(...) for 1.0s (asyncio.exceptions.TimeoutError)
2023-03-10 09:09:58.422 ERROR (MainThread) [custom_components.huawei_solar] Timeout fetching TA2230026735_data_update_coordinator data
2023-03-10 09:10:38.423 INFO (MainThread) [backoff] Backing off _do_read(...) for 1.0s (asyncio.exceptions.TimeoutError)
2023-03-10 09:11:28.423 INFO (MainThread) [backoff] Backing off _do_read(...) for 1.0s (asyncio.exceptions.TimeoutError)
2023-03-10 09:12:18.424 INFO (MainThread) [backoff] Backing off _do_read(...) for 1.0s (asyncio.exceptions.TimeoutError)
2023-03-10 09:13:08.422 INFO (MainThread) [backoff] Backing off _do_read(...) for 1.0s (asyncio.exceptions.TimeoutError)
2023-03-10 09:13:58.423 INFO (MainThread) [backoff] Backing off _do_read(...) for 1.0s (asyncio.exceptions.TimeoutError)
2023-03-10 09:14:48.423 INFO (MainThread) [backoff] Backing off _do_read(...) for 1.0s (asyncio.exceptions.TimeoutError)
2023-03-10 09:15:38.426 INFO (MainThread) [backoff] Backing off _do_read(...) for 1.0s (asyncio.exceptions.TimeoutError)
2023-03-10 09:16:28.423 INFO (MainThread) [backoff] Backing off _do_read(...) for 1.0s (asyncio.exceptions.TimeoutError)
2023-03-10 09:17:18.424 INFO (MainThread) [backoff] Backing off _do_read(...) for 1.0s (asyncio.exceptions.TimeoutError)
2023-03-10 09:18:08.422 INFO (MainThread) [backoff] Backing off _do_read(...) for 1.0s (asyncio.exceptions.TimeoutError)
2023-03-10 09:18:58.423 INFO (MainThread) [backoff] Backing off _do_read(...) for 1.0s (asyncio.exceptions.TimeoutError)
2023-03-10 09:19:48.422 INFO (MainThread) [backoff] Backing off _do_read(...) for 1.0s (asyncio.exceptions.TimeoutError)
2023-03-10 09:20:38.423 INFO (MainThread) [backoff] Backing off _do_read(...) for 1.0s (asyncio.exceptions.TimeoutError)
2023-03-10 09:21:28.423 INFO (MainThread) [backoff] Backing off _do_read(...) for 1.0s (asyncio.exceptions.TimeoutError)
2023-03-10 09:22:18.425 INFO (MainThread) [backoff] Backing off _do_read(...) for 1.0s (asyncio.exceptions.TimeoutError)
2023-03-10 09:23:02.145 INFO (MainThread) [backoff] Backing off _do_read(...) for 1.0s (asyncio.exceptions.TimeoutError)
2023-03-10 09:23:12.143 ERROR (MainThread) [custom_components.huawei_solar] Timeout fetching TA2230026735_configuration_update_coordinator data

seeker1973, 2023-03-10:
> I have the same problem since I installed the 153 on the inverter. There is no other device and the connection is via lan.
> 
> 2023-03-10 07:51:51.215 ERROR (MainThread) [huawei_solar.huawei_solar] Could not read register value, has another device interrupted the connection? 2023-03-10 07:51:51.218 INFO (MainThread) [backoff] Backing off _do_read(...) for 1.0s (huawei_solar.exceptions.ConnectionInterruptedException: Could not read register value, has another device interrupted the connection?) 2023-03-10 07:51:52.219 ERROR (MainThread) [huawei_solar.huawei_solar] Modbus client is not connected to the inverter. NoneType: None 2023-03-10 07:51:52.220 INFO (MainThread) [backoff] Backing off _do_read(...) for 2.0s (huawei_solar.exceptions.ConnectionInterruptedException: Modbus client is not connected to the inverter.) 2023-03-10 07:51:54.220 ERROR (MainThread) [huawei_solar.huawei_solar] Modbus client is not connected to the inverter. NoneType: None 2023-03-10 07:51:54.221 INFO (MainThread) [backoff] Backing off _do_read(...) for 4.0s (huawei_solar.exceptions.ConnectionInterruptedException: Modbus client is not connected to the inverter.) 2023-03-10 07:51:58.223 ERROR (MainThread) [huawei_solar.huawei_solar] Modbus client is not connected to the inverter. NoneType: None 2023-03-10 07:51:58.223 INFO (MainThread) [backoff] Backing off _do_read(...) for 8.0s (huawei_solar.exceptions.ConnectionInterruptedException: Modbus client is not connected to the inverter.) 2023-03-10 07:52:16.137 WARNING (MainThread) [homeassistant.components.sensor] Setup of sensor platform huawei_solar is taking over 10 seconds. 2023-03-10 09:09:48.424 INFO (MainThread) [backoff] Backing off _do_read(...) for 1.0s (asyncio.exceptions.TimeoutError) 2023-03-10 09:09:58.422 ERROR (MainThread) [custom_components.huawei_solar] Timeout fetching TA2230026735_data_update_coordinator data 2023-03-10 09:10:38.423 INFO (MainThread) [backoff] Backing off _do_read(...) for 1.0s (asyncio.exceptions.TimeoutError) 2023-03-10 09:11:28.423 INFO (MainThread) [backoff] Backing off _do_read(...) for 1.0s (asyncio.exceptions.TimeoutError) 2023-03-10 09:12:18.424 INFO (MainThread) [backoff] Backing off _do_read(...) for 1.0s (asyncio.exceptions.TimeoutError) 2023-03-10 09:13:08.422 INFO (MainThread) [backoff] Backing off _do_read(...) for 1.0s (asyncio.exceptions.TimeoutError) 2023-03-10 09:13:58.423 INFO (MainThread) [backoff] Backing off _do_read(...) for 1.0s (asyncio.exceptions.TimeoutError) 2023-03-10 09:14:48.423 INFO (MainThread) [backoff] Backing off _do_read(...) for 1.0s (asyncio.exceptions.TimeoutError) 2023-03-10 09:15:38.426 INFO (MainThread) [backoff] Backing off _do_read(...) for 1.0s (asyncio.exceptions.TimeoutError) 2023-03-10 09:16:28.423 INFO (MainThread) [backoff] Backing off _do_read(...) for 1.0s (asyncio.exceptions.TimeoutError) 2023-03-10 09:17:18.424 INFO (MainThread) [backoff] Backing off _do_read(...) for 1.0s (asyncio.exceptions.TimeoutError) 2023-03-10 09:18:08.422 INFO (MainThread) [backoff] Backing off _do_read(...) for 1.0s (asyncio.exceptions.TimeoutError) 2023-03-10 09:18:58.423 INFO (MainThread) [backoff] Backing off _do_read(...) for 1.0s (asyncio.exceptions.TimeoutError) 2023-03-10 09:19:48.422 INFO (MainThread) [backoff] Backing off _do_read(...) for 1.0s (asyncio.exceptions.TimeoutError) 2023-03-10 09:20:38.423 INFO (MainThread) [backoff] Backing off _do_read(...) for 1.0s (asyncio.exceptions.TimeoutError) 2023-03-10 09:21:28.423 INFO (MainThread) [backoff] Backing off _do_read(...) for 1.0s (asyncio.exceptions.TimeoutError) 2023-03-10 09:22:18.425 INFO (MainThread) [backoff] Backing off _do_read(...) for 1.0s (asyncio.exceptions.TimeoutError) 2023-03-10 09:23:02.145 INFO (MainThread) [backoff] Backing off _do_read(...) for 1.0s (asyncio.exceptions.TimeoutError) 2023-03-10 09:23:12.143 ERROR (MainThread) [custom_components.huawei_solar] Timeout fetching TA2230026735_configuration_update_coordinator data

core66, 2023-03-11:
I have the same problem.

Greets

wlcrs, 2023-03-12:
I'm closing this issue as it is attracting a lot of useless 'me too' comments which are unrelated to the issue mentioned in the initial bugreport: EVCC working fine but this integration triggering TimeoutErrors. If you haven't used EVCC and are experiencing TimeoutErrors then I suspect a poor connection to your inverter is to blame for the issue.

I will reopen this issue when I'm provided with actionable information: logs of modbus communication with the inverter by both EVCC and this integration (with timestamps please!). I would expect that a modbus proxy would be a good candidate to sniff this traffic.

For anyone else: I invite you to tinker with the code of this integration and the underlying library (cfr. https://gitlab.com/Emilv2/huawei-solar/) to try to find a pattern which causes the integration to malfunction for you. Do the timeouterrors stop if you remove some registers from the update method? (cfr. https://gitlab.com/Emilv2/huawei-solar/-/blob/master/src/huawei_solar/bridge.py#L164) Only when you provide me with actionable information I will be able to identify the problem and will attempt to fix it.

closed 2023-03-12 by wlcrs

core66, 2023-03-17:
I have no EVCC, only 30KTL and DTSU

2023-03-17 05:56:50.805 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 30000 with length 35 from slave 1
2023-03-17 05:56:51.319 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 30071 with length 1 from slave 1
2023-03-17 05:56:51.700 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37200 with length 1 from slave 1
2023-03-17 05:56:52.084 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 47000 with length 1 from slave 1
2023-03-17 05:56:52.127 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 47089 with length 1 from slave 1
2023-03-17 05:56:52.350 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37100 with length 1 from slave 1
2023-03-17 05:56:52.784 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37125 with length 1 from slave 1
2023-03-17 05:56:53.034 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064 with length 52 from slave 1
2023-03-17 05:56:53.373 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32000 with length 32 from slave 1
2023-03-17 05:56:53.833 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37100 with length 38 from slave 1
2023-03-17 05:56:54.318 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064 with length 52 from slave 1
2023-03-17 05:56:54.966 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32000 with length 32 from slave 1
2023-03-17 05:56:55.805 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37100 with length 38 from slave 1
2023-03-17 05:57:06.279 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064 with length 52 from slave 1
2023-03-17 05:57:07.413 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32000 with length 32 from slave 1
2023-03-17 05:57:07.865 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37100 with length 38 from slave 1
2023-03-17 05:57:38.448 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064 with length 52 from slave 1
2023-03-17 05:57:39.034 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32000 with length 32 from slave 1
2023-03-17 05:57:39.490 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37100 with length 38 from slave 1
2023-03-17 05:58:09.448 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064 with length 52 from slave 1
2023-03-17 05:58:09.686 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32000 with length 32 from slave 1
2023-03-17 05:58:09.978 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37100 with length 38 from slave 1
2023-03-17 05:58:40.448 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064 with length 52 from slave 1
2023-03-17 05:58:40.733 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32000 with length 32 from slave 1
2023-03-17 05:58:41.189 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37100 with length 38 from slave 1
2023-03-17 05:59:11.448 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064 with length 52 from slave 1
2023-03-17 05:59:11.760 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32000 with length 32 from slave 1
2023-03-17 05:59:12.056 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37100 with length 38 from slave 1
2023-03-17 05:59:42.447 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064 with length 52 from slave 1
2023-03-17 05:59:42.819 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32000 with length 32 from slave 1
2023-03-17 05:59:43.276 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37100 with length 38 from slave 1
2023-03-17 06:00:13.448 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064 with length 52 from slave 1
2023-03-17 06:00:13.546 ERROR (MainThread) [custom_components.huawei_solar] Error fetching 6T2249046955_data_update_coordinator data: Could not update 6T2249046955 values: Got error while reading from register 32064 with length 52: Exception Response(131, 3, IllegalAddress)
2023-03-17 06:00:43.447 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064 with length 52 from slave 1
2023-03-17 06:00:44.173 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32000 with length 32 from slave 1
2023-03-17 06:00:44.762 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37100 with length 38 from slave 1
2023-03-17 06:01:15.447 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064 with length 52 from slave 1
2023-03-17 06:01:15.986 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32000 with length 32 from slave 1
2023-03-17 06:01:16.441 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37100 with length 38 from slave 1
2023-03-17 06:01:47.446 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064 with length 52 from slave 1
2023-03-17 06:01:47.719 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32000 with length 32 from slave 1
2023-03-17 06:01:48.007 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37100 with length 38 from slave 1
2023-03-17 06:02:18.448 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064 with length 52 from slave 1
[... cut ...]
2023-03-17 06:35:45.797 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32000 with length 32 from slave 1
2023-03-17 06:35:46.095 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37100 with length 38 from slave 1
2023-03-17 06:36:16.447 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064 with length 52 from slave 1
2023-03-17 06:36:16.789 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32000 with length 32 from slave 1
2023-03-17 06:36:17.245 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37100 with length 38 from slave 1

wlcrs, 2023-03-24:
I've found that EVCC implemented a delay upon connection to the Huawei Solar inverter, cfr. discussion https://github.com/evcc-io/evcc/discussions/1928 and PR https://github.com/evcc-io/evcc/pull/2109.

While this library already waits a bit before sending the first request upon initial connection, this is not the case when a reconnection happens, as this is automatically handled by pymodbus.

I've created a PR in pymodbus to make it possible to specify a delay upon (re)connect: https://github.com/pymodbus-dev/pymodbus/pull/1452