wlcrs/huawei_solar #322: Data pulled from battery connected on secondary inverter became unknown after latest update v1.2.5

norbergmats, 2023-03-14:
### Describe the issue

Data pulled from battery connected on secondary inverter became unknown after latest update v1.2.5
[image]

### Bescribe your Huawei Solar Setup

Inverter Type: Sun2000-10KTL-M1 (2 inveters in cascade)
Inverter Firmware version: V100R001C00SPC153
SDongle present: yes
Power meter present: three phase
Battery:  LUNA2000 15kWh (2 batteries, 15kwh each connected to separate inveters)
Battery Firmware version: V100R002C00SPC118

### How do you connect to the inverter?

Via the SDongle, wireless connection

### Upload your Diagnostics File

Drag & Drop your Diagnostics File here.
[config_entry-huawei_solar-d70db850c6b4361c6c20fcbbca695ef0.json (1).txt](https://github.com/wlcrs/huawei_solar/files/10970290/config_entry-huawei_solar-d70db850c6b4361c6c20fcbbca695ef0.json.1.txt)

### Upload your relevant debug logs

```text
2023-03-14 16:05:01.132 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 182, in _handle_refresh_interval
    await self._async_refresh(log_failures=True, scheduled=True)
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 330, in _async_refresh
    self.async_update_listeners()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 135, in async_update_listeners
    update_callback()
  File "/config/custom_components/huawei_solar/sensor.py", line 851, in _handle_coordinator_update
    value = self.coordinator.data[self._register_key].value
KeyError: 'power_meter_active_power'
2023-03-14 16:12:07.478 WARNING (MainThread) [homeassistant.components.sensor] Setup of sensor platform huawei_solar is taking over 10 seconds.
2023-03-14 16:12:25.737 ERROR (MainThread) [custom_components.huawei_solar] Error fetching TA2260030610_data_update_coordinator data: Could not update TA2260030610 values: Got error while reading from register 32000 with length 20: Exception Response(131, 3, SlaveFailure)
2023-03-14 16:12:54.711 ERROR (MainThread) [custom_components.huawei_solar] Unexpected exception from <bound method DataUpdateCoordinator.async_refresh of <custom_components.huawei_solar.HuaweiSolarUpdateCoordinator object at 0x7f7a72bb9f90>>
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/debounce.py", line 109, in _handle_timer_finish
    await task
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 215, in async_refresh
    await self._async_refresh(log_failures=True)
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 330, in _async_refresh
    self.async_update_listeners()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 135, in async_update_listeners
    update_callback()
  File "/config/custom_components/huawei_solar/sensor.py", line 851, in _handle_coordinator_update
    value = self.coordinator.data[self._register_key].value
KeyError: 'power_meter_active_power'
```

### Please confirm the following:

- [X] The problem is still present in the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [X] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [X] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

wlcrs, 2023-03-14:
Can you test again with the latest beta?

cfr. #313

norbergmats, 2023-03-14:
Hi, thanks alot, the beta 126b solved my problem and I can see data from all devices. 
Just one question. I I only have a power meter on inverter#1 but I got an additional power meter device now, attached to inverter#2 hsowing Unavailable on all data.

[image]

wlcrs, 2023-03-14:
You will need to remove those entities manually

closed 2023-03-14 by wlcrs