wlcrs/huawei_solar #323: [Bug]: AttributeError: 'PrivateHuaweiModbusResponse' object has no attribute 'registers'

nicx, 2023-03-15:
### Describe the issue

I try to use the integration with an USB-to-Serial Stick via MODBUS connected to my Huawei SUN2000-10KTL-M1. Because the Modbus-TCP connection via the SDongle is very unstable I want to switch over to MODBUS RTU.

When trying the modbus connection directly on my linux server it seems to work, even there are sometimes CRC errors:

```
root@nas:~/modpoll/x86_64-linux-gnu# ./modpoll -m rtu -a 1 -r 32016 -c 4 -b 9600 -d 8 -p none /dev/ttyUSB0
modpoll 3.10 - FieldTalk(tm) Modbus(R) Master Simulator
Copyright (c) 2002-2021 proconX Pty Ltd
Visit https://www.modbusdriver.com for Modbus libraries and tools.

Protocol configuration: Modbus RTU, FC3
Slave configuration...: address = 1, start reference = 32016, count = 4
Communication.........: /dev/ttyUSB0, 9600, 8, 1, none, t/o 1.00 s, poll rate 1000 ms
Data type.............: 16-bit register, output (holding) register table

-- Polling slave... (Ctrl-C to stop)
[32016]: 7031
[32017]: 3636
[32018]: 378
[32019]: 4734
-- Polling slave... (Ctrl-C to stop)
Checksum error!
-- Polling slave... (Ctrl-C to stop)
[32016]: 7031
[32017]: 3632
[32018]: 379
[32019]: 4774
-- Polling slave... (Ctrl-C to stop)
[32016]: 7031
[32017]: 3632
[32018]: 379
[32019]: 4774
-- Polling slave... (Ctrl-C to stop)
^C
root@nas:~/modpoll/x86_64-linux-gnu# 
```

Anything I could do to get the integration working, too?

### Bescribe your Huawei Solar Setup

Inverter Type: HUAWEI SUN2000-10KTL-M1
Inverter Firmware version: V100R001C00SPC153
SDongle present: yes
SDongle Firmware Version: V100R001C00SPC133
Power meter present: three phase
Battery: LUNA2000 5kWh
Battery Firmware version: V100R002C00SPC118 

### How do you connect to the inverter?

Over serial, with a serial-to-USB stick

### Upload your Diagnostics File

Drag & Drop your Diagnostics File here.

### Upload your relevant debug logs

```text
2023-03-15 08:47:18.134 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 43006
2023-03-15 08:47:18.232 ERROR (MainThread) [huawei_solar.huawei_solar] Aborting client creation due to error.
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 169, in create_rtu
    await huawei_solar._initialize()
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 90, in _initialize
    self.time_zone = (await self.get(rn.TIME_ZONE)).value
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 215, in get
    return (await self.get_multiple([name], slave))[0]
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 250, in get_multiple
    response = await self._read_registers(registers[0].register, total_length, slave)
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 328, in _read_registers
    result = await _do_read()
  File "/usr/local/lib/python3.10/site-packages/backoff/_async.py", line 151, in retry
    ret = await target(*args, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 313, in _do_read
    if len(response.registers) != length:
AttributeError: 'PrivateHuaweiModbusResponse' object has no attribute 'registers'
2023-03-15 08:47:21.031 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 43006
2023-03-15 08:47:23.199 ERROR (MainThread) [huawei_solar.huawei_solar] Aborting client creation due to error.
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 169, in create_rtu
    await huawei_solar._initialize()
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 90, in _initialize
    self.time_zone = (await self.get(rn.TIME_ZONE)).value
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 215, in get
    return (await self.get_multiple([name], slave))[0]
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 250, in get_multiple
    response = await self._read_registers(registers[0].register, total_length, slave)
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 328, in _read_registers
    result = await _do_read()
  File "/usr/local/lib/python3.10/site-packages/backoff/_async.py", line 151, in retry
    ret = await target(*args, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py", line 313, in _do_read
    if len(response.registers) != length:
AttributeError: 'PrivateHuaweiModbusResponse' object has no attribute 'registers'
```

### Please confirm the following:

- [X] The problem is still present in the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [ ] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [X] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

wlcrs, 2023-03-15:
This is with 99% certainty because a mismatch in pymodbus versions. This integration expects pyModbus 3.x.x . Do you have any other integrations using pyModbus 2.5.3 or lower? If so, then they cannot be used together.

wlcrs, 2023-03-15:
(removed earlier wrong reply)

Duplicate of #156, #135 and #117 

Please search Issues for your error the next time.

closed 2023-03-15 by wlcrs