wlcrs/huawei_solar #324: [Bug]: "Unavailable" very often

core66, 2023-03-15:
### Describe the issue

i have no connection problem -> wired
Pings steady

### Bescribe your Huawei Solar Setup

Inverter Type: 30 KTL
Inverter Firmware version: last
SDongle present: yes
Power meter present: three phase
Battery: None
Battery Firmware version:

### How do you connect to the inverter?

Via the SDongle, wired connection

### Upload your Diagnostics File

[config_entry-huawei_solar-d18c14c346f7faad116d34249dd9c89a.json.txt](https://github.com/wlcrs/huawei_solar/files/10979574/config_entry-huawei_solar-d18c14c346f7faad116d34249dd9c89a.json.txt)

### Upload your relevant debug logs

```text
Logger: custom_components.huawei_solar
Source: helpers/update_coordinator.py:182
Integration: Huawei Solar
First occurred: 13:04:21 (5 occurrences)
Last logged: 13:15:49

Timeout fetching xxxxxxxxx_data_update_coordinator data
Error fetching xxxxxxxxx_data_update_coordinator data: Could not update xxxxxxxxx values: Got error while reading from register 32000 with length 32: Exception Response(131, 3, IllegalAddress)
Error fetching xxxxxxxxx_data_update_coordinator data: Could not update xxxxxxxxx values: Got error while reading from register 32064 with length 52: Exception Response(131, 3, IllegalAddress)
```

### Please confirm the following:

- [X] The problem is still present in the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [X] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [X] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

wlcrs, 2023-03-15:
Your inverter is responding with Modbus error messages to normal read requests. The type of error (IllegalAddress) that your inverter is reporting is typically fatal and permanent. The integration will periodically retry reading, and in the meantime report the values that it was trying to read as Unavailable.

As far as I can see, the integration works as best as can be expected.

Maybe related to #248 ?

core66, 2023-03-15:
Hmm, ok, I was reading the #248 ... ok thank you mate

zsgodor, 2023-03-21:
I have same problem, sometime the inverter connection is failed, a little bit time the inverter reconnect, and work again.
My conecction is wired, via modbus. 
Logger: custom_components.huawei_solar
Source: helpers/update_coordinator.py:182
Integration: Huawei Solar
First occurred: 2023. március 17. 15:03:28 (34 occurrences)
Last logged: 12:49:30

Timeout fetching HV20C0067666_data_update_coordinator data

wlcrs, 2023-03-21:
Please use #248 for further discussion

closed 2023-03-21 by wlcrs