wlcrs/huawei_solar #326: [Bug]: Sensors cannot be added to dashboard

luotinen, 2023-03-17:
### Describe the issue

After adding the intergration, sensors appear for the device. Upon clicking the "Add to dashboard" button the following error is produced: "You don't seem to be in control of any dashboard, please take control first."

### Bescribe your Huawei Solar Setup

Inverter Type: SUN2000-10KTL-M1
Inverter Firmware version: V100R001C00SPC150
SDongle present: yes
Power meter present: three phase
Battery:  No
Battery Firmware version: N/A

### How do you connect to the inverter?

Via the SDongle, wireless connection

### Upload your Diagnostics File

{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.3.5",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.10",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Europe/Helsinki",
    "os_name": "Linux",
    "os_version": "5.15.84-v8",
    "supervisor": "2023.03.1",
    "host_os": "Home Assistant OS 9.5",
    "docker_version": "20.10.22",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {
    "huawei_solar": {
      "version": "1.2.4",
      "requirements": [
        "huawei-solar==2.2.4"
      ]
    },
    "pax_ble": {
      "version": "1.0.0",
      "requirements": []
    },
    "hacs": {
      "version": "1.30.1",
      "requirements": [
        "aiogithubapi>=22.10.1"
      ]
    },
    "daikin_residential": {
      "version": "2.3.0",
      "requirements": [
        "oic==1.4.0"
      ]
    },
    "nordpool": {
      "version": "0.0.11",
      "requirements": [
        "nordpool>=0.2"
      ]
    }
  },
  "integration_manifest": {
    "domain": "huawei_solar",
[... 280 lines cut ...]
2023-03-17 10:15:16.450 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32000
2023-03-17 10:15:47.193 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064
2023-03-17 10:15:47.520 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32000
2023-03-17 10:16:17.193 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064
2023-03-17 10:16:17.710 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32000
2023-03-17 10:16:48.193 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064
2023-03-17 10:16:49.069 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32000
2023-03-17 10:17:19.193 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064
2023-03-17 10:17:19.410 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32000
2023-03-17 10:17:50.194 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064
2023-03-17 10:17:50.649 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32000
2023-03-17 10:18:20.193 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064
2023-03-17 10:18:20.573 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32000
2023-03-17 10:18:51.194 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064
2023-03-17 10:18:51.426 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32000
2023-03-17 10:19:21.194 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064
2023-03-17 10:19:21.430 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32000
```

### Please confirm the following:

- [X] The problem is still present in the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [X] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [X] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

wlcrs, 2023-03-17:
That error is unrelated to this integration. Check the account permissions in your HA.

closed 2023-03-17 by wlcrs

luotinen, 2023-03-17:
This is my only integration which is unable to add sensors to UI controlled dashboard. I interpret that this is the intended behaviour.

If someone else is trying to add sensors, you need to go to the dashboard you want to take control, click on the three dots on top right and take control. After this, the UI will no longer keep the dashboard up-to-date and you will have to manually add the items you want to see.