wlcrs/huawei_solar #332: [Bug]: Modbus client is not connected to the inverter

johanvh83, 2023-03-19:
### Describe the issue

I checked the discussion but nothing could help me and it seems nobody is very responsive on the discussions.
The integrations worked for months, almost 2 years, without problem. 
Since 2 days, I don't have any data anymore. Tried the almost I could but can't succeed.
When I ping the inverter, I get a reply and on FusionSolar all data is available.
I was a bit desperate and finally removed and re-installed the integration but after the popup with the IP address/port/ID it gives: 'Failed to connect' so I'm not able to check logs.
The inverter gets Wifi access from my router. In the past I connected by the local IP 192.168.100.117 to the inverter.
This option is not available from the drop-down 'how do you connect'. Is the setup changed?

The HA logs:

Logger: huawei_solar.huawei_solar
Source: /usr/local/lib/python3.10/site-packages/huawei_solar/huawei_solar.py:293
First occurred: 7:57:50 PM (12 occurrences)
Last logged: 8:11:00 PM

Modbus client is not connected to the inverter.
NoneType: None

and

This error originated from a custom integration.

Logger: backoff
Source: custom_components/huawei_solar/config_flow.py:114
Integration: Huawei Solar
First occurred: 7:58:21 PM (2 occurrences)
Last logged: 8:11:00 PM

Giving up _do_read(...) after 6 tries (huawei_solar.exceptions.ConnectionInterruptedException: Modbus client is not connected to the inverter.)

### Bescribe your Huawei Solar Setup

Inverter Type: SUN2000-4.6KTL-L1
Inverter Firmware version: V200R001C00SPC130
SDongle present: no
Power meter present: single phase
Battery: LUNA2000 10kWh
Battery Firmware version: V100R002C00SPC106

### How do you connect to the inverter?

Via the `SUN2000-<serial_no>` WiFi

### Upload your Diagnostics File

Drag & Drop your Diagnostics File here.

### Upload your relevant debug logs

```text
N/A
```

### Please confirm the following:

- [ ] The problem is still present in the latest release of this integration.
- [ ] I did not find an existing issue describing this problem.
- [ ] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [ ] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

tremebundo, 2023-03-20:
I had the same problem a few days ago and it worked again reinstaling this pakage.
Anyway, I connect to the installer access point which I gess is not your case.

johanvh83, 2023-03-20:
Hi @tremebundo, which package you installed? The package is intend for Home Assistant or for the inverter?

MariuszWoszczynski, 2023-03-21:
I also have a problem with connecting to inverter last days

johanvh83, 2023-03-21:
@tarmor1 & @tremebundo, how guys do you connect to the inverter, by Wifi? Do you have the Dongle?
According my information Huawei changed 'something' remotely and disabled Modbus by Wifi.
Seems there are 2 options now: Buy the Dongle to enable this feature again or use the serial output on the communication port. The last option required additional hardware. See https://github.com/wlcrs/huawei_solar/wiki/Connecting-to-the-inverter

diegomontanaduarte, 2023-03-21:
@johanvh83 even with new inverter firmware is still be possible to connect by using a WiFi USB stick on your Home Assistant server. In my case when I restarted Home Assistant works fine for some minutes and then disconnected from the inverter

tremebundo, 2023-03-21:
Yes, I am connected right now to the wifi for the installer in the inverter. The one with fixed IP 192.168.1.200 and I get the data. I do not have any dongle or serial connection. And it has the latest firmware revision.
@diegomontanaduarte , when this problem appeared, I had to uninstall this component, install again and configure to get it work again. I had the same symptoms as you. But since then, the sensor manes have changed.