wlcrs/huawei_solar #335: [Bug]: Randomly error on reading sensors from Inverter and Smart Meter

colyro, 2023-03-22:
### Describe the issue

Hi

There are intermitent error in HA with the error messages that the Inverter and Smart Meter values are not shown anymore and cannot be read. 
It is happening randomly. i've tried Dongle firmware SPC127/130/133 but same issue.
Connection tried via WiFI and LAN cable but same results.
Tried the connection via an openwrt router connected as client directly to the inverter (SUN2000-SN) AP (192.168.200.1) but still no success.
During the blackout period, the sdongle IP is pingable and the 502 port is open - checked via telnet command.
I am running latest HA and Huawei Solar. 

### Bescribe your Huawei Solar Setup

Inverter Type:  SUN2000-15KTL-M2
Inverter Firmware version: SUN2000MAV100R001C00SPC153
SDongle present: yes
Power meter present: three phase
Battery: None
Battery Firmware version: NA

### How do you connect to the inverter?

Via the SDongle, wired connection

### Upload your Diagnostics File

Drag & Drop your Diagnostics File here.
[home-assistant_2023-03-22T06-16-58.666Z.log](https://github.com/wlcrs/huawei_solar/files/11037537/home-assistant_2023-03-22T06-16-58.666Z.log)
[home-assistant_2023-03-22T06-16-58.666Z.log](https://github.com/wlcrs/huawei_solar/files/11037552/home-assistant_2023-03-22T06-16-58.666Z.log)
[config_entry-huawei_solar-ab093ea8bd7dac6fd65a63920a589a3c.json.txt](https://github.com/wlcrs/huawei_solar/files/11037590/config_entry-huawei_solar-ab093ea8bd7dac6fd65a63920a589a3c.json.txt)

### Upload your relevant debug logs

```text
all full logs uploaded.
```

### Please confirm the following:

- [X] The problem is still present in the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [X] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [X] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

wlcrs, 2023-03-22:
There is no obvious bug on the side of the integration, but your inverter does not always answer modbus requests in a timely fashion (which is something else than accepting a TCP connection to the modbus port).

Can you please test if the following fix improves the number of errors:

https://github.com/wlcrs/huawei_solar/issues/248#issuecomment-1474034533

colyro, 2023-03-22:
i've implemented the aformentioned code into __init__.py and restarted the HA.
I'll monitor for the next 1-2 days as it will be more than conclusive as i am getting these hicups 2-3/hour

thanks

colyro, 2023-03-22:
happened again with above settings at 13:23 and 14:05.
[logs.zip](https://github.com/wlcrs/huawei_solar/files/11039778/logs.zip)

dont know what actions to perform further
maybe its not a bug but rather a sdongle modbus implementation issue?

core66, 2023-03-24:
Many Users got that problem with "Timeout on data_update_coordinator". i tested 1.2.6b2. Nothing fixed so far at my side.
WLCRS you always tell us that nothing is wrong with your code...i cant believe it anymore.

pls believe that it is a real problem and something is wrong with the code in some way.

core66, 2023-03-24:
ok you define my post as "abuse"...lol but ok.

i want to say, look, when I use for example BOPV for reading the values there are absolutely NO DROPS.
When i use your HA Add-on, there are drops...so sorry, but that's a fact you cannot deny.

I hope you understand what i wanna say.

wlcrs, 2023-03-24:
@core66 your comments are adding nothing to the discussion. Try be helpful to figure out what is causing the issue, or even better: solve it yourself and create a PR. 

So stop littering issues with your complaints, this is your final warning.

colyro, 2023-03-25:
@core66 - no offence but i am strongly against non constructive feedback/blaiming for somebody's work which is absolutely for free. no matter how good or bad it is that one. i fully understand your frustration but still, please lets respect others (in general) free work or lets start doing this work by ourselves.

Related to the initial topic - i think i found the issue. After your confirmation that it is more likely an issue with the connection to the modbus of SDongle, i started to look in this direction (after did a lot of troubleshooting in firewalls/IPS/IDS /VPN Site2Site etc).
I have downgraded from SDognle firmware verc SPC 133 gradually to SPC127 which is far away more stable. 
Maybe, there is only max 1 dropping connection per day instead of ~10.

So as a conclusion, for those who have this issue and they are connecting to SDongle Modbus via Wifi or LAN, try to use firmware SPC127.

closed 2023-03-25 by wlcrs

core66, 2023-03-27:
@colyro Thank you for your comment but your workaround with downpatching to 127 is not really helpful in my case.
I have ~20 timeouts an hour before and after, and no, i have no connection problem.

@wlcrs THIS WAS A CONSTRUCTIVE COMMENT TO A CONSTRUCTIVE PROPOSED SOLUTION.

core66, 2023-03-29:
The only thing i can add for constructive cooperation is:

i tried to connect directly to the inverter and this works flawless. No timeouts anymore.
So anything with the dongle communication is broken in my opinion.