wlcrs/huawei_solar #347: [Bug]: lost connection to all the optimizers - 1.2.6 beta 4

etamtlosz, 2023-04-03:
### Describe the issue

As of the 28th of march I'm not getting data from my optimizers. It seems it started after I upgraded to b4.

There were similar issues reported in the past, that's why I did not tick `I did not find an existing issue describing this problem.`, but I think this one is something new, has nothing to do with the previous issues.

### Bescribe your Huawei Solar Setup

Inverter Type: SUN2000-5KTL-L1
Inverter Firmware version: V200R001C00SPC130
SDongle present: yes
Power meter present: single phase
Battery: LUNA2000 10kWh
Battery Firmware version: V100R002C00SPC116

### How do you connect to the inverter?

Via the SDongle, wired connection

### Upload your Diagnostics File

Drag & Drop your Diagnostics File here.
[config_entry-huawei_solar-4c0d9bad6cf39b762c7431d14686cc4a.json.txt](https://github.com/wlcrs/huawei_solar/files/11134795/config_entry-huawei_solar-4c0d9bad6cf39b762c7431d14686cc4a.json.txt)

### Upload your relevant debug logs

```text
-04-03 16:32:12.531 INFO (MainThread) [pymodbus.logging] Protocol lost connection.
2023-04-03 16:32:12.531 DEBUG (MainThread) [pymodbus.logging] Client disconnected from modbus server: trying to send
2023-04-03 16:32:12.531 DEBUG (MainThread) [pymodbus.logging] Getting transaction 25
2023-04-03 16:32:12.531 ERROR (MainThread) [backoff] Giving up _perform_request(...) after 6 tries (asyncio.exceptions.TimeoutError)
2023-04-03 16:32:12.535 INFO (MainThread) [pymodbus.logging] Protocol lost connection.
2023-04-03 16:32:12.535 WARNING (MainThread) [pymodbus.logging] Ignoring launch of delayed reconnection, another is in progress
2023-04-03 16:32:12.536 DEBUG (MainThread) [pymodbus.logging] Client disconnected from modbus server: None
2023-04-03 16:32:12.536 DEBUG (MainThread) [pymodbus.logging] Waiting 1000 ms before next connection attempt.
2023-04-03 16:32:12.551 INFO (MainThread) [huawei_solar.huawei_solar] Waiting for connection ...
2023-04-03 16:32:13.537 DEBUG (MainThread) [pymodbus.logging] Connecting.
2023-04-03 16:32:13.538 DEBUG (MainThread) [pymodbus.logging] Client connected to modbus server
2023-04-03 16:32:13.539 INFO (MainThread) [pymodbus.logging] Protocol made connection.
2023-04-03 16:32:13.539 DEBUG (MainThread) [huawei_solar.modbus] Waiting for 1500 milliseconds after connection before performing operations
2023-04-03 16:32:13.539 INFO (MainThread) [pymodbus.logging] Connected to 192.168.2.170:502.
```

### Please confirm the following:

- [X] The problem is still present in the latest release of this integration.
- [ ] I did not find an existing issue describing this problem.
- [X] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [X] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

wlcrs, 2023-04-03:
Please enable debug logging for at least an hour and upload the resulting file. https://www.home-assistant.io/docs/configuration/troubleshooting/#debug-logs-and-diagnostics

The provided logs are too limited to diagnose the issue.

etamtlosz, 2023-04-03:
Here you go, the last two hours.
[home-assistant_huawei_solar_2023-04-03T06-14-11.395Z.log](https://github.com/wlcrs/huawei_solar/files/11135265/home-assistant_huawei_solar_2023-04-03T06-14-11.395Z.log)

wlcrs, 2023-04-03:
After inspecting the logs, I see that the integration tries to fetch the Optimizer system information 6 times, and each time it loses it's connection to the inverter upon doing this:

```
2023-04-03 16:30:22.505 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading file 0x45 from slave 1
2023-04-03 16:30:22.506 DEBUG (MainThread) [pymodbus.logging] send: 0x0 0x14 0x0 0x0 0x0 0x5 0x0 0x41 0x5 0x1 0x45
2023-04-03 16:30:22.506 DEBUG (MainThread) [pymodbus.logging] Adding transaction 20
2023-04-03 16:30:32.508 INFO (MainThread) [pymodbus.logging] Protocol lost connection.
2023-04-03 16:30:32.508 DEBUG (MainThread) [pymodbus.logging] Client disconnected from modbus server: trying to send
2023-04-03 16:30:32.509 DEBUG (MainThread) [pymodbus.logging] Getting transaction 20
2023-04-03 16:30:32.510 INFO (MainThread) [backoff] Backing off _perform_request(...) for 10.0s (asyncio.exceptions.TimeoutError)
2023-04-03 16:30:32.510 DEBUG (MainThread) [huawei_solar.huawei_solar] Received <class 'asyncio.exceptions.TimeoutError'>: backing off file upload for 10.0 seconds after 1 tries
2023-04-03 16:30:32.510 INFO (MainThread) [pymodbus.logging] Protocol lost connection.
2023-04-03 16:30:32.510 WARNING (MainThread) [pymodbus.logging] Ignoring launch of delayed reconnection, another is in progress
2023-04-03 16:30:32.514 DEBUG (MainThread) [pymodbus.logging] Client disconnected from modbus server: None
2023-04-03 16:30:32.515 DEBUG (MainThread) [pymodbus.logging] Waiting 1000 ms before next connection attempt.
2023-04-03 16:30:33.516 DEBUG (MainThread) [pymodbus.logging] Connecting.
2023-04-03 16:30:33.518 DEBUG (MainThread) [pymodbus.logging] Client connected to modbus server
2023-04-03 16:30:33.518 INFO (MainThread) [pymodbus.logging] Protocol made connection.
2023-04-03 16:30:33.518 DEBUG (MainThread) [huawei_solar.modbus] Waiting for 1500 milliseconds after connection before performing operations
2023-04-03 16:30:33.519 INFO (MainThread) [pymodbus.logging] Connected to 192.168.2.170:502.
2023-04-03 16:30:42.511 DEBUG (MainThread) [pymodbus.logging] send: 0x0 0x15 0x0 0x0 0x0 0x5 0x0 0x41 0x5 0x1 0x45
2023-04-03 16:30:42.512 DEBUG (MainThread) [pymodbus.logging] Adding transaction 21
2023-04-03 16:30:52.514 INFO (MainThread) [pymodbus.logging] Protocol lost connection.
2023-04-03 16:30:52.514 DEBUG (MainThread) [pymodbus.logging] Client disconnected from modbus server: trying to send
2023-04-03 16:30:52.514 DEBUG (MainThread) [pymodbus.logging] Getting transaction 21
2023-04-03 16:30:52.514 INFO (MainThread) [backoff] Backing off _perform_request(...) for 10.0s (asyncio.exceptions.TimeoutError)
2023-04-03 16:30:52.514 DEBUG (MainThread) [huawei_solar.huawei_solar] Received <class 'asyncio.exceptions.TimeoutError'>: backing off file upload for 10.0 seconds after 2 tries
2023-04-03 16:30:52.514 INFO (MainThread) [pymodbus.logging] Protocol lost connection.
2023-04-03 16:30:52.515 WARNING (MainThread) [pymodbus.logging] Ignoring launch of delayed reconnection, another is in progress
2023-04-03 16:30:52.516 DEBUG (MainThread) [pymodbus.logging] Client disconnected from modbus server: None
2023-04-03 16:30:52.516 DEBUG (MainThread) [pymodbus.logging] Waiting 1000 ms before next connection attempt.
2023-04-03 16:30:53.516 DEBUG (MainThread) [pymodbus.logging] Connecting.
2023-04-03 16:30:53.518 DEBUG (MainThread) [pymodbus.logging] Client connected to modbus server
2023-04-03 16:30:53.518 INFO (MainThread) [pymodbus.logging] Protocol made connection.
2023-04-03 16:30:53.518 DEBUG (MainThread) [huawei_solar.modbus] Waiting for 1500 milliseconds after connection before performing operations
2023-04-03 16:30:53.518 INFO (MainThread) [pymodbus.logging] Connected to 192.168.2.170:502.
2023-04-03 16:30:54.828 WARNING (MainThread) [homeassistant.components.zha.core.channels.base] [0x85FF:1:0x0006]: async_initialize: all attempts have failed: [DeliveryError('Failed to deliver packet: <TXStatus.NWK_ROUTE_DISCOVERY_FAILED: 208>'), DeliveryError('Failed to deliver packet: <TXStatus.NWK_ROUTE_DISCOVERY_FAILED: 208>'), DeliveryError('Failed to deliver packet: <TXStatus.NWK_ROUTE_DISCOVERY_FAILED: 208>'), DeliveryError('Failed to deliver packet: <TXStatus.NWK_ROUTE_DISCOVERY_FAILED: 208>')]
2023-04-03 16:30:57.007 WARNING (MainThread) [homeassistant.bootstrap] Waiting on integrations to complete setup: huawei_solar
2023-04-03 16:31:02.515 DEBUG (MainThread) [pymodbus.logging] send: 0x0 0x16 0x0 0x0 0x0 0x5 0x0 0x41 0x5 0x1 0x45
2023-04-03 16:31:02.515 DEBUG (MainThread) [pymodbus.logging] Adding transaction 22
2023-04-03 16:31:02.995 WARNING (MainThread) [custom_components.localtuya.common] [128...a76] Failed to connect to 192.168.2.211: [Errno 113] Connect call failed ('192.168.2.211', 6668)
2023-04-03 16:31:12.517 INFO (MainThread) [pymodbus.logging] Protocol lost connection.
2023-04-03 16:31:12.517 DEBUG (MainThread) [pymodbus.logging] Client disconnected from modbus server: trying to send
2023-04-03 16:31:12.518 DEBUG (MainThread) [pymodbus.logging] Getting transaction 22
2023-04-03 16:31:12.518 INFO (MainThread) [backoff] Backing off _perform_request(...) for 10.0s (asyncio.exceptions.TimeoutError)
2023-04-03 16:31:12.518 DEBUG (MainThread) [huawei_solar.huawei_solar] Received <class 'asyncio.exceptions.TimeoutError'>: backing off file upload for 10.0 seconds after 3 tries
2023-04-03 16:31:12.519 INFO (MainThread) [pymodbus.logging] Protocol lost connection.
[... cut ...]
2023-04-03 16:32:12.531 DEBUG (MainThread) [pymodbus.logging] Client disconnected from modbus server: trying to send
2023-04-03 16:32:12.531 DEBUG (MainThread) [pymodbus.logging] Getting transaction 25
2023-04-03 16:32:12.531 ERROR (MainThread) [backoff] Giving up _perform_request(...) after 6 tries (asyncio.exceptions.TimeoutError)
2023-04-03 16:32:12.535 INFO (MainThread) [pymodbus.logging] Protocol lost connection.
```

Can you please do a downgrade to an older version to see if the behavior is different then? Please upload a diagnostics file + debug logs while running an older version as well.

etamtlosz, 2023-04-04:
With b2 the optimizers are back.
log and diag attached
[home-assistant_huawei_solar_2023-04-04T01-12-59.016Z.log](https://github.com/wlcrs/huawei_solar/files/11143720/home-assistant_huawei_solar_2023-04-04T01-12-59.016Z.log)
[config_entry-huawei_solar-4c0d9bad6cf39b762c7431d14686cc4a.json_04_04_23_1.txt](https://github.com/wlcrs/huawei_solar/files/11143722/config_entry-huawei_solar-4c0d9bad6cf39b762c7431d14686cc4a.json_04_04_23_1.txt)

wlcrs, 2023-04-05:
Not an easy problem to diagnose to be honest. In 1.2.6b2 the pymodbus library version used had an unexpected change limiting the amount of logging to happen. I contributed a fix to pymodbus to solve this, which took effect in 1.2.6b4.

Could you please add the following lines to `__init__.py` of this integration, restart your HA and capture some new logs? These should be more complete so that I can compare them with your logs of 1.2.6b4 .

```
from pymodbus import pymodbus_apply_logging_config
from pymodbus.logging import Log

pymodbus_apply_logging_config(Log.DEBUG)
```

etamtlosz, 2023-04-06:
A few hours of logging with b2 and the changed `__init__.py`
[config_entry-huawei_solar-4c0d9bad6cf39b762c7431d14686cc4a.json.txt](https://github.com/wlcrs/huawei_solar/files/11165229/config_entry-huawei_solar-4c0d9bad6cf39b762c7431d14686cc4a.json.txt)
[home-assistant_huawei_solar_2023-04-06T00-21-24.387Z.log](https://github.com/wlcrs/huawei_solar/files/11165230/home-assistant_huawei_solar_2023-04-06T00-21-24.387Z.log)

wlcrs, 2023-04-06:
Thanks a lot! With these logs it was only a case of "spot the difference". Apparently pyModbus changed it's API subtly which caused the slave-id to not be passed on for certain Modbus requests. They did document the API change, but put it under the wrong version: one that is not release yet 🤦‍♂️

Can you test again with 1.2.6b5?

etamtlosz, 2023-04-06:
Just in time before sunset.
b5 installed, optimizers online and recognized by the integration.
I'm going to keep an eye on it, just in case something else pops up.
I see that `__init__.py` got overwritten and my change is gone.

Thanks for the fast reaction and solution.

closed 2023-04-06 by etamtlosz