wlcrs/huawei_solar #349: [Feature Request]: Inverter temperature

nicx, 2023-04-05:
### Describe your feature request

Hey, as far as I can see the inverter and the battery temperature is not monitored via the integration. It would be great, if that sensor "internal temperature" of both devices (inverter & battery) would be monitored, too. Why? The inverter and the battery are in my boiler room, and i want to monitor the various temperatures closely to be able to take cooling countermeasures early. Automation in home assistant would be very helpful here. Today i still have to use the huawei app for this.

### Proper usage

- [X] I confirm that this is not a bug report
- [X] I confirm that this feature request is within the stated scope of the integration

wlcrs, 2023-04-05:
I have to strike a balance between the number of registers that is read by this integration and the overall performance. 

These temperature registers cannot be bundled together in the requests that I already do to the inverter. As other users are already facing performance and reliability problems due to the large amount of registers being read, I'm thus not inclined to add these to the integration.

You are free to extend the integration yourself using these instructions: https://github.com/wlcrs/huawei_solar/wiki/Defining-custom-sensors

nicx, 2023-04-05:
@wlcrs why not just add more seniors directly, but as "disabled" sensors. You everybody could enable only the sensors really needed. Define a basic set of sensors which are enabled b default when installing the integration, all other sensors should be enabled manually.

This could help even for all other issues with "performance", too, when reducing the basic set of enabled sensors in default.

wlcrs, 2023-04-05:
The amount of effort to make this more dynamic is not in proportion to the expected benefit. The `DataUpdateCoordinator` interface of HA does not expose the necessary information about which entities are enabled/disabled by itself, so it would mean a lot of extra effort, code and thus potential bugs and support load.

You are however very welcome to add this functionality yourself: create the necessary PR's on this integration and the underlying library and I will gladly review them.

If the `DataUpdateCoordinator` interface were to evolve in a sense that they better expose which entities are enabled/disabled, then I'm willing to revisit this myself.

jires, 2023-05-09:
> The amount of effort to make this more dynamic is not in proportion to the expected benefit. The `DataUpdateCoordinator` interface of HA does not expose the necessary information about which [...]

Hello @wlcrs 

Do I understand this correct? 
Integration will always pool/request all known/available standard registers from inverter and it doesn't matter if some of respectively entities are disabled in HA or not.    

If so, maybe someone can write a short wiki, how to modify the code, to remove some standard registers from reading, fx. in section [Defining custom sensors](https://github.com/wlcrs/huawei_solar/wiki/Defining-custom-sensors), so people can remove those registers which is not in use. I will personally sacrifice some registers, which I not use, for more stable connection and less traffic.

WiebKastanje, 2023-07-10:
Hi  @wlcrs  or any other capable person.

As I'm not a software engineer (although I manage yaml programming) I would appreciate to read instructions how to remove/add Huawei registers within this excellent HA integration. 

I really want to read/write some registers currently not available. I understand fully that the implemented large number of registers is the maximum the inverter can possibly handle. Also understand that DataUpdateCoordinator has described limitations on selecting/deselecting polled registers.

I know that changing code is my own responsibility and without any support/issue handling.
Nevertheless, I would like to try with instructions of a knowledgeable person.

Would appreciate any help on this subject.
Thanks a lot!