wlcrs/huawei_solar #352: [Feature Request]: Modbus Proxy

broodro0ster, 2023-04-14:
### Describe your feature request

This integration is working very reliable for me and I like it a lot. Thanks a lot for you work!
I've come across some other automation software for specific taks that aren't possible in HA (yet) and it also needs to access my Huawei inverter. I found only 1 device can make a connection and I would like that HA stays the "master". Would it be possible for you to add a modbus proxy to integration so other devices/application can share the connection of your integration?

### Proper usage

- [X] I confirm that this is not a bug report
- [X] I confirm that this feature request is within the stated scope of the integration

wlcrs, 2023-04-14:
You can use a general purpose modbus proxy like https://github.com/Akulatraxas/ha-modbusproxy  . Put it in between this integration and the inverter and then you can also use other applications that connect to that proxy.

wlcrs, 2023-04-14:
By editing the config_enties file in the hidden .storage folder.

Deleting and re-adding will also work, but can cause some entities to be renamed due to alignment with the HA entity naming scheme that came into effect last year.

closed 2023-04-14 by wlcrs

broodro0ster, 2023-09-07:
I'm using the modbus proxy for a while now (https://github.com/Akulatraxas/ha-modbusproxy), but when Home Assistant is starting after an update, the Huawei integration fails to load. I'm guessing it's because the modbus proxy is not up yet, because when I notice it and I click reload integration, it starts fine.
Is there a way to make your integration retry a few times before it gives up?