wlcrs/huawei_solar #353: [Feature Request]: Add funktion Grid Power to zero by phase

ivoryrea, 2023-04-18:
### Describe your feature request

Writing in the right Modbus Address, to control the maximum power injected by the inverter in the Grid network.

You already integrated limitation where is possible to set limits by Watts or prosentage. Is it possible to add functionality where is possible to set zero power limit based on phase?

[image]

Reason why I ask this funktionality is, that electrisity company want to get fee if i selling energy back to net. Sometimes energy price is smaller than fee what I need to pay. 

### Proper usage

- [X] I confirm that this is not a bug report
- [X] I confirm that this feature request is within the stated scope of the integration

wlcrs, 2023-04-18:
Yes. Alas, when looking in the [modbus registers pdf](https://github.com/wlcrs/huawei_solar/files/10920652/Solar.Inverter.Modbus.Interface.Definitions.v05.pdf), I do not find anything that allows me to set the limitation mode.

I cannot implement something which is not documented in that PDF I'm afraid.

ivoryrea, 2023-04-18:
Now I tested again, if I set limitation mode to single-phase power then this stay when I switch mode to no limit and back to limited power. 
Services what before created is enough for me.
huawei_solar.reset_maximum_feed_grid_power
huawei_solar.set_maximum_feed_grid_power

Sorry for bothering you.
You have done good integration.

closed 2023-04-18 by ivoryrea

osos, 2023-05-28:
I cannot seem to find the option in the Home-Assistant integration to limit the export.

Would like to set up an automation to limit export, dependent on the selling price. Sometimes is costs money to export to grid.