wlcrs/huawei_solar #365: [Feature Request]:

NiFu90, 2023-05-05:
### Describe your feature request

First of all thank you very much for this integration and the work you have done! Create job!!

I want to use some of the entities from the solar plant inside the HA energy tab, to see the statistics and flows (I know there are also visualisation cards for it, but I would like to have it in the native energy tab).
For some reason the entities that I would like to use are not selectable, as for example the inverter_input_power:
[image]

The two entities present in the screen shot are the only ones to be selected that include the name inverter.

Is it possible to adapt the sensor settings such that the physically appropriate ones can be used for the different inputs towards the energy tab?

Thank you.

### Proper usage

- [X] I confirm that this is not a bug report
- [X] I confirm that this feature request is within the stated scope of the integration

wlcrs, 2023-05-05:
That is not how the HA energy tab works

closed 2023-05-05 by wlcrs

JacoboSanchez, 2023-05-05:
If you want to know why you must understand the difference between energy (KWh) and Power (KW). Energy tab tracks energy so only energy sensors can be used. Most of the data provided by the inverter is instant power and that is why you can not use them in the energy tab.

As for you questions, from the [HA documentation](https://www.home-assistant.io/docs/energy/):
`If you have a sensor that returns instantaneous power readings (W or kW), then to add a sensor that returns energy usage or generation (kWh) refer to [Riemann sum integral integration](https://www.home-assistant.io/integrations/integration/#energy)`

But probably you don't want that, what you want is to set up a card to show the power data and the energy tab wont give you that

NiFu90, 2023-05-05:
Thanks @JacoboSanchez for the explanation. I didn't know that is was supposed to show only the energy and not also the current power readings because I always saw these visualisation cards with the power distribution and I thought why doesn't it work with the native energy tab, but well, that's obviously not intended.
Thank you!