wlcrs/huawei_solar #384: [Bug]: Elevate permisisons does not do anything

radudorurus, 2023-05-31:
### Describe the issue

Everything is fine (connection, data)
When adding the integration with elevate permissions checked, the only "setting" I get is Power On/OFF but I do not get any prompt for the installer password.
Is that how it should work?

If the Integration Huawei is added without being connected with the app Fusion Solar, the integration sees 2 devices 25 entities (the power meter has no entity).

When I connect to the Inverter (with the app) by "device commissioning" method (and enter the Installer password), I get 2 devices and 43 entities (10 are from the power meter)

Without checking "elevate permissions" I get only 1 device/6 entities.
After I log in to device commissioning, I get 2 devices/42 entities (power on/off is missing)

### Bescribe your Huawei Solar Setup

Inverter Type: SUN2000-5KTL-L1
Inverter Firmware version: V200R001C00SPC130
SDongle present: yes
Power meter present: single phase 
Battery: None
Battery Firmware version: n/a

### How do you connect to the inverter?

Via the SDongle, wired connection

### Upload your Diagnostics File

Drag & Drop your Diagnostics File here.
[config_entry-huawei_solar-0c086f1083fea94a6f473d583afd7d1b.json.txt](https://github.com/wlcrs/huawei_solar/files/11617631/config_entry-huawei_solar-0c086f1083fea94a6f473d583afd7d1b.json.txt)

### Upload your relevant debug logs

```text
Not sure if the debuglog helps with something since the issue relates to the adding phase of the integration.
If necessary I can upload upon request.
```

### Please confirm the following:

- [X] The problem is still present in the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [X] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [ ] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

radudorurus, 2023-06-02:
My mistake. Sorry.
I just observed after some time the power limitation are called by service in the Developer section of the HA.

jayjayseal, 2023-07-18:
I can confirm this. Give it some time. It loads 1 entity at first but then starts to add more (33 in my case)

SUN2000-6KTL-L1 | V200R001C00SPC130

Also give it a couple of tries when adding the device in the admin panel (unknown error). Make sure you have disconnected from the Wlan interface of the inverter and close down the fusion solar and Sun 2000 application (i had read in the docs that only 1 device can access at a time. It can take a while before it notices the connection has closed from your mobile phone.

The wiki states:
`You need to enable "Local O&M" in "Settings" -> "Router connection settings" in the "Device Commissioning" part of the FusionSolar App. This needs to be done using an Installer account. (Note that this toggle is somewhat buggy, and possibly needs multiple tries to effectively enable it. Make sure to use the latest version of the app). This will open port 6607.`

Maybe there was a change but you will have to reinitialize the WLANconnection to your home router again and then toggle the switch Local O&M to ON. Check the network SSID again, Enter your personal WLAN password again and flick the O&M switch. It will work then!.

closed 2023-07-19 by wlcrs