wlcrs/huawei_solar #386: [Bug]: name 'bridge' is not defined

harnyd, 2023-06-04:
### Describe the issue

If I want to turn on or off the inverter (switch.inverter_inverter_on_off) I see following message in logs which is causing malfunction of my NodeRed setup. From HA the switch is working fine. 
_name 'bridge' is not defined_

### Bescribe your Huawei Solar Setup

Inverter Type: SUN2000-5KTL-M1
Inverter Firmware version: V100R001C00SPC153
SDongle present: V100R001C00SPC133
Power meter present: three phase 
Battery: LUNA2000 xxkWh 
Battery Firmware version:

### How do you connect to the inverter?

Over serial, with a serial-to-USB stick

### Upload your Diagnostics File

Drag & Drop your Diagnostics File here.
[config_entry-huawei_solar.json.txt](https://github.com/wlcrs/huawei_solar/files/11645848/config_entry-huawei_solar.json.txt)

### Upload your relevant debug logs

```text
Logger: homeassistant.components.websocket_api.http.connection
Source: custom_components/huawei_solar/switch.py:279
Integration: Home Assistant WebSocket API (documentation, issues)
First occurred: 15:59:29 (2 occurrences)
Last logged: 16:14:03

[2760810016] name 'bridge' is not defined
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 199, in handle_call_service
    await hass.services.async_call(
  File "/usr/src/homeassistant/homeassistant/core.py", line 1849, in async_call
    task.result()
  File "/usr/src/homeassistant/homeassistant/core.py", line 1889, in _execute_service
    await cast(Callable[[ServiceCall], Awaitable[None]], handler.job.target)(
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 226, in handle_service
    await service.entity_service_call(
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 798, in entity_service_call
    future.result()  # pop exception if have
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 980, in async_request_call
    await coro
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 838, in _handle_entity_call
    await result
  File "/config/custom_components/huawei_solar/switch.py", line 279, in async_turn_off
    device_status = (await self.bridge.client.get(rn.DEVICE_STATUS, bridge.slave_id)).value
NameError: name 'bridge' is not defined
--------------------------------------------
Logger: homeassistant.components.websocket_api.http.connection
Source: custom_components/huawei_solar/switch.py:261
Integration: Home Assistant WebSocket API (documentation, issues)
First occurred: 16:00:15 (2 occurrences)
Last logged: 16:14:30

[2760810016] name 'bridge' is not defined
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 199, in handle_call_service
    await hass.services.async_call(
  File "/usr/src/homeassistant/homeassistant/core.py", line 1849, in async_call
    task.result()
  File "/usr/src/homeassistant/homeassistant/core.py", line 1889, in _execute_service
    await cast(Callable[[ServiceCall], Awaitable[None]], handler.job.target)(
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 226, in handle_service
    await service.entity_service_call(
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 798, in entity_service_call
    future.result()  # pop exception if have
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 980, in async_request_call
    await coro
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 838, in _handle_entity_call
    await result
  File "/config/custom_components/huawei_solar/switch.py", line 261, in async_turn_on
    device_status = (await self.bridge.client.get(rn.DEVICE_STATUS, bridge.slave_id)).value
NameError: name 'bridge' is not defined
```

### Please confirm the following:

- [X] The problem is still present in the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [X] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [X] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

wlcrs, 2023-06-04:
I don't provide support for NodeRED as I'm not familiar with it. Feel free to submit a PR if you're able to fix the issue.

Please also note that I strongly advice against using that switch on a regular basis. Switching the inverter on and off when it's under load will negatively impact the relays and lead to premature failure of your device.

If you just want to prevent the inverter from feeding to the grid when the price is negative, them you're better of looking into the services that this integration (ie. `huawei_solar.set_maximum_feed_grid_power_percent`)

harnyd, 2023-06-04:
@wlcrs I know, but the problem is regarding HA and huawei solar extension.
This is from HA log and refenrences to python.

File "/config/custom_components/huawei_solar/switch.py", line 261, in async_turn_on
    device_status = (await self.bridge.client.get(rn.DEVICE_STATUS, bridge.slave_id)).value
NameError: name 'bridge' is not defined

closed 2023-07-19 by wlcrs