wlcrs/huawei_solar #387: [Feature Request]: Expose services for other Active Power Control modes

petr5250, 2023-06-04:
### Describe your feature request

Hey, would it be possible to expose services for setting the inverter into DI_ACTIVE_SCHEDULING and ZERO_POWER_GRID_CONNECTION Active Power Control modes?

DI_ACTIVE_SCHEDULING mode is sometimes used in conjunction with DI inputs of the inverter for remote control of the inverter by electricity distribution companies (ie. for remote shutdown of the PV power plant). In this situation it is not possible to use UNLIMITED mode which will disable this remote control. 

ZERO_POWER_GRID_CONNECTION mode can be used in situation when there are low (negative) feed-in electricity prices, but consumption prices are fixed (independent of the feed-in prices). In this situation it is practical to stop delivering the surplus energy to the grid but still be able to make some energy for own consumption.

### Proper usage

- [X] I confirm that this is not a bug report
- [X] I confirm that this feature request is within the stated scope of the integration

dim123dim123, 2023-07-17:
doest Modbus have this register? Where is possible find latest Modbus spec ?

closed 2023-07-17 by wlcrs

petr5250, 2023-07-17:
Is the same register (41415 - rn.ACTIVE_POWER_CONTROL_MODE) as for the other Active power control modes, that are already exposed in services (Huawei Solar: Set Active Power Control to Unlimited mode, Huawei Solar: Limit the power fed to the grid and Huawei Solar: Limit the power fed to the grid to percentage)

Corresponding values are already available in huawei-solar-lib/src/huawei_solar/register_values.py, class ActivePowerControlMode, constants DI_ACTIVE_SCHEDULING and ZERO_POWER_GRID_CONNECTION. There are jus no services in huawei_solar module to use them.