wlcrs/huawei_solar #393: [Bug]: HA Core 2023.6.1 fails to start integration

bengtner, 2023-06-10:
### Describe the issue

I just made an HA Core update to 2023.6.1 and I got a notification this integration didn't start. No special information available in logs. Just a statement it fails. 

### Bescribe your Huawei Solar Setup

Inverter Type:
Inverter Firmware version:
SDongle present: yes / no
Power meter present: three phase / single phase / no
Battery: LG RESU xxkWh / LUNA2000 xxkWh / None
Battery Firmware version:

N/A

### How do you connect to the inverter?

Over serial, with a serial-to-USB stick

### Upload your Diagnostics File

Not possible to get any more info. I wonder if this is related to the Python update in HA Core 2023.6?  Unfortunately I have not checked which release of this integration I use. Will do that later after I have restored my backup. But I have looked into the commits but they are at least one month old. So an issue the 2023.6 it most likely not solved by those.

### Upload your relevant debug logs

```text
N/A
```

### Please confirm the following:

- [ ] The problem is still present in the latest release of this integration.
- [ ] I did not find an existing issue describing this problem.
- [ ] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [ ] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

bengtner, 2023-06-10:
I have restored my 2024.5.4 backup and everything is fine again, I'm running a quite old release of this integration 1.1.6 and I should of course update this. But it is good to know if anyone can confirm it works on 2023.6.1.

johansolve, 2023-06-10:
I had the same issue but I installed a new HACS (had an old version that didn’t run on the 2023.6.1 update), then I could update the Huawei integration and now it works fine.

bengtner, 2023-06-10:
Thanks for this info. I will upgrade to the latest release as well. If this solves the problem with 2023.6, I will close this issue.

Raffifumi, 2023-06-11:
i updated to 2023.6 and having same issue. restored a previous backup, and huawei solar is working. Came back to 2023.6 and it says again:"Invalid handler specified"

bengtner, 2023-06-12:
@Raffifumi, which version of this integration did you run on 2023.6 when you had the failure?

Raffifumi, 2023-06-12:
Last one.

bengtner, 2023-06-12:
Ok. Then I assume it is an Python issue as i first assumed. I guess @wlcrs has to look into this. I will not make any further attempts since it is quite time-consuming to restore a backup even if the procedure is smooth.

wlcrs, 2023-06-12:
No logs = nothing for me to do

closed 2023-06-12 by wlcrs