wlcrs/huawei_solar #398: [Bug]: Impossibile caricare il flusso di configurazione: {"message":"Invalid handler specified"}

Raffifumi, 2023-06-19:
### Describe the issue

I updated to home assistant 2023.6, and, since that moment, the integration is not working. I updated huawei solar to last version and it's not working. If i restore a previous versioni of Home Assistant, integration works.
 when i try to reinstall it says: : Impossibile caricare il flusso di configurazione: {"message":"Invalid handler specified"}

This is the log:

Error occurred loading flow for integration huawei_solar: Exception importing custom_components.huawei_solar-main.config_flow
15:17:21 – (ERRORE) config_entries.py
Unexpected exception importing component custom_components.huawei_solar-main
15:17:21 – (ERRORE) huawei_solar-main (integrazione personalizzata) - il messaggio si è verificato per la prima volta alle 15:02:18 e compare 4 volte

### Bescribe your Huawei Solar Setup

Inverter Type: SUN2000-6KTL-L1
SDongle present: yes 
Power meter present: / no
Battery: LG RESU / None

### How do you connect to the inverter?

Via the SDongle, wireless connection

### Upload your Diagnostics File

Drag & Drop your Diagnostics File here.

### Upload your relevant debug logs

```text
.
```

### Please confirm the following:

- [X] The problem is still present in the latest release of this integration.
- [ ] I did not find an existing issue describing this problem.
- [ ] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [ ] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

wlcrs, 2023-06-20:
no proper logs = nothing for me to do.

closed 2023-06-20 by wlcrs

ragesoft, 2023-07-08:
Hi there. I've just started with my New solar Inverter.
Try this Integration but failed with same message.
Invalid handler specified

I allready have one Integration using modbus (solarfocus buffer / heater) 
Maybe there is a version conflict... 

Logger: homeassistant.config_entries
Source: config_entries.py:849
First occurred: 10:27:00 (3 occurrences)
Last logged: 10:31:02

Error occurred loading configuration flow for integration huawei_solar: cannot import name 'AsyncModbusSerialClient' from 'pymodbus.client' (/usr/local/lib/python3.10/site-packages/pymodbus/client/__init__.py)