wlcrs/huawei_solar #401: [Feature Request]:

vwtuner, 2023-06-21:
### Describe your feature request

„Backup Batterie Kapazität“( Reservierte Batterie Kapazität)
Einstellung bitte hinzufügen 
[image]

### Proper usage

- [X] I confirm that this is not a bug report
- [X] I confirm that this feature request is within the stated scope of the integration

streamthing, 2023-07-02:
Write in English!!!

streamthing, 2023-07-02:
@vwtuner Write in English!!!

vwtuner, 2023-07-02:
"Backup Battery Capacity" (Reserved Battery Capacity)
Please add setting

WiebKastanje, 2023-07-02:
Hi @vwtuner, I'm too interested in this described Reserved Battery Capacityf eature. We have to make clear, in plain english, what is needed.
Please provide the link to Huawei document are you referring to, describing the "netzentkoppelte modes"? And abbildung 3-15?   
Thank you very much.

vwtuner, 2023-07-02:
[Back up Anleitung .pdf](https://github.com/wlcrs/huawei_solar/files/11929948/Back.up.Anleitung.pdf)

[image]

Here is the English version.

WiebKastanje, 2023-07-02:
The feature request is: 

Kindly add input field to set the ""Backup power SOC" parameter to a certain percentage. Maximum = 100%. 
Default value = NAN
[image]

This field is an off-grid control parameter, specially important for users with a Huawei backup-box connected to the inverter, which provides energy when utility powergrid is down.

Thank you.

vwtuner, 2023-07-10:
Hi, thank you very much for your effort to add the backup soc.
unfortunately there is always an error when entering a value.
when I want to activate loading from the network, an error also occurs (which worked before)Gesendet mit der mobilen Mail AppAm 10.07.23 um 15:43 schrieb Thijs W.
                
            
            
                
                    Von: "Thijs W." ***@***.***>Datum: 10. Juli 2023An: "wlcrs/huawei_solar" ***@***.***>Cc: "vwtuner" ***@***.***>,"Mention" ***@***.***>Betreff: Re: [wlcrs/huawei_solar] [Feature Request]: Backup Power SOC entity (Issue #401)
                
                
This feature was implemented in 1.2.6 beta 8, but as I cannot detect if a battery box is installed the entity is disabled by default.

Please also note that the valid range of this register is influenced by the Capacity Control SOC (when enabled). I haven't implemented any checks on whether a valid value is being configured and as I do not own a backup box I have no idea what happens when you do so.
Use at your own risk.

vwtuner, 2023-07-13:
[image]
The SOC backup is from 0-100% 
Not from 20-100%

wlcrs, 2023-07-15:
Please test again with 1.2.6 beta 9

vwtuner, 2023-07-15:
currently it works as it should, thank you for the great work. may I ask where the error was?

closed 2023-07-19 by wlcrs