wlcrs/huawei_solar #408: [Bug]:

trisplay, 2023-07-05:
### Describe the issue

Integration crashes after updating HA to 2023.7

### Bescribe your Huawei Solar Setup

Inverter Type:
Inverter Firmware version:
SDongle present: yes / no
Power meter present: three phase / single phase / no
Battery: LG RESU xxkWh / LUNA2000 xxkWh / None
Battery Firmware version:

### How do you connect to the inverter?

Via the SDongle, wireless connection

### Upload your Diagnostics File

Logger: homeassistant.setup
Source: custom_components/huawei_solar/__init__.py:25
Integration: Huawei Solar
First occurred: 22:46:16 (1 occurrences)
Last logged: 22:46:16

Setup failed for custom integration huawei_solar: Unable to import component: cannot import name 'make_byte_string' from 'pymodbus.utilities' (/usr/local/lib/python3.11/site-packages/pymodbus/utilities.py)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 215, in _async_setup_component
    component = integration.get_component()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 813, in get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1147, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/config/custom_components/huawei_solar/__init__.py", line 25, in <module>
    from huawei_solar import HuaweiSolarBridge, HuaweiSolarException, InvalidCredentials
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/__init__.py", line 6, in <module>
    from .bridge import HuaweiSolarBridge  # noqa
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/bridge.py", line 16, in <module>
    from .huawei_solar import DEFAULT_BAUDRATE, DEFAULT_SLAVE, DEFAULT_TCP_PORT, AsyncHuaweiSolar, Result
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 17, in <module>
    from pymodbus.payload import BinaryPayloadBuilder, BinaryPayloadDecoder
  File "/usr/local/lib/python3.11/site-packages/pymodbus/payload.py", line 18, in <module>
    from pymodbus.utilities import (
ImportError: cannot import name 'make_byte_string' from 'pymodbus.utilities' (/usr/local/lib/python3.11/site-packages/pymodbus/utilities.py)

### Upload your relevant debug logs

```text
2023-07-05 22:44:35.721 WARNING (SyncWorker_2) [homeassistant.loader] We found a custom integration huawei_solar which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2023-07-05 22:46:16.628 ERROR (MainThread) [homeassistant.setup] Setup failed for custom integration huawei_solar: Unable to import component: cannot import name 'make_byte_string' from 'pymodbus.utilities' (/usr/local/lib/python3.11/site-packages/pymodbus/utilities.py)
File "/config/custom_components/huawei_solar/__init__.py", line 25, in <module>
from huawei_solar import HuaweiSolarBridge, HuaweiSolarException, InvalidCredentials
File "/usr/local/lib/python3.11/site-packages/huawei_solar/__init__.py", line 6, in <module>
File "/usr/local/lib/python3.11/site-packages/huawei_solar/bridge.py", line 16, in <module>
from .huawei_solar import DEFAULT_BAUDRATE, DEFAULT_SLAVE, DEFAULT_TCP_PORT, AsyncHuaweiSolar, Result
File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 17, in <module>
```

### Please confirm the following:

- [X] The problem is still present in the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [X] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [X] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

wlcrs, 2023-07-06:
I've done the upgrade myself and the integration just kept working. This is thus not an universal issue with this integration and HA 2023.07.

The stacktrace can only be caused by an incompatible version of pyModbus being installed on your system.

In your logs I do see that you have another integration that is also using pyModbus. I suspect there is a version conflict for that underlying library. Can you tell me what that other integration is? What about @jaxer and @netmb : which integrations do you have that are also communicating over Modbus to the device ?

Can you please submit the  'Download diagnostics' file also? If this is not possible for this integration, you can upload the file of any other integration as that also offers some insights into your installation.

jaxer, 2023-07-06:
Hey, thanks for suggestions. You are probably right. I tried "re-download integration" in HACS and restarted Home Assistant and it worked alright this time. So huawei sensors are back reporting values for me.

I only have few HACS integrations and only Huawei is using modbus i think. I do use core [modbus integration](https://www.home-assistant.io/integrations/modbus/) though for couple other devices.

wlcrs, 2023-07-06:
The core Modbus integration has a dependency on a newer version of pymodbus: https://github.com/home-assistant/core/blob/de24860c87b2042fdba548367d6d70dc5f9a38d0/homeassistant/components/modbus/manifest.json#L9 .

I'll need to upgrade this library to match the same pymodbus version. This requires some testing as upgrades have caused bugs in the past.

trisplay, 2023-07-06:
Thanks for diagnosis. I use indeed the core modbus integration for my Pluggit ventilation system. After disabling this your integration is working fine. I'll wait for your update to reintegrate vents.

luzik, 2023-07-07:
Same here. I am using pymodbus in several integrations, so there is no way to go. Be my hero, and save my energy stats history please :) I can test beta whole day long

guevara777, 2023-07-07:
Same Here. Using Modbus Integration for my Heat pump. Will also bei able to Beta Test.

nhogblom, 2023-07-08:
I also had this issue, restarting home assistant solved the problem. But i noticed that the integration fails again next 00:00.
I rolled back to previous HA verision.

compuvision2, 2023-07-09:
I have also this problem with missing Huawei data. After downgrading to HA 6.3 everything is fine now.

artibea, 2023-07-09:
Hi. I have a similar problem. Unfortunately I cannot downgrade HA to version 6.3. Tried "ha core update --version 2023.6.3" but no luck. Is there any solution to reinstate huawei_solar integration?

Jordan87, 2023-07-09:
For me "Redownload" via HACS helped, withoput downgading HA.

artibea, 2023-07-09:
Didn't helped that way. Tried multiple versions of this integration. It is however interesting as it stopped working yesterday. Was working fine for a couple days without issue.

artibea, 2023-07-09:
Ok. Reverted HA back to version 6.3. Working fine.

boku78, 2023-07-10:
> Thanks for diagnosis. I use indeed the core modbus integration for my Pluggit ventilation system. After disabling this your integration is working fine. I'll wait for your update to reintegrate vents.

I do have the exact same problem. I also use the core modbus integration for my Pluggit ventilaation System and after removing the core modbus the Huawei Solar integration works. I'm going to test the beta 8

boku78, 2023-07-10:
The 1.2.6 beta 8 seems to be working as well as my core modbus!

trisplay, 2023-07-10:
> The 1.2.6 beta 8 seems to be working as well as my core modbus!

Same here. Both Integrations working side by side at first sight.
Well done and thank you for sharing this.

guevara777, 2023-07-10:
> The 1.2.6 beta 8 seems to be working as well as my core modbus!

Working fine with 1.2.6b8

boku78, 2023-07-11:
Unfortunately it seems to crash from time to time, and i have to restart HA for the Huawei solar integration to work again!

wlcrs, 2023-07-11:
@boku78 Please enable diagnostics and create a new issue with the logs you gathered once the problem shows itself.

closed 2023-07-11 by wlcrs

lauer, 2023-07-12:
Just a note, if you cant update to beta. I have an automation running the last weeks, to reload modbus when one of the entities from this integration gets unavailable. 

````yaml
alias: Huawei - restart service unavailable
description: ""
trigger:
  - platform: state
    entity_id:
      - sensor.huawei_battery_status
    to: unavailable
condition: []
action:
  - service: modbus.reload
    data: {}
  - delay:
      hours: 0
      minutes: 0
      seconds: 10
      milliseconds: 0
  - service: homeassistant.reload_core_config
    data: {}
mode: single
````

On my system, this is automatic triggered every evening around 11:50PM.