wlcrs/huawei_solar #409: Cant connect with sin 2000 [Bug]:

Marlov11, 2023-07-08:
### Describe the issue

I can see the Sdongel via modbus poll, but the installation says "Unexpected error"
salve ID: 1
Advanced: elevate permissions : yes 

 

### Bescribe your Huawei Solar Setup

Inverter Type:
Inverter Firmware version:
SDongle present: yes / no
Power meter present: three phase / single phase / no
Battery: LG RESU xxkWh / LUNA2000 xxkWh / None
Battery Firmware version:

### How do you connect to the inverter?

Over serial, with a serial-to-USB stick

### Upload your Diagnostics File

none

### Upload your relevant debug logs

```text
Denne fejl stammer fra en brugerdefineret integration.

Logger: custom_components.huawei_solar.config_flow
Source: custom_components/huawei_solar/config_flow.py:159
Integration: Huawei Solar
First occurred: 18.26.31 (2 occurrences)
Last logged: 18.41.30

object NoneType can't be used in 'await' expression
Traceback (most recent call last):
  File "/config/custom_components/huawei_solar/config_flow.py", line 384, in async_step_setup_network
    info = await validate_network_setup(user_input)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/huawei_solar/config_flow.py", line 159, in validate_network_setup
    await bridge.stop()
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/bridge.py", line 300, in stop
    return await self.client.stop()
           ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 203, in stop
    await self._client.close()
TypeError: object NoneType can't be used in 'await' expression
```

### Please confirm the following:

- [X] The problem is still present in the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [X] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [X] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

wlcrs, 2023-07-10:
Can you please test again with 1.2.6 beta 8 ?

Marlov11, 2023-07-10:
Hi It works

;o)

but there is a problem with the optimizer, they are not available, I have 20 pieces. but inverter and battery are coming in... great..

from my heart thanks..

BR/ Martin

closed 2023-07-10 by wlcrs