wlcrs/huawei_solar #415: [Bug]: Integration crashes from time to time

boku78, 2023-07-13:
### Describe the issue

After the HA Update to 2023.07.01 i updated the huawei_solar integration to 1.2.6b8. The problem with the python libaries seems to be gone. But from time to time the integration crashes, after restarting HA Core the integration runs for several hours and crashes again.
The error occurs at 2023-07-12 09:57:08.379

### Bescribe your Huawei Solar Setup

Inverter Type: SUN2000-8KTL-M1 
Inverter Firmware version: V100R001C00SPC153
SDongle present: yes
Power meter present:  three phase 
Battery: LUNA2000
Battery Firmware version:

### How do you connect to the inverter?

Over serial, with a serial-to-USB stick

### Upload your Diagnostics File

Drag & Drop your Diagnostics File here.
[config_entry-huawei_solar-0fb10aad00efb384d23ddc00dbc8c83f.json.txt](https://github.com/wlcrs/huawei_solar/files/12036765/config_entry-huawei_solar-0fb10aad00efb384d23ddc00dbc8c83f.json.txt)

### Upload your relevant debug logs

```text
023-07-12 09:57:04.355 DEBUG (SyncWorker_38) [pymodbus.logging] Getting transaction 39516
2023-07-12 09:57:04.355 DEBUG (SyncWorker_38) [pymodbus.logging] Changing transaction state from "PROCESSING REPLY" to "TRANSACTION_COMPLETE"
2023-07-12 09:57:08.379 INFO (MainThread) [backoff] Backing off _do_read(...) for 1.0s (TimeoutError)
2023-07-12 09:57:08.379 DEBUG (MainThread) [huawei_solar.huawei_solar] Received <class 'TimeoutError'>: backing off reading for 1.0 seconds after 1 tries
2023-07-12 09:57:08.380 DEBUG (MainThread) [pymodbus.logging] Connection lost comm due to None
2023-07-12 09:57:08.381 DEBUG (MainThread) [pymodbus.logging] Getting transaction 5256
2023-07-12 09:57:09.381 ERROR (MainThread) [huawei_solar.huawei_solar] Modbus client is not connected to the inverter.
NoneType: None
2023-07-12 09:57:09.382 INFO (MainThread) [backoff] Backing off _do_read(...) for 1.0s (huawei_solar.exceptions.ConnectionInterruptedException: Modbus client is not connected to the inverter.)
2023-07-12 09:57:09.382 DEBUG (MainThread) [huawei_solar.huawei_solar] Received <class 'huawei_solar.exceptions.ConnectionInterruptedException'>: backing off reading for 1.0 seconds after 1 tries
2023-07-12 09:57:10.384 ERROR (MainThread) [huawei_solar.huawei_solar] Modbus client is not connected to the inverter.
NoneType: None
2023-07-12 09:57:10.385 INFO (MainThread) [backoff] Backing off _do_read(...) for 2.0s (huawei_solar.exceptions.ConnectionInterruptedException: Modbus client is not connected to the inverter.)
2023-07-12 09:57:10.385 DEBUG (MainThread) [huawei_solar.huawei_solar] Received <class 'huawei_solar.exceptions.ConnectionInterruptedException'>: backing off reading for 2.0 seconds after 2 tries
2023-07-12 09:57:12.386 ERROR (MainThread) [huawei_solar.huawei_solar] Modbus client is not connected to the inverter.
NoneType: None
2023-07-12 09:57:12.387 INFO (MainThread) [backoff] Backing off _do_read(...) for 4.0s (huawei_solar.exceptions.ConnectionInterruptedException: Modbus client is not connected to the inverter.)
2023-07-12 09:57:12.387 DEBUG (MainThread) [huawei_solar.huawei_solar] Received <class 'huawei_solar.exceptions.ConnectionInterruptedException'>: backing off reading for 4.0 seconds after 3 tries
2023-07-12 09:57:16.387 ERROR (MainThread) [huawei_solar.huawei_solar] Modbus client is not connected to the inverter.
NoneType: None
2023-07-12 09:57:16.389 INFO (MainThread) [backoff] Backing off _do_read(...) for 8.0s (huawei_solar.exceptions.ConnectionInterruptedException: Modbus client is not connected to the inverter.)
2023-07-12 09:57:16.389 DEBUG (MainThread) [huawei_solar.huawei_solar] Received <class 'huawei_solar.exceptions.ConnectionInterruptedException'>: backing off reading for 8.0 seconds after 4 tries
2023-07-12 09:57:18.430 ERROR (MainThread) [custom_components.huawei_solar] Timeout fetching BT2240017925_data_update_coordinator data
2023-07-12 09:57:18.431 DEBUG (MainThread) [custom_components.huawei_solar] Finished fetching BT2240017925_data_update_coordinator data in 20.060 seconds (success: False)
2023-07-12 09:57:18.447 ERROR (MainThread) [homeassistant.helpers.template_entity] TemplateError('ValueError: Template error: float got invalid input 'unavailable' when rendering template '{% set meter_power = states('sensor.power_meter_active_power') | float %} {% set inverter_power = states('sensor.inverter_active_power') | float %} {{ (inverter_power - meter_power )  | float(0) }}' but no default was specified') while processing template 'Template<template=({% set meter_power = states('sensor.power_meter_active_power') | float %} {% set inverter_power = states('sensor.inverter_active_power') | float %} {{ (inverter_power - meter_power )  | float(0) }}) renders=4250>' for attribute '_attr_native_value' in entity 'sensor.house_power_consumption'
2023-07-12 09:57:19.079 DEBUG (SyncWorker_25) [pymodbus.logging] Current transaction state - TRANSACTION_COMPLETE
```

### Please confirm the following:

- [X] The problem is still present in the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [X] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [X] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

VaabenDK, 2023-07-13:
not solving anything - but came here as I'm seeing the same in my installation - using the same versions and products - just a 6KW inverter instead

lagge78, 2023-07-14:
I have the same issue, for me the integration crashes a couple of minutes after midnight each night.
I have the latest xx8 beta of the integration running on latest HA 2023.07.1

wlcrs, 2023-07-14:
Can someone provide me with a longer debugging log? The few lines that I got in the initial report are too limited to understand what is going on. I need at least a few minutes of the querying going OK before it starts to fail.

Thanks in advance.

boku78, 2023-07-15:
Hi,

i do have the whole logfile ... please have alook here:

[home-assistant_huawei_solar_2023-07-12T09-51-37.967Z-short.log](https://github.com/wlcrs/huawei_solar/files/12060687/home-assistant_huawei_solar_2023-07-12T09-51-37.967Z-short.log)

boku78, 2023-07-15:
I do have the complete Logfile, but is 78MB in Size ... i shortended it and attached it.

Please have a look here:
[home-assistant_huawei_solar_2023-07-12T09-51-37.967Z_not_so_short.log](https://github.com/wlcrs/huawei_solar/files/12060696/home-assistant_huawei_solar_2023-07-12T09-51-37.967Z_not_so_short.log)

peixoto482, 2023-07-15:
También se me queda colgada la comunicación. No se si desde la actualización de HA o desde la actualización de la integración de huawei. Se soluciona recargando la integración, no hace falta reiniciar HA. Yo me conecto con inversor Huawei a través de Ewin11, hasta ahora había ido bien.

wlcrs, 2023-07-15:
I have the impression that the automatic reconnect feature of pyModbus is broken. I've added some explicit reconnects when I detect the connection fails.

Please test again with 1.2.6 beta 9.

peixoto482, 2023-07-15:
Lo probaré cuando lo pueda instalar desde HACS. Todavía no está en HACS esa
versión.

Un saludo y gracias.

boku78, 2023-07-16:
It seems to run better now. I installed 1.26.b9 yesterday and until it didn't crash!

THX

peixoto482, 2023-07-16:
Good to hear!

Lo acabo de instalar. Entiendo que problema solucionado. Gracias

nurzen, 2023-07-17:
There is an error in the manifest.json, there is a comma too much at the end of line 11
2023-07-17 10:17:25.758 ERROR (SyncWorker_0) [homeassistant.loader] Error parsing manifest.json file at /config/custom_components/huawei_solar/manifest.json: trailing comma is not allowed: line 11 column 44 (char 414)

fodi666, 2023-07-18:
I also had the disconnect issue since updating to the july version of home assistant, with the 1.2.6b9 update seems to solve it (no disconnect since yesterday)

Misiu, 2023-07-19:
I was going to report this issue.
I've noticed that for the last couple of days, I don't see solar production in the energy dashboard.
I will try the latest beta, hopefully, it will work.

P.S. Is there a way to fill the gaps in the energy dashboard? I see data in the Fusion Solar App, so ideally missing data should be downloaded from there.

fodi666, 2023-07-19:
> P.S. Is there a way to fill the gaps in the energy dashboard? I see data in the Fusion Solar App, so ideally missing data should be downloaded from there.

If you add the "Inverter total yield" sensor to your energy dashboard, then you can at least have the total amount missing to a single hour (the hour when you restart the integration), so your monthly totals can still be fine. However, no hourly data will be added since that would need you to retroactively add data to Homa Assistant recorder (not going to work unless you make some manual database hacking)

Misiu, 2023-07-19:
I've done a quick search on the forum and I found this: https://community.home-assistant.io/t/import-old-energy-readings-for-use-in-energy-dashboard/341406/9?u=misiu
According to the info I found it is possible to import missing data, but let's keep this issue clean.
I'll create a separate issue (feature request) for this.

wlcrs, 2023-07-19:
I've released 1.2.6 as a stable version

closed 2023-07-19 by wlcrs