wlcrs/huawei_solar #421: [Feature Request]: Import historical data and fill the gaps is recorder (statistics)

Misiu, 2023-07-19:
### Describe your feature request

For reasons unknown to me (library errors, network connectivity, or others), I noticed that I have holes in solar production in energy dashboard.
If integration is unavailable or has problems with modbus communication we get no data hor that time.
According to the community thread https://community.home-assistant.io/t/import-old-energy-readings-for-use-in-energy-dashboard/341406/9?u=misiu it is possible to add historical data to energy dashboard.

In my case I see that I don't have data for a couple of days this month:
[image]
but the data is available in Fusion Solar App:
[image]

I know this integration is using Modbus to communicate with the inverter, but I'd like to propose an option to add login/pass for fusion solar and download and fill data for missing days/hours.
I have solar panels for 1.5 years now, but this integration is set up only for two months, so the historical data would be useful to compare usages.

not sure if there is an official API for Fusion Solar, but I think that downloading the data would be the easiest part, finding the gaps and filling them with the correct data would be the hardest.

### Proper usage

- [X] I confirm that this is not a bug report or support request
- [X] I confirm that this feature request is within the stated scope of the integration

wlcrs, 2023-07-19:
The modbus definitions file contains no information on how to retrieve historical data from the inverter.

I would need to reverse engineer the app to obtain the necessary info on how to do that. As it is undocumented behavior it can change and break at any moment.

This is too much work and too much of a future maintenance headache to risk implementing such a non-essential feature.

If somebody feels like volunteering: please react in this issue. I can give some pointers on how I think it will be implemented on Huawei-side.

Misiu, 2023-07-19:
@wlcrs  thank you for the reply.
I'm aware that Modbus doesn't allow retrieving historical data.
After looking at web interface (https://region02eu5.fusionsolar.huawei.com/) I was able to get the request that returns Daily energy:
[image]

https://region02eu5.fusionsolar.huawei.com/rest/pvms/web/device/v1/device-history-data?signalIds=30016&deviceDn=NE%3DXXXXXXXX&date=1689600057049&_=1689765761549 (I've redacted some info from URL,

The API probably isn't documented, but with it, we don't have to revere engineer the app.

wlcrs, 2023-07-19:
FusionSolar web is outside the scope of this integration

Misiu, 2023-07-19:
I'm sure I'll be able to download the required data from FusionSolar Web, but I need to read more about how to import it to Home Assistant. I'll share the results if I'll have success.

yousaf465, 2023-07-20:
I face a similar issue, when there is loadshedding / blackout,s data for that time is missing. It could be great if someone could create a PR for this.

karliwalti, 2023-09-12:
Hi
Huawei has an API called Northbound. Maybe you can find there a solution on how to get access to historical data. 
The rather complicated approach to get access through your installer is described here (german)
https://docs.clever-pv.com/electricmeter/huawei

wlcrs, 2023-09-12:
Using any Web API from Huawei is out of scope for this integration. I will not allow any code connecting to a Huawei server in this integration. For anyone who wants to work on this issue: historical data must be retrieved from the inverter itself. I know it does store this info locally, but it is undocumented and must be reverse engineered.

FYI: there is an excellent integration out there which does connect to Huawei FusionSolar Web: https://github.com/tijsverkoyen/HomeAssistant-FusionSolar .

yousaf465, 2023-09-12:
> @wlcrs the idea behind this issue was simple - missing data on the graph in HA. In my case, I had to restart HA to get everything working. Maybe instead of trying to fill the gaps with data from web API, we should focus on the main issue - what is happening and how to solve that. If I remember correctly, I had a power issue in my house (HA was down and my router was down), so when everything got up I started to have missing data. Maybe because the router takes longer to start or HA needs some time or the inverter needs some time to connect to my wifi, I'm not sure about that. I can enable logging and turn the power off and on in my house to simulate that situation. Maybe the logs will tell a bit more.
> 
> What components I should log and on what level? Is there anything else I can do to help diagnose this issue?

Do you mean this issue? I had this issue too if the power is down after sunset. rebooting HA fixes it. 
[image]
[image]

Misiu, 2023-09-13:
[image]
This is the situation from this month. I have no data from 2nd to 5th September and no data from today.

yousaf465, 2023-10-14:
anyone willing to contribute for a solution to this problem?

closed 2023-11-26 by github-actions[bot]