wlcrs/huawei_solar #422: [Bug]: failed to setup

yousaf465, 2023-07-20:
### Describe the issue

[image]

since two days of facing this issue. how to download the debug logs? 

### Bescribe your Huawei Solar Setup

Inverter Type:
Inverter Firmware version:
SDongle present: yes / no
Power meter present: three phase / single phase / no
Battery: LG RESU xxkWh / LUNA2000 xxkWh / None
Battery Firmware version:

### How do you connect to the inverter?

Via the `SUN2000-<serial_no>` WiFi

### Upload your Diagnostics File

Drag & Drop your Diagnostics File here.
[home-assistant_huawei_solar_2023-07-20T05-38-52.296Z.log](https://github.com/wlcrs/huawei_solar/files/12102380/home-assistant_huawei_solar_2023-07-20T05-38-52.296Z.log)

### Upload your relevant debug logs

```text
2023-07-20 04:22:55.333 ERROR (MainThread) [huawei_solar.huawei_solar] Timeout while waiting for connection. Reconnecting...
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/asyncio/tasks.py", line 490, in wait_for
    return fut.result()
           ^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/asyncio/locks.py", line 213, in wait
    await fut
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 141, in _communication_lock
    await asyncio.wait_for(self._client.connected_event.wait(), WAIT_FOR_CONNECTION_TIMEOUT)
  File "/usr/local/lib/python3.11/asyncio/tasks.py", line 492, in wait_for
    raise exceptions.TimeoutError() from exc
TimeoutError
2023-07-20 04:22:55.452 ERROR (MainThread) [huawei_solar.huawei_solar] Aborting client creation due to error.
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/asyncio/tasks.py", line 490, in wait_for
    return fut.result()
           ^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/asyncio/locks.py", line 213, in wait
    await fut
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 178, in create
    await huawei_solar._initialize()
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 107, in _initialize
    self.time_zone = (await self.get(rn.TIME_ZONE)).value
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 241, in get
    return (await self.get_multiple([name], slave))[0]
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 276, in get_multiple
    response = await self._read_registers(registers[0].register, total_length, slave)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 396, in _read_registers
    async with self._communication_lock():
  File "/usr/local/lib/python3.11/contextlib.py", line 204, in __aenter__
    return await anext(self.gen)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 145, in _communication_lock
    raise err
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 141, in _communication_lock
    await asyncio.wait_for(self._client.connected_event.wait(), WAIT_FOR_CONNECTION_TIMEOUT)
  File "/usr/local/lib/python3.11/asyncio/tasks.py", line 492, in wait_for
    raise exceptions.TimeoutError() from exc
TimeoutError
2023-07-20 04:22:55.514 ERROR (MainThread) [homeassistant.config_entries] Error setting up entry SUN2000-10KTL-M1
```

### Please confirm the following:

- [X] The problem is still present in the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [ ] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [X] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

danise76, 2023-07-22:
I have the same problem after updating from 1.25 to 1.26
I have 2 inverters connected with Serial Modbus. 1 inverter has a Power meter and the 2nd one only has the inverter.
Is added with "slave IDs" "1, 2" in the configuration.

I tried removing the component, reloading, etc. Nothing works, same errors as you indicate in the log.

If I add it with ONLY 1 slave, it works, but i only see one of the inverter (Either the main one + power meter, or the 2nd one).

For now i'm with 1.25 still

wlcrs, 2023-07-22:
pyModbus 3.4.0 has been released 2 days ago, mentioning some bugfixes concerning the serial connections (https://github.com/pymodbus-dev/pymodbus/releases/tag/v3.4.0).

Can you add an explicit dependency "pyModbus==3.4.0" in your manifest.json, restart HA and check if that fixes things? I cannot test on my own inverter as I seem to have fried its serial port somewhere last year during development 🙈

danise76, 2023-07-23:
Changed manifest.json to:

{
    "domain": "huawei_solar",
    "name": "Huawei Solar",
    "config_flow": true,
    "dependencies": [],
    "documentation": "https://github.com/wlcrs/huawei_solar/wiki",
    "issue_tracker": "https://github.com/wlcrs/huawei_solar/issues",
    "requirements": ["huawei-solar==2.2.8", "pyModbus==3.4.0"],
    "codeowners": ["@wlcrs"],
    "iot_class": "local_polling",
    "version": "1.2.6",
    "loggers": ["huawei_solar", "pymodbus"]
}

Not working. The debug:

`2023-07-23` 08:59:29.465 DEBUG (MainThread) [pymodbus.logging] Connecting to /dev/serial/by-id/usb-1a86_USB2.0-Ser_-if00-port0.
2023-07-23 08:59:29.465 DEBUG (MainThread) [pymodbus.logging] Connecting comm
2023-07-23 08:59:29.469 DEBUG (MainThread) [pymodbus.logging] Connected to comm
2023-07-23 08:59:29.469 DEBUG (MainThread) [pymodbus.logging] callback_connected called
2023-07-23 08:59:29.470 DEBUG (MainThread) [huawei_solar.modbus] Waiting for 1500 milliseconds after connection before performing operations
2023-07-23 08:59:30.472 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 43006 with length 1 from slave 1
2023-07-23 08:59:30.472 DEBUG (MainThread) [pymodbus.logging] send: 0x1 0x3 0xa7 0xfe 0x0 0x1 0xc6 0x8e
2023-07-23 08:59:30.472 DEBUG (MainThread) [pymodbus.logging] Adding transaction 1
2023-07-23 08:59:30.505 DEBUG (MainThread) [pymodbus.logging] recv: 0x1 0x3 0x2 0x0 0x3c 0xb8 0x55 addr=None
2023-07-23 08:59:30.506 DEBUG (MainThread) [pymodbus.logging] Processing: 0x1 0x3 0x2 0x0 0x3c 0xb8 0x55
2023-07-23 08:59:30.506 DEBUG (MainThread) [pymodbus.logging] Getting Frame - 0x3 0x2 0x0 0x3c
2023-07-23 08:59:30.506 DEBUG (MainThread) [pymodbus.logging] Factory Response[ReadHoldingRegistersResponse': 3]
2023-07-23 08:59:30.506 DEBUG (MainThread) [pymodbus.logging] Frame advanced, resetting header!!
2023-07-23 08:59:30.506 DEBUG (MainThread) [pymodbus.logging] Getting transaction 1
2023-07-23 08:59:30.507 DEBUG (MainThread) [pymodbus.logging] [60]
2023-07-23 08:59:30.559 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 47000 with length 1 from slave 1
2023-07-23 08:59:30.559 DEBUG (MainThread) [pymodbus.logging] send: 0x1 0x3 0xb7 0x98 0x0 0x1 0x22 0x51
2023-07-23 08:59:30.559 DEBUG (MainThread) [pymodbus.logging] Adding transaction 1
2023-07-23 08:59:30.595 DEBUG (MainThread) [pymodbus.logging] recv: 0x1 0x3 0x2 0x0 0x0 0xb8 0x44 addr=None
2023-07-23 08:59:30.595 DEBUG (MainThread) [pymodbus.logging] Processing: 0x1 0x3 0x2 0x0 0x0 0xb8 0x44
2023-07-23 08:59:30.595 DEBUG (MainThread) [pymodbus.logging] Getting Frame - 0x3 0x2 0x0 0x0
2023-07-23 08:59:30.595 DEBUG (MainThread) [pymodbus.logging] Factory Response[ReadHoldingRegistersResponse': 3]
2023-07-23 08:59:30.595 DEBUG (MainThread) [pymodbus.logging] Frame advanced, resetting header!!
2023-07-23 08:59:30.595 DEBUG (MainThread) [pymodbus.logging] Getting transaction 1
2023-07-23 08:59:30.595 DEBUG (MainThread) [pymodbus.logging] [0]
2023-07-23 08:59:30.646 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 47089 with length 1 from slave 1
2023-07-23 08:59:30.647 DEBUG (MainThread) [pymodbus.logging] send: 0x1 0x3 0xb7 0xf1 0x0 0x1 0xf2 0x4d
2023-07-23 08:59:30.648 DEBUG (MainThread) [pymodbus.logging] Adding transaction 1
2023-07-23 08:59:30.823 DEBUG (MainThread) [pymodbus.logging] recv: 0x1 0x3 0x2 0x0 0x0 0xb8 0x44 addr=None
2023-07-23 08:59:30.824 DEBUG (MainThread) [pymodbus.logging] Processing: 0x1 0x3 0x2 0x0 0x0 0xb8 0x44
2023-07-23 08:59:30.824 DEBUG (MainThread) [pymodbus.logging] Getting Frame - 0x3 0x2 0x0 0x0
2023-07-23 08:59:30.824 DEBUG (MainThread) [pymodbus.logging] Factory Response[ReadHoldingRegistersResponse': 3]
2023-07-23 08:59:30.825 DEBUG (MainThread) [pymodbus.logging] Frame advanced, resetting header!!
2023-07-23 08:59:30.825 DEBUG (MainThread) [pymodbus.logging] Getting transaction 1
2023-07-23 08:59:30.825 DEBUG (MainThread) [pymodbus.logging] [0]
2023-07-23 08:59:30.877 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 30000 with length 35 from slave 1
2023-07-23 08:59:30.878 DEBUG (MainThread) [pymodbus.logging] send: 0x1 0x3 0x75 0x30 0x0 0x23 0x1e 0x10
2023-07-23 08:59:30.878 DEBUG (MainThread) [pymodbus.logging] Adding transaction 1
2023-07-23 08:59:30.937 DEBUG (MainThread) [pymodbus.logging] recv: 0x1 0x3 0x46 0x53 0x55 0x4e 0x32 0x30 0x30 0x30 0x2d 0x35 0x4b 0x54 0x4c 0x2d 0x4c 0x31 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 addr=None
2023-07-23 08:59:30.937 DEBUG (MainThread) [pymodbus.logging] Processing: 0x1 0x3 0x46 0x53 0x55 0x4e 0x32 0x30 0x30 0x30 0x2d 0x35 0x4b 0x54 0x4c 0x2d 0x4c 0x31 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0
2023-07-23 08:59:30.938 DEBUG (MainThread) [pymodbus.logging] Frame - not ready
2023-07-23 08:59:30.974 DEBUG (MainThread) [pymodbus.logging] recv: 0x0 0x31 0x30 0x32 0x30 0x41 0x30 0x30 0x32 0x36 0x32 0x39 0x38 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x30 0x31 0x30 0x37 0x35 0x33 0x34 0x36 0x0 0x0 0x0 addr=None
2023-07-23 08:59:30.974 DEBUG (MainThread) [pymodbus.logging] Processing: 0x0 0x31 0x30 0x32 0x30 0x41 0x30 0x30 0x32 0x36 0x32 0x39 0x38 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x30 0x31 0x30 0x37 0x35 0x33 0x34 0x36 0x0 0x0 0x0
2023-07-23 08:59:30.974 DEBUG (MainThread) [pymodbus.logging] Frame - not ready
2023-07-23 08:59:30.989 DEBUG (MainThread) [pymodbus.logging] recv: 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x9 0x1 addr=None
[... cut ...]
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 423, in _perform_request
    response = await self._client.execute(request)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pymodbus/client/base.py", line 201, in async_execute
    resp = await asyncio.wait_for(
           ^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/asyncio/tasks.py", line 479, in wait_for
    return fut.result()
           ^^^^^^^^^^^^
asyncio.exceptions.CancelledError

wlcrs, 2023-07-23:
@danise76 , someone else submitted a similar bugreport for serial connections in #426 . Can you please follow the instructions in https://github.com/wlcrs/huawei_solar/issues/426#issuecomment-1646795769 to help find the regression in pyModbus?

danise76, 2023-07-24:
Same here as in the #426. It works setting the version 3.2.2 (works up to that one) in manifest.json:

`"requirements": ["huawei-solar==2.2.8", "pyModbus==3.2.2"],`

I left it updated to 1.2.6 but manually setting the manifest.json to that pyModbus version.

closed 2023-09-04 by wlcrs