wlcrs/huawei_solar #426: [Bug]: Optimizers missing after HA upgrade

julianbl, 2023-07-23:
### Describe the issue

Hi! After HA upgraded to 2023.7.3, all optimizer information became unavailable. I was with the integration in 1.2.5, so I upgraded to 1.2.6 with no luck. Also uninstalled it and re installed it, and no I don't even see them.
All the optimizers report correctly on FusionSolar.

### Bescribe your Huawei Solar Setup

Inverter Type: SUN2000-6KTL-L1
Inverter Firmware version: V200R001C00SPC130
SDongle present: no
Power meter present: single phase
Battery: LUNA2000 5kWh
Battery Firmware version: V100R002C00SPC119
Optimizer model: SUN2000-450W-P2
Optimizer firmware version: V100R002C10SPC019

### How do you connect to the inverter?

Over serial, with a serial-to-USB stick

### Upload your Diagnostics File

Drag & Drop your Diagnostics File here.
[config_entry-huawei_solar-489c6d71f49df8a11e064f95770f6de1.json.txt](https://github.com/wlcrs/huawei_solar/files/12138342/config_entry-huawei_solar-489c6d71f49df8a11e064f95770f6de1.json.txt)
[home-assistant_huawei_solar_2023-07-23T09-00-21.347Z.log](https://github.com/wlcrs/huawei_solar/files/12138351/home-assistant_huawei_solar_2023-07-23T09-00-21.347Z.log)

### Upload your relevant debug logs

```text
File uploaded in diag section
```

### Please confirm the following:

- [X] The problem is still present in the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [X] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [X] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

wlcrs, 2023-07-23:
The following snippet shows the problem: pyModbus receives the answer from the inverter in two parts, and discards the first part because it is not (yet) a valid message. This seems to be a regression to earlier 

Both parts together are a valid Modbus RTU message:

[image]

```
2023-07-23 10:59:47.460 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading file 0x45 from slave 1
2023-07-23 10:59:47.460 DEBUG (MainThread) [pymodbus.logging] send: 0x1 0x41 0x5 0x1 0x45 0xdd 0x9e
2023-07-23 10:59:47.460 DEBUG (MainThread) [pymodbus.logging] Adding transaction 1
2023-07-23 10:59:47.534 DEBUG (MainThread) [pymodbus.logging] recv: 0x1 0x41 0x5 0xa 0x45 0x0 0x0 0x7 0x4e 0xf6
2023-07-23 10:59:47.534 DEBUG (MainThread) [pymodbus.logging] recv: 0x1 0x41 0x5 0xa 0x45 0x0 0x0 0x7 0x4e 0xf6
2023-07-23 10:59:47.534 DEBUG (MainThread) [pymodbus.logging] Frame check failed, ignoring!!
2023-07-23 10:59:47.534 DEBUG (MainThread) [pymodbus.logging] Resetting frame - Current Frame in buffer - 0x41 0x5 0xa 0x45 0x0 0x0 0x7 0x4e 0xf6
2023-07-23 10:59:47.534 DEBUG (MainThread) [pymodbus.logging] Frame - [b'\x01A\x05\nE\x00\x00\x07N\xf6'] not ready
2023-07-23 10:59:47.550 DEBUG (MainThread) [pymodbus.logging] recv: 0xff 0xff 0xff 0xff 0xd8 0x3e
2023-07-23 10:59:47.550 DEBUG (MainThread) [pymodbus.logging] recv: 0xff 0xff 0xff 0xff 0xd8 0x3e
2023-07-23 10:59:47.551 DEBUG (MainThread) [pymodbus.logging] Frame check failed, ignoring!!
2023-07-23 10:59:47.551 DEBUG (MainThread) [pymodbus.logging] Resetting frame - Current Frame in buffer - 0x0 0x7 0x4e 0xf6 0xff 0xff 0xff 0xff 0xd8 0x3e
2023-07-23 10:59:57.462 INFO (MainThread) [backoff] Backing off _perform_request(...) for 10.0s (TimeoutError)
```

I sadly had to upgrade the pymodbus library to stay in sync with the version used by the native modbus-integration of HA (#408).

Can you change your `manifest.json` to fix the pymodbus version to an older one? Restarting HA should then install that older dependency:

```
{
"domain": "huawei_solar",
 ...
"requirements": ["huawei-solar==2.2.8", "pyModbus==3.1.3"],
}
```

If that works for you: can you please gradually raise the version number (cfr. here: https://pypi.org/project/pymodbus/#history) until it starts failing again? Please let me know the last version that worked for you.

julianbl, 2023-07-23:
Thanks for your quick replay and fix!
Tried it and the latest version where pymodbus works is 3.2.2. It stops working with 3.3.0.

Thanks again!

wlcrs, 2023-07-24:
Hi @julianbl and/or @danise76 , Can you please upload some debug logs with your working setup using pymodbus 3.2.2 ? I'll extract the relevant logs for the maintainer of pyModbus to have a look at.

Only with your help will we be able to fix this problem in the pyModbus library (cfr. https://github.com/pymodbus-dev/pymodbus/issues/1695 )

julianbl, 2023-07-24:
[home-assistant_huawei_solar_2023-07-24T10-20-03.271Z.log](https://github.com/wlcrs/huawei_solar/files/12144715/home-assistant_huawei_solar_2023-07-24T10-20-03.271Z.log)

Hi @wlcrs. Sure!
Not sure why it failed a couple of times while reloading but it is still working.

Thanks!

julianbl, 2023-07-24:
[home-assistant_huawei_solar_2023-07-24T10-29-26.410Z.log](https://github.com/wlcrs/huawei_solar/files/12144778/home-assistant_huawei_solar_2023-07-24T10-29-26.410Z.log)
Just in case, here is a clean log.

danise76, 2023-07-24:
Here it is. If a longer one is needed, let me know (it's cleaned).

[home-assistant_huawei_solar_2023-07-24T10-25-08.916Z.log](https://github.com/wlcrs/huawei_solar/files/12144816/home-assistant_huawei_solar_2023-07-24T10-25-08.916Z.log)

finnok, 2023-07-24:
> ### Describe the issue
> Hi! After HA upgraded to 2023.7.3, all optimizer information became unavailable. I was with the integration in 1.2.5, so I upgraded to 1.2.6 with no luck. Also uninstalled it and re installed it, and no I don't even see them. All the optimizers report correctly on FusionSolar.
> 
> ### Bescribe your Huawei Solar Setup
> Inverter Type: SUN2000-6KTL-L1 Inverter Firmware version: V200R001C00SPC130 SDongle present: no Power meter present: single phase Battery: LUNA2000 5kWh Battery Firmware version: V100R002C00SPC119 Optimizer model: SUN2000-450W-P2 Optimizer firmware version: V100R002C10SPC019
> 
> ### How do you connect to the inverter?
> Over serial, with a serial-to-USB stick
> 
> ### Upload your Diagnostics File
> Drag & Drop your Diagnostics File here. [config_entry-huawei_solar-489c6d71f49df8a11e064f95770f6de1.json.txt](https://github.com/wlcrs/huawei_solar/files/12138342/config_entry-huawei_solar-489c6d71f49df8a11e064f95770f6de1.json.txt) [home-assistant_huawei_solar_2023-07-23T09-00-21.347Z.log](https://github.com/wlcrs/huawei_solar/files/12138351/home-assistant_huawei_solar_2023-07-23T09-00-21.347Z.log)
> 
> ### Upload your relevant debug logs
> ```
> File uploaded in diag section
> ```
> 
> ### Please confirm the following:
> * [x]  The problem is still present in the latest release of this integration.
> * [x]  I did not find an existing issue describing this problem.
> * [x]  I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
> * [x]  I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)
 
Exact same hardware / Optimizer and other information missing.

HI, similar issue. Can you confirm if battery Max charge and other changeable parameters are greyed out ?

nesnajkneh, 2023-07-27:
it seems i have the same problem!

wlcrs, 2023-07-28:
pyModbus [3.4.1](https://github.com/pymodbus-dev/pymodbus/releases/tag/v3.4.1) has been released. Can you please check if the problem is solved in this version?

nesnajkneh, 2023-07-28:
I tested it, but it's even worse... so i changed back to 3.1.3

nesnajkneh, 2023-07-29:
I tried to uninstall the huawei solar entity and integration and now i installled the huawei addon again, but i can't add the entity anymore.

tried to change the modbus version in the manifest.json file

nothing works to add the entity again.... 😵‍💫😰

nesnajkneh, 2023-07-31:
got everything up and running again with:

  "requirements": ["huawei-solar==2.2.8", "pyModbus==3.2.2"],

julianbl, 2023-07-31:
[home-assistant_huawei_solar_2023-07-31T14-35-42.143Z.log](https://github.com/wlcrs/huawei_solar/files/12216186/home-assistant_huawei_solar_2023-07-31T14-35-42.143Z.log)
Does not work with 3.4.1. Attaching log. tnx!

wlcrs, 2023-08-01:
After some debugging, I found a bug in pyModbus that probably explains the behavior that we are seeing here.

The PR with the fix (https://github.com/pymodbus-dev/pymodbus/pull/1716) has already been merged, so hopefully we can confirm the fix with the next release.

JFHCFC, 2023-08-18:
after update to 2023.8.2 optimizators stops working (inverter, battery nad power meter works).

add to /config/custom_components/huawei_solar/manifest.json - fix problem
`"requirements": ["huawei-solar==2.2.8", "pyModbus==3.1.3"],`

I had before (without pyModbus)
``"requirements": ["huawei-solar==2.2.8"`

Sh1bbyz, 2023-08-20:
> after update to 2023.8.2 optimizators stops working (inverter, battery nad power meter works).
>
> add to /config/custom_components/huawei_solar/manifest.json - fix problem `"requirements": [...]

Does not work for me.

Also new update 2023.8.3 does not work either. Any solution anyone????

JFHCFC, 2023-08-22:
> Does not work for me.
>
> Also new update 2023.8.3 does not work either. Any solution anyone????

try (work on HA 2023.8.3)

/config/custom_components/huawei_solar/manifest.json

```
{
    "domain": "huawei_solar",
    "name": "Huawei Solar",
    "config_flow": true,
    "documentation": "https://github.com/wlcrs/huawei_solar/wiki",
    "issue_tracker": "https://github.com/wlcrs/huawei_solar/issues",
    "requirements": ["huawei-solar==2.2.8", "pyModbus==3.2.2"],
    "codeowners": ["@wlcrs"],
    "iot_class": "local_polling",
    "version": "1.2.6",
    "loggers": ["huawei_solar", "pymodbus"]
```

wlcrs, 2023-09-03:
@nesnajkneh can you please test and confirm that it works with pyModbus 3.5.0 as well? If so, I'll release a new version with the updated dependency.

julianbl, 2023-09-03:
Hi @wlcrs .
It does not work on 3.5.0. Adding the log.
[home-assistant_huawei_solar_2023-09-03T08-58-10.919Z.log](https://github.com/wlcrs/huawei_solar/files/12505588/home-assistant_huawei_solar_2023-09-03T08-58-10.919Z.log)

wlcrs, 2023-09-04:
Can you please test with release 1.2.7 beta 2 of this integration? Please open a new bug report with new logs if necessary.

closed 2023-09-04 by wlcrs

julianbl, 2023-09-04:
Works great with 1.2.7 beta 2!!!

JFHCFC, 2023-09-04:
Updtated to 1.2.7b2, optimizers falls down

i've change
/config/custom_components/huawei_solar/manifest.json

to 

`{
    "domain": "huawei_solar",
    "name": "Huawei Solar",
    "config_flow": true,
    "documentation": "https://github.com/wlcrs/huawei_solar/wiki",
    "issue_tracker": "https://github.com/wlcrs/huawei_solar/issues",
    "requirements": [
        "huawei-solar==2.2.9b1", **"pyModbus==3.5.0"],**
    ],
    "codeowners": [
        "@wlcrs"
    ],
    "iot_class": "local_polling",
    "version": "1.2.7b2",
    "loggers": [
        "huawei_solar",
        "pymodbus"
    ]
}`

and now works

Home Assistant 2023.8.4
Supervisor 2023.08.1
Operating System 10.2
Interfejs użytkownika: 20230802.1 - latest

Inverter - SPC155 (2 days ago SPC156 relased)
SDongielA-05 - SPC133
Optimizators (SUN-450-P2) - SPC019
MBUS - SPC330

JFHCFC, 2023-09-05:
SUN2000MA V100R001C00SPC156 Release Documents

Optimized Features
1. Optimized the relay failure issue when PV strings are reversely connected in
scenarios with energy storage.
2. Optimized the upgrade function of SUN2000.
3. Optimized the issue that batteries may be exhausted when the SUN2000 is shut
down due to an optimizer fault and the battery SOC is set to a low value.

1.2.7b2 works well with SPC156