wlcrs/huawei_solar #427: [Feature Request]:  Include SOH Calibration Status

kaciker, 2023-07-23:
### Describe your feature request

The SOH Battery Check is rarely enforced, but when it is, there are no alerts in the app or on the Inverter that warn about the calibration being performed.
When this calibration is performed on the battery, it charges up to 100% and then proceeds to discharge down to 0%. During this time, the inverter does NOT generate power, and the only way to know what is happening is by checking the calibration status in the Huawei Solar application. If you are unaware of this process (as happened to me), you might think there is a malfunction. The inverter will not resume generating active power until the battery has completed its discharge

https://support.huawei.com/enterprise/en/doc/EDOC1100167258/91437c2/battery-soh-check

My improvement proposal is to add the "SOH (Status of Health) sensor" for the installed batteries. (Idle / Calibrating / Calibration completed)

[image]

Whith that sensor, we will be able to program some alarm in case of calibrating status

### Proper usage

- [X] I confirm that this is not a bug report or support request
- [X] I confirm that this feature request is within the stated scope of the integration

wlcrs, 2023-07-24:
Afaik this information is not exposed via Modbus and hence out of scope for this integration.

closed 2023-07-24 by wlcrs

kaciker, 2023-07-25:
Hello.
Colud be posible to add SOH %, and warnings  ?

[image]
[image]

wlcrs, 2023-07-25:
https://github.com/wlcrs/huawei_solar/wiki/Defining-custom-sensors

kaciker, 2023-07-25:
> https://github.com/wlcrs/huawei_solar/wiki/Defining-custom-sensors

I don't have enough knowledge of python to do this, ok, thanks anyway.

hvorragend, 2024-03-28:
By any chance, has anyone already integrated SOH as a custom sensor?