wlcrs/huawei_solar #429: [Bug]: Not a valid 'Inverter' device

petr5250, 2023-07-23:
### Describe the issue

Hi,
when using the Huawei Solar Integration with only one inverter connected, everything works fine and I can call the services to set up the Active power of that inverter, but when I add the second inverter (which is a separate device with no connection to the first one), then the problems occur. From that point onward I am unable to call services mentioned above on either inverter. Every time I try, it returns "Not a valid 'Inverter' device" error message. The only way to make it work again is to remove both Inverters and set up only one of them.

### Bescribe your Huawei Solar Setup

Inverter Type: SUN2000 15KTL-M2
Inverter Firmware version: V100R001C00SPC153
SDongle present: yes
Power meter present: three phase
Battery: None
Battery Firmware version: -
(both inverters are the same)

### How do you connect to the inverter?

Over serial, with a serial-to-USB stick

### Upload your Diagnostics File

[config_entry-huawei_solar-577a1a4d8a6dee5074a15ebe922ec202 - first inverter befor adding the second inverter.json.txt](https://github.com/wlcrs/huawei_solar/files/12139899/config_entry-huawei_solar-577a1a4d8a6dee5074a15ebe922ec202.-.first.inverter.befor.adding.the.second.inverter.json.txt)
[config_entry-huawei_solar-577a1a4d8a6dee5074a15ebe922ec202 - first inverter after adding the second inverter.json.txt](https://github.com/wlcrs/huawei_solar/files/12139900/config_entry-huawei_solar-577a1a4d8a6dee5074a15ebe922ec202.-.first.inverter.after.adding.the.second.inverter.json.txt)
[config_entry-huawei_solar-4d5c98bd51ed60a3a91bda1cb2458cf7 - second inverter.json.txt](https://github.com/wlcrs/huawei_solar/files/12139901/config_entry-huawei_solar-4d5c98bd51ed60a3a91bda1cb2458cf7.-.second.inverter.json.txt)

### Upload your relevant debug logs

```text
2023-07-23 22:17:14.461 ERROR (MainThread) [custom_components.huawei_solar.services] The provided device is not an inverter
2023-07-23 22:17:14.486 ERROR (MainThread) [homeassistant.helpers.script.websocket_api_script] websocket_api script: Error executing script. Unexpected error for call_service at pos 1: Not a valid 'Inverter' device
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 468, in _async_step
    await getattr(self, handler)()
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 703, in _async_call_service_step
    response_data = await self._async_run_long_action(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 665, in _async_run_long_action
    return long_task.result()
           ^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 1965, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2005, in _execute_service
    return await cast(
           ^^^^^^^^^^^
  File "/config/custom_components/huawei_solar/services.py", line 326, in set_di_active_power_scheduling
    bridge = get_inverter_bridge(service_call)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/huawei_solar/services.py", line 198, in get_inverter_bridge
    raise HuaweiSolarServiceException("Not a valid 'Inverter' device")
custom_components.huawei_solar.services.HuaweiSolarServiceException: Not a valid 'Inverter' device
2023-07-23 22:17:14.492 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [140283706946128] Error handling message: Unknown error (unknown_error) Admin from 109.224.97.1 (Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:109.0) Gecko/20100101 Firefox/115.0)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/decorators.py", line 26, in _handle_async_response
    await func(hass, connection, msg)
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 680, in handle_execute_script
    response = await script_obj.async_run(msg.get("variables"), context=context)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 1569, in async_run
    return await asyncio.shield(run.async_run())
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 420, in async_run
    await self._async_step(log_exceptions=False)
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 470, in _async_step
    self._handle_exception(
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 493, in _handle_exception
    raise exception
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 468, in _async_step
    await getattr(self, handler)()
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 703, in _async_call_service_step
[... 60 lines cut ...]
    response_data = await self._async_run_long_action(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 665, in _async_run_long_action
    return long_task.result()
           ^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 1965, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2005, in _execute_service
    return await cast(
           ^^^^^^^^^^^
  File "/config/custom_components/huawei_solar/services.py", line 326, in set_di_active_power_scheduling
    bridge = get_inverter_bridge(service_call)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/huawei_solar/services.py", line 198, in get_inverter_bridge
    raise HuaweiSolarServiceException("Not a valid 'Inverter' device")
custom_components.huawei_solar.services.HuaweiSolarServiceException: Not a valid 'Inverter' device
```

### Please confirm the following:

- [ ] The problem is still present in the latest release of this integration.
- [ ] I did not find an existing issue describing this problem.
- [ ] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [ ] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

wlcrs, 2023-07-25:
Can you please try again with 1.2.7 beta 1?

closed 2023-07-25 by wlcrs

Taraman17, 2023-09-18:
This also fixed my two converters always getting in each other's way on receiving values.
THX!

commit 1b39063e (2023-07-25): Fix services when integration is setup multiple times

Fixes #429