wlcrs/huawei_solar #431: [Bug]: HA Services, multiple with same name

ThomDietrich, 2023-07-26:
### Describe the issue

Hello, check the screenshot below. As you can see, three services are described by "Huawei Solar: Set Active Power Control to Unlimited mode", which I suspect is not intended.

[image]

### Bescribe your Huawei Solar Setup

Inverter Type: SUN2000-10KTL-M1-HC
Inverter Firmware version: latest
SDongle present: no
Power meter present: no
Battery: None
Battery Firmware version:

### How do you connect to the inverter?

Over serial, with a serial-to-WiFi stick

### Upload your Diagnostics File

[config_entry-huawei_solar-f61543400e1b6c1346d3dd0813ab9633.json.txt](https://github.com/wlcrs/huawei_solar/files/12171659/config_entry-huawei_solar-f61543400e1b6c1346d3dd0813ab9633.json.txt)

### Upload your relevant debug logs

```text
I hope no logs are required for this issue. Very happy to provide all the log data needed to resolve this issue though.
```

### Please confirm the following:

- [X] The problem is still present in the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [X] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [X] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

ThomDietrich, 2023-07-26:
Btw not fixed in 1.2.7b1

wlcrs, 2023-07-27:
Already fixed: https://github.com/wlcrs/huawei_solar/commit/1b39063e9e37395d278332bd63ad6cf77d084d21#diff-476759a1d4ca0225d35e990ad3c57a4a2d5193563a9b764dee3044c6aeecfc0c

closed 2023-07-27 by wlcrs

ThomDietrich, 2023-07-27:
Maybe I need to restart the system for this to take effect. Thanks!!