wlcrs/huawei_solar #433: [Bug]: Have lost some device data, related to TOU Working Mode

NickWhitNTES, 2023-07-27:
### Describe the issue

[image]
I have lost since latest upgrade the ability to set these settings.

### Bescribe your Huawei Solar Setup

Inverter Type: SUN2000-6KTL-L1
Inverter Firmware version: V200R001C00SPC130 
SDongle present: yes 
Power meter present: single phase 
Battery: LUNA2000 15KW
Battery Firmware version: V100R002C00SPC11

### How do you connect to the inverter?

Over serial, with a serial-to-USB stick

### Upload your Diagnostics File

Drag & Drop your Diagnostics File here.
[config_entry-huawei_solar-1e8173dbb7b3a552ec66ae59d2dd6e72.json(1).txt](https://github.com/wlcrs/huawei_solar/files/12187521/config_entry-huawei_solar-1e8173dbb7b3a552ec66ae59d2dd6e72.json.1.txt)

### Upload your relevant debug logs

```text
NA
```

### Please confirm the following:

- [X] The problem is still present in the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [X] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [X] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

NickWhitNTES, 2023-08-05:
Is this related to bug #426

wlcrs, 2023-08-11:
no logs = nothing for me to work with

closed 2023-08-11 by wlcrs