wlcrs/huawei_solar #435: [Bug?]: Data sometimes not valid

t0bse, 2023-07-31:
### Describe the issue

i am runnig this integration for a while now, since a few days we got bad weather here in germany, so sometimes i have the sun changing in one minute for a few times.

now i noticed, that the sensors are wrong sometimes. (i think it's only happening if the clouds are coming/going) 

see screenshots, all values have the same timestamp

inverter_active_power:
[image]

power_meter_active_power:
[image]

my sensor calculates "inverter minus power_meter active power" and so it says
[image]

i can also see this the other way:
inverter_active_power:
[image]
power_meter_active_power:
[image]
so my sensor:
[image]

### Bescribe your Huawei Solar Setup

Inverter Type: SUN2000-10KTL-M0
Inverter Firmware version: V100R001C00SPC153
SDongle present: yes 
Power meter present: three phase 
Battery: None
Battery Firmware version:

### How do you connect to the inverter?

Via the SDongle, wired connection

### Upload your Diagnostics File

{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.7.3",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.11.4",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Berlin",
    "os_name": "Linux",
    "os_version": "6.1.34",
    "supervisor": "2023.07.1",
    "host_os": "Home Assistant OS 10.3",
    "docker_version": "23.0.6",
    "chassis": "vm",
    "run_as_root": true
  },
  "custom_components": {
    "rewe": {
      "version": "2023.06.0",
      "requirements": [
        "cloudscraper==1.2.71"
      ]
    },
    "rika_firenet": {
      "version": "0.0.1",
      "requirements": [
        "bs4",
        "requests"
      ]
    },
    "alexa_media": {
      "version": "4.6.5",
      "requirements": [
        "alexapy==1.26.8",
        "packaging>=20.3",
        "wrapt>=1.12.1"
      ]
    },
    "huawei_solar": {
      "version": "1.2.6",
      "requirements": [
        "huawei-solar==2.2.8"
      ]
    },
    "hacs": {
      "version": "1.32.1",
      "requirements": [
        "aiogithubapi>=22.10.1"
      ]
    },
    "oilfox": {
      "version": "0.1.10",
      "requirements": [
        "aiohttp"
      ]
    },
    "xiaomi_cloud_map_extractor": {
      "version": "v2.2.0",
      "requirements": [
        "pillow",
        "pybase64",
        "python-miio",
        "requests",
        "pycryptodome"
      ]
    },
    "gardena_smart_system": {
      "version": "1.0.0",
      "requirements": [
        "py-smart-gardena==1.3.7"
      ]
    },
    "dwains_dashboard": {
      "version": "3.4.3",
      "requirements": []
    },
    "pyscript": {
      "version": "1.4.0",
      "requirements": [
        "croniter==1.3.8",
        "watchdog==2.2.1"
      ]
    },
    "fontawesome": {
      "version": "2.1.5",
      "requirements": []
    },
    "fordpass": {
      "version": "0.1.48",
      "requirements": []
    },
    "enigma": {
      "version": "1.4",
      "requirements": [
        "beautifulsoup4==4.6.3"
      ]
    }
  },
  "integration_manifest": {
    "domain": "huawei_solar",
    "name": "Huawei Solar",
    "config_flow": true,
    "documentation": "https://github.com/wlcrs/huawei_solar/wiki",
    "issue_tracker": "https://github.com/wlcrs/huawei_solar/issues",
    "requirements": [
      "huawei-solar==2.2.8"
    ],
    "codeowners": [
      "@wlcrs"
    ],
    "iot_class": "local_polling",
    "version": "1.2.6",
    "loggers": [
      "huawei_solar",
      "pymodbus"
    ],
    "is_built_in": false
  },
  "data": {
    "config_entry_data": {
      "host": "192.168.1.60",
      "port": 502,
      "slave_ids": [
        1
      ],
      "enable_parameter_configuration": false,
      "username": null,
      "password": null
    },
    "slave_0": {
      "model_name": "SUN2000-10KTL-M1\u0000\u0000\u0000\u0000SUN2000-12",
      "pv_string_count": 2,
      "has_optimizers": 0,
      "battery_type": 0,
      "battery_1_type": 0,
      "battery_2_type": 0,
      "power_meter_type": 1,
      "supports_capacity_control": false
    },
    "slave_0_data": {
      "input_power": [
        0,
        "W"
      ],
      "line_voltage_A_B": [
        0.0,
        "V"
      ],
      "line_voltage_B_C": [
        0.0,
        "V"
      ],
      "line_voltage_C_A": [
        0.0,
        "V"
      ],
      "phase_A_voltage": [
        0.0,
        "V"
      ],
      "phase_B_voltage": [
        0.0,
        "V"
      ],
      "phase_C_voltage": [
        0.0,
        "V"
      ],
      "phase_A_current": [
        0.0,
        "A"
      ],
      "phase_B_current": [
        0.0,
        "A"
      ],
      "phase_C_current": [
        0.0,
        "A"
      ],
      "day_active_power_peak": [
        11000,
        "W"
      ],
      "active_power": [
        0,
        "W"
      ],
      "reactive_power": [
        0,
        "VA"
      ],
      "power_factor": [
        0.0,
        null
      ],
      "grid_frequency": [
        0.0,
        "Hz"
      ],
      "efficiency": [
        null,
        "%"
      ],
      "internal_temperature": [
        0.0,
        "\u00b0C"
      ],
      "insulation_resistance": [
        30.0,
        "MOhm"
      ],
      "device_status": [
        "Standby: no irradiation",
        null
      ],
      "fault_code": [
        0,
        null
      ],
      "startup_time": [
        "2023-07-31T04:59:53+00:00",
        null
      ],
      "shutdown_time": [
        "2023-07-31T19:48:51+00:00",
        null
      ],
      "accumulated_yield_energy": [
        843.36,
        "kWh"
      ],
      "daily_yield_energy": [
        32.23,
        "kWh"
      ],
      "state_1": [
        [
          "Standby"
        ],
        null
      ],
      "state_2": [
        [
          "Locked",
          "PV disconnected",
          "DSP data collection"
        ],
        null
      ],
      "state_3": [
        [
          "On-grid",
          "Off-grid switch disabled"
        ],
        null
      ],
      "alarm_1": [
        [],
        null
      ],
      "alarm_2": [
        [],
        null
      ],
      "alarm_3": [
        [],
        null
      ],
      "pv_01_voltage": [
        0.0,
        "V"
      ],
      "pv_01_current": [
        0.0,
        "A"
      ],
      "pv_02_voltage": [
        0.0,
        "V"
      ],
      "pv_02_current": [
        0.0,
        "A"
      ],
      "meter_status": [
        1,
        null
      ],
      "grid_A_voltage": [
        235.5,
        "V"
      ],
      "grid_B_voltage": [
        236.3,
        "V"
      ],
      "grid_C_voltage": [
        235.1,
        "V"
      ],
      "active_grid_A_current": [
        -0.74,
        "I"
      ],
      "active_grid_B_current": [
        -0.99,
        "I"
      ],
      "active_grid_C_current": [
        -0.6,
        "I"
      ],
      "power_meter_active_power": [
        -343,
        "W"
      ],
      "power_meter_reactive_power": [
        337,
        "Var"
      ],
      "active_grid_power_factor": [
        0.62,
        null
      ],
      "active_grid_frequency": [
        49.99,
        "Hz"
      ],
      "grid_exported_energy": [
        191.82,
        "kWh"
      ],
      "grid_accumulated_energy": [
        335.59,
        "kWh"
      ],
      "grid_accumulated_reactive_power": [
        0.0,
        "kVarh"
      ],
      "meter_type": [
        1,
        null
      ],
      "active_grid_A_B_voltage": [
        408.6,
        "V"
      ],
      "active_grid_B_C_voltage": [
        408.2,
        "V"
      ],
      "active_grid_C_A_voltage": [
        407.5,
        "V"
      ],
      "active_grid_A_power": [
        -62,
        "W"
      ],
      "active_grid_B_power": [
        -214,
        "W"
      ],
      "active_grid_C_power": [
        -66,
        "W"
      ]
    }
  }
}

### Upload your relevant debug logs

```text
-
```

### Please confirm the following:

- [X] The problem is still present in the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [X] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [X] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

jires, 2023-07-31:
I'm experiencing same issues.
I have figured out, that sometimes Integration has problems to read all registers and retrying. Those registers which are not read at the first time, cannot be updated together with others registers on pooling time, so there can be 2 options. Either they will be "unavailable" until they get new reading, or they will keep current value. Thankfully to newly changes, they keeping current value. Unfortunately, it can be a situation, where different registers values are "shifted" in time for some seconds, therefore is mismatch in values when they changing very fast.

I think this is good trade off, what we get from integration and what inverter/usb dongle is capable of deliver. 
Only way to avoid this is to turn off completely all those registers which are not used/ not essentials, and make less load on Modbus module in the inverter.

wlcrs, 2023-08-01:
Thank you @jires for the fine explanation.

This integration just exposes what it can read from the inverter. I batch-read as much as possible to get consistent values. If the values that you see are strange, it is the fault of the inverter, not of the integration.

closed 2023-08-01 by wlcrs