wlcrs/huawei_solar #438: [Feature Request]: Quicker entity update via service call

ThomDietrich, 2023-08-02:
### Describe your feature request

Hey there,
thanks for your integration! My setup: SUN2000-10KTL-M1-HC via Protoss PE11-H.

As documented in your README, the integration retrieved values periodically every 30 seconds. That's generally fine but I need a few values more frequently.

After a bit of testing I have questions, and the hope that we can either improve current functionality or the README.

1. Why can't I go below a polling rate of 10 seconds? Is that a limitation of the inverter itself?
1. Is there a way to define an automation for ALL entities, rather than specifying individual entity_ids (e.g. When I want all secondary entities to update every 60s)?

### Proper usage

- [X] I confirm that this is not a bug report or support request
- [X] I confirm that this feature request is within the stated scope of the integration

wlcrs, 2023-08-11:
This is a bug tracker, not a support channel

closed 2023-08-11 by wlcrs

ThomDietrich, 2023-08-11:
Hey, you might be mistaken. 

1. is a feature request,
2. is a feature request,
3. is a bug report,
4. is a question, agreed.

Please have a look at number 3 at least. Thanks you

cgernert, 2023-08-11:
As far as I know, this is because of the SDongle
https://skyboo.net/2022/02/huawei-sun2000-why-using-a-usb-dongle-for-monitoring-is-not-a-good-idea/