wlcrs/huawei_solar #443: Feature Request: Service to enable/disable MPPT multi peak scanning

wlcrs, 2023-08-11:
### Discussed in https://github.com/wlcrs/huawei_solar/discussions/439

<div type='discussions-op-text'>

<sup>Originally posted by **michi1984** August  7, 2023</sup>
Hi,

I would like to be able to enable and disable the MPPT multi peak scanning of my Huawei inverter via a service. This would be to optimise the use of the global peak MPP scan in scenarios where the PV-array is affected by shading only during 1-2 hours a day, depending on the season. Disabling the scan during the rest of the day where there is no shading could potentially slighly increase electricitiy production, because the energy loss during scanning is avoided. 

Currently this is only possible via the fusionsolar portal or the app (with direct wifi connection to the inverter), which is not really feasible to do. I would like to set up an automation which depending on the current situation switches the scan on and off.

This can be done via Modbus:
Signal Name: MPPT multimodal scanning 
Read and write: RW
Type: E16
Address: 42054
Number of Registers: 1

0 is off, 1 is on. 
</div>

michi1984, 2023-09-06:
Just tested reading the mppt registers, seems to work. Will have to check predicted power once sun is up.

michi1984, 2023-09-08:
Predicted power is always at 0. At least with my 30KTL M3. Mppt on/Off sind Intervall schön corrxtly and responding to changes made in Fusionsolar

michi1984, 2023-09-08:
Perhaps predicted Power only works if a solar radiation and module temperature sensor is connected to calculate a theoretical power expectation.

wlcrs, 2023-10-09:
I've decided to not add these registers to the integration as they would require an extra read-request during every update cycle of the configuration entities. I'm trying to balance the number of modbus address ranges being read on each update with their utility. Currently I'm already getting quite some issues from people of which the inverter answers too slowly, especially when multiple inverters are stringed together. Adding a new range that needs to be read would exacerbate those issues.

As you already seem to be using those newly added registers: can you share the improvement that you were able to get by manually enabling/disabling the MPPT scanning based on your intuition? If there is a measurable and significant difference I might reconsider.

closed 2023-11-19 by github-actions[bot]