wlcrs/huawei_solar #450: [Bug]: Connection completely fails and only resumes after integration reload

bogyxp, 2023-08-24:
### Describe the issue

Hello! I have a new Huawei Luna 3000 inverter SUN2000-3KTL-L1. I have a twisted pair coming into a 485 to Usb adapter as the one in the guide.  
The second day I have managed to get it in HA and it worked for a while.
My problem is that after some time, the connection breaks, everything becomes unavailable and it only recovers after an integration restart. This is for both inverter and smart meter.
I made a mistake at the beginning by deactivating the plugins 30 second polling and calling an update every 10 seconds via an automation. I have since removed that and decided i will leave it to the default 30s and it worked for a day, then failed again.

### Bescribe your Huawei Solar Setup

Inverter Type:  SUN2000-3KTL-L1
Inverter Firmware version: V200R001C00SPC130
SDongle present: yes (but the one that is reported to work, not SDongleA-05)
Power meter present: single phase 
Battery: LG RESU  None
Battery Firmware version: none

### How do you connect to the inverter?

Over serial, with a serial-to-USB stick

### Upload your Diagnostics File

Drag & Drop your Diagnostics File here.
[redacted.txt](https://github.com/wlcrs/huawei_solar/files/12429966/redacted.txt)

### Upload your relevant debug logs

```text
2023-08-23 12:08:41.299 INFO (MainThread) [backoff] Backing off _do_read(...) for 2.0s (huawei_solar.exceptions.ConnectionInterruptedException: Modbus client is not connected to the inverter.)
2023-08-23 12:08:41.299 DEBUG (MainThread) [huawei_solar.huawei_solar] Received <class 'huawei_solar.exceptions.ConnectionInterruptedException'>: backing off reading for 2.0 seconds after 2 tries
2023-08-23 12:08:41.517 DEBUG (MainThread) [pymodbus.logging] recv: 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff
2023-08-23 12:08:41.518 DEBUG (MainThread) [pymodbus.logging] recv: 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff
2023-08-23 12:08:41.519 DEBUG (MainThread) [pymodbus.logging] Frame - [b'\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff'] not ready
2023-08-23 12:08:42.028 DEBUG (MainThread) [pymodbus.logging] recv: 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff
2023-08-23 12:08:42.028 DEBUG (MainThread) [pymodbus.logging] recv: 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff
2023-08-23 12:08:42.029 DEBUG (MainThread) [pymodbus.logging] Frame - [b'\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff'] not ready
2023-08-23 12:08:42.397 DEBUG (MainThread) [pymodbus.logging] recv: 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff
2023-08-23 12:08:42.398 DEBUG (MainThread) [pymodbus.logging] recv: 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff
2023-08-23 12:08:42.399 DEBUG (MainThread) [pymodbus.logging] Frame - [b'\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff'] not ready
2023-08-23 12:08:42.727 DEBUG (MainThread) [pymodbus.logging] recv: 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff
2023-08-23 12:08:42.728 DEBUG (MainThread) [pymodbus.logging] recv: 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff
2023-08-23 12:08:42.729 DEBUG (MainThread) [pymodbus.logging] Frame - [b'\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff'] not ready
2023-08-23 12:08:43.247 DEBUG (MainThread) [pymodbus.logging] recv: 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff
2023-08-23 12:08:43.248 DEBUG (MainThread) [pymodbus.logging] recv: 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff
2023-08-23 12:08:43.249 DEBUG (MainThread) [pymodbus.logging] Frame - [b'\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff'] not ready
2023-08-23 12:08:43.300 ERROR (MainThread) [huawei_solar.huawei_solar] Modbus client is not connected to the inverter.
NoneType: None
2023-08-23 12:08:43.300 INFO (MainThread) [backoff] Backing off _do_read(...) for 4.0s (huawei_solar.exceptions.ConnectionInterruptedException: Modbus client is not connected to the inverter.)
2023-08-23 12:08:43.301 DEBUG (MainThread) [huawei_solar.huawei_solar] Received <class 'huawei_solar.exceptions.ConnectionInterruptedException'>: reconnecting and backing off reading for 4.0 seconds after 3 tries
2023-08-23 12:08:43.301 DEBUG (MainThread) [pymodbus.logging] Connecting to None.
2023-08-23 12:08:43.301 DEBUG (MainThread) [pymodbus.logging] Connecting comm
2023-08-23 12:08:43.304 DEBUG (MainThread) [pymodbus.logging] Connected to comm
2023-08-23 12:08:43.304 DEBUG (MainThread) [huawei_solar.modbus] Waiting for 1500 milliseconds after connection before performing operations
2023-08-23 12:08:43.548 DEBUG (MainThread) [pymodbus.logging] recv: 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff
2023-08-23 12:08:43.548 DEBUG (MainThread) [pymodbus.logging] recv: 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff
2023-08-23 12:08:43.548 DEBUG (MainThread) [pymodbus.logging] Frame - [b'\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff'] not ready
2023-08-23 12:08:44.028 DEBUG (MainThread) [pymodbus.logging] recv: 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff
2023-08-23 12:08:44.028 DEBUG (MainThread) [pymodbus.logging] recv: 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff
2023-08-23 12:08:44.028 DEBUG (MainThread) [pymodbus.logging] Frame - [b'\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff'] not ready
2023-08-23 12:08:44.417 DEBUG (MainThread) [pymodbus.logging] recv: 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff
2023-08-23 12:08:44.418 DEBUG (MainThread) [pymodbus.logging] recv: 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff
2023-08-23 12:08:44.418 DEBUG (MainThread) [pymodbus.logging] Frame - [b'\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff'] not ready
2023-08-23 12:08:44.908 DEBUG (MainThread) [pymodbus.logging] recv: 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff
2023-08-23 12:08:44.908 DEBUG (MainThread) [pymodbus.logging] recv: 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff
2023-08-23 12:08:44.908 DEBUG (MainThread) [pymodbus.logging] Frame - [b'\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff'] not ready
2023-08-23 12:08:45.208 DEBUG (MainThread) [pymodbus.logging] recv: 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff
2023-08-23 12:08:45.208 DEBUG (MainThread) [pymodbus.logging] recv: 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff
2023-08-23 12:08:45.208 DEBUG (MainThread) [pymodbus.logging] Frame - [b'\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff'] not ready
2023-08-23 12:08:45.608 DEBUG (MainThread) [pymodbus.logging] recv: 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff
2023-08-23 12:08:45.608 DEBUG (MainThread) [pymodbus.logging] recv: 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff
[... 227 lines cut ...]
2023-08-23 12:09:10.301 INFO (MainThread) [backoff] Backing off _do_read(...) for 1.0s (TimeoutError)
2023-08-23 12:09:10.302 DEBUG (MainThread) [huawei_solar.huawei_solar] Received <class 'TimeoutError'>: backing off reading for 1.0 seconds after 1 tries
2023-08-23 12:09:10.303 DEBUG (MainThread) [pymodbus.logging] Connection lost comm due to None
2023-08-23 12:09:10.303 DEBUG (MainThread) [pymodbus.logging] Getting transaction 1
2023-08-23 12:09:10.458 DEBUG (MainThread) [pymodbus.logging] recv: 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff
2023-08-23 12:09:10.459 DEBUG (MainThread) [pymodbus.logging] recv: 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff
2023-08-23 12:09:10.460 DEBUG (MainThread) [pymodbus.logging] Frame - [b'\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff'] not ready
[... cut ...]
2023-08-23 12:09:11.304 DEBUG (MainThread) [huawei_solar.huawei_solar] Received <class 'huawei_solar.exceptions.ConnectionInterruptedException'>: backing off reading for 2.0 seconds after 2 tries
```

### Please confirm the following:

- [X] The problem is still present in the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [ ] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [X] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

bogyxp, 2023-08-24:
Here is today`s log. for some reason all HA logs are in the file.
[24.08logs.txt](https://github.com/wlcrs/huawei_solar/files/12432475/24.08logs.txt)

renatod47, 2023-08-25:
https://github.com/wlcrs/huawei_solar/issues/447
Integration doesn't even start

bogyxp, 2023-08-27:
After updating HA to latest OS and latest hypervisor and a complete restart of the server it seems to be consistent now. It has been running for the past 2 days.
I have also removed most of the elements except the important ones.
All that`s available now is: 
From  inverter: 
- Active power
- Daily yield
- Internal temperature
- PV 1 Current
- PV 1 Voltage
- Device status
from Smart Meter:
- Active power
- Consumption
- Current
- Exported
- Voltage
Everything is updated by the integration, every 30s and i still get the occasional Unavailable but overall it`s working again.

renatod47, 2023-08-28:
Hi @bogyxp, does "hypervisor" stand for the HA supervisor? 
 Is there any logical explanation for you to remove most of the elements? 
 I updated everything for the last version and it still won't run so I'm trying to understand a little bit more.

ivanuski9, 2023-08-30:
Same problem. Some more detail to fix it.

bogyxp, 2023-08-31:
> Hi @bogyxp, Is there any logical explanation for you to remove most of the elements? I updated everything for the last version and it still won't run so I'm trying to understand a little bit more.

I just assumed that based on the enabled parameters, they will be requested from the inverter. i only use those and don`t need the extra overhead of communication with the inverter. 

I can see the communication led of the inverter blinking when communication is ongoing, and every 30s it blinks constantly for about 15-20 seconds. 

> Same problem. Some more detail to fix it.
Honestly, just updated HA to latest version, and removed my automation for updating the sensors from the inverter as it was done every 5 seconds. I now rely on the default polling rate of 30 seconds offered by the integration.  

 Have not had an issue since about a week now. There is the odd unavailable every now and then, but overall it`s quite  good.

ivanuski9, 2023-09-01:
In my case, the other day I solved it by reconnecting the Wifi connection through which I reach the inverter. But I have observed that the ping takes a long time (more than 2 seconds) and perhaps the problem is due to timeouts or an excessive number of requests. The problems happened to me when updating the OS version to 10.5. Maybe there is something unstable in the Wifi connection and the connection is lost and it is the root of the problem.

closed 2023-09-04 by wlcrs

bogyxp, 2023-09-06:
@wlcrs  can you please provide in the documentation the means to restart the integration in case it fails? For literally no reason it started failing again every so often.
Just pasting this here as it might help the others.

shell_command:
  reset_huawei_integration: '/usr/bin/curl -X POST -H "Authorization: Bearer <YOUR_TOKEN>" -H "Content-Type: application/json" http://192.168.0.10:8123/api/config/config_entries/entry/<integration id>/reload'
your token  - generate a long acces stoken by clicking on your name in left bottom menu and scrolling way down
integration id - something like 6e217723901fe9031b4c03ef68333359(redacted) can be found by getting to the devices that are part of the integration and copying it from the address bar. (settings - devices and services - Huawei solar -  2 devices)
Make sure to escape every special character (change " to \" until vscode stops complaining) 

You have to restart HA to get the shell command to appear in automations.

automation 
alias: Autorestart HUAWEI SOLAR
description: ""
trigger:
  - platform: state
    entity_id:
      - sensor.inverter_active_power
      - sensor.power_meter_active_power
    to: unavailable
    for:
      hours: 0
      minutes: 2
      seconds: 0
condition: []
action:
  - service: shell_command.reset_huawei_integration
    data: {}
  - service: notify.mobile_phone(redacted, optional)
    data:
      title: INFO
      message: Huawei integration failed. Restarting
mode: single