wlcrs/huawei_solar #463: [Bug]: Not initialized after network connection was disconnected

powerscooter, 2023-09-09:
### Describe the issue

If the network connection between HomeAssistant and the Huawei Solar SDongle is disconnected during operation, you must reinitialize the integration in HomeAssistant so that a connection to the SDongle is established again. Otherwise, there is always the error "not initialized".

### Bescribe your Huawei Solar Setup

SDongle present: yes

### How do you connect to the inverter?

Via the SDongle, wired connection

### Upload your Diagnostics File

Drag & Drop your Diagnostics File here.

### Upload your relevant debug logs

```text
No
```

### Please confirm the following:

- [X] The problem is still present in the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [X] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [X] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

powerscooter, 2023-09-09:
I think adding a function that automatically reinitializes the integration in such a case would solve the problem

wlcrs, 2023-09-10:
It already has such functionality on multiple levels. If you don't provide debug logs I cannot  diagnose the problem

powerscooter, 2023-09-10:
[home-assistant_huawei_solar_2023-09-10T07-41-38.697Z.log](https://github.com/wlcrs/huawei_solar/files/12567741/home-assistant_huawei_solar_2023-09-10T07-41-38.697Z.log)

Ok sorry, here is the debug file :-)

wlcrs, 2023-09-10:
At what time does the issue manifest itself? The debuglog is quite big and I don't see anything relevant after reading through it a first time

powerscooter, 2023-09-10:
[home-assistant_huawei_solar_2023-09-10T14-14-59.965Z.log](https://github.com/wlcrs/huawei_solar/files/12568404/home-assistant_huawei_solar_2023-09-10T14-14-59.965Z.log)

Hi i hope that one is better.
From line 1 to ~464 was the inverter not reachable.

After that line the inverter was reachable again and the integration was able to connect to the device

wlcrs, 2023-09-10:
So the integration automatically reconnected once the inverter came back online? This is the behaviour you were requesting, right? What am I missing?

powerscooter, 2023-09-10:
No, that is exactly the problem. The integration does not automatically reconnect.
I always have to manually reload the integration

[image]

Now the integration works again until there is a network failure.

40federico, 2023-09-11:
In my case, after a network failure, the reload is not working. Attached log
This error originated from a custom integration.

Logger: homeassistant.config_entries
Source: custom_components/huawei_solar/__init__.py:87
Integration: Huawei Solar (documentation, issues)
First occurred: 08:47:48 (9 occurrences)
Last logged: 11:37:22

Error setting up entry SUN2000-6KTL-L1 for huawei_solar
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/asyncio/tasks.py", line 490, in wait_for
    return fut.result()
           ^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/asyncio/locks.py", line 213, in wait
    await fut
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 178, in create
    await huawei_solar._initialize()
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 107, in _initialize
    self.time_zone = (await self.get(rn.TIME_ZONE)).value
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 241, in get
    return (await self.get_multiple([name], slave))[0]
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 276, in get_multiple
    response = await self._read_registers(registers[0].register, total_length, slave)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 396, in _read_registers
    async with self._communication_lock():
  File "/usr/local/lib/python3.11/contextlib.py", line 204, in __aenter__
    return await anext(self.gen)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 145, in _communication_lock
    raise err
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 141, in _communication_lock
    await asyncio.wait_for(self._client.connected_event.wait(), WAIT_FOR_CONNECTION_TIMEOUT)
  File "/usr/local/lib/python3.11/asyncio/tasks.py", line 492, in wait_for
    raise exceptions.TimeoutError() from exc
TimeoutError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 387, in async_setup
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/huawei_solar/__init__.py", line 199, in async_setup_entry
    raise err
  File "/config/custom_components/huawei_solar/__init__.py", line 87, in async_setup_entry
    primary_bridge = await HuaweiSolarBridge.create(
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/bridge.py", line 66, in create
    client = await AsyncHuaweiSolar.create(host, port, slave_id)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 187, in create
    await client.close()
TypeError: object NoneType can't be used in 'await' expression

wlcrs, 2023-09-13:
Can you please test again with 1.2.7? I don't expect it to be solved, but I fixed the error that I saw popping up in the logs. That error was probably obscuring the real underlying issue. Thanks in advance!

powerscooter, 2023-09-16:
Hey, I've been testing it for a few days now with the new version and it seems to be working fine now!
Thanks for your quick help and reply

closed 2023-09-16 by wlcrs