wlcrs/huawei_solar #465: [Bug]: Since update Home Assistant 2023.9.1 the Integration is broken

kpat69, 2023-09-10:
### Describe the issue

Hello, since the update to 2023.9.1 all my huawei sensors are unavailable.
With 2023.9 all was o.k..
The connection to the Dongle is o.k. and Fusion Solar works all so.

### Bescribe your Huawei Solar Setup

Inverter Type: SUN2000-10KTL-M1
Inverter Firmware version: V100R001C00SPC153
SDongle present: yes
Power meter present: no
Battery: None

### How do you connect to the inverter?

Via the SDongle, wired connection

### Upload your Diagnostics File

Drag & Drop your Diagnostics File here.
[home-assistant_huawei_solar_2023-09-10T08-00-39.251Z.zip](https://github.com/wlcrs/huawei_solar/files/12567777/home-assistant_huawei_solar_2023-09-10T08-00-39.251Z.zip)

### Upload your relevant debug logs

```text
2023-09-10 09:49:57.125 ERROR (MainThread) [huawei_solar.huawei_solar] Aborting client creation due to error.
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 178, in create
    await huawei_solar._initialize()
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 107, in _initialize
    self.time_zone = (await self.get(rn.TIME_ZONE)).value
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 241, in get
    return (await self.get_multiple([name], slave))[0]
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 278, in get_multiple
    decoder = BinaryPayloadDecoder.fromRegisters(response.registers, byteorder=Endian.Big, wordorder=Endian.Big)
                                                                               ^^^^^^^^^^
  File "/usr/local/lib/python3.11/enum.py", line 784, in __getattr__
    raise AttributeError(name) from None
AttributeError: Big
2023-09-10 09:49:57.133 ERROR (MainThread) [homeassistant.config_entries] Error setting up entry SUN2000-10KTL-M1    SUN2000-12 for huawei_solar
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 178, in create
    await huawei_solar._initialize()
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 107, in _initialize
    self.time_zone = (await self.get(rn.TIME_ZONE)).value
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 241, in get
    return (await self.get_multiple([name], slave))[0]
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 278, in get_multiple
    decoder = BinaryPayloadDecoder.fromRegisters(response.registers, byteorder=Endian.Big, wordorder=Endian.Big)
                                                                               ^^^^^^^^^^
  File "/usr/local/lib/python3.11/enum.py", line 784, in __getattr__
    raise AttributeError(name) from None
AttributeError: Big

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 387, in async_setup
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/huawei_solar/__init__.py", line 199, in async_setup_entry
    raise err
  File "/config/custom_components/huawei_solar/__init__.py", line 87, in async_setup_entry
    primary_bridge = await HuaweiSolarBridge.create(
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/bridge.py", line 66, in create
    client = await AsyncHuaweiSolar.create(host, port, slave_id)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 187, in create
    await client.close()
TypeError: object NoneType can't be used in 'await' expression
2023-09-10 09:49:57.149 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event system_log_event[L]: name=huawei_solar.huawei_solar, message=['Aborting client creation due to error.'], level=ERROR, source=('/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py', 184), timestamp=1694332197.1255002, exception=Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 178, in create
    await huawei_solar._initialize()
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 107, in _initialize
    self.time_zone = (await self.get(rn.TIME_ZONE)).value
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 241, in get
    return (await self.get_multiple([name], slave))[0]
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 278, in get_multiple
    decoder = BinaryPayloadDecoder.fromRegisters(response.registers, byteorder=Endian.Big, wordorder=Endian.Big)
                                                                               ^^^^^^^^^^
  File "/usr/local/lib/python3.11/enum.py", line 784, in __getattr__
    raise AttributeError(name) from None
AttributeError: Big
, count=1, first_occurred=1694332197.1255002>
2023-09-10 09:49:57.150 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event system_log_event[L]: name=homeassistant.config_entries, message=['Error setting up entry SUN2000-10KTL-M1\x00\x00\x00\x00SUN2000-12 for huawei_solar'], level=ERROR, source=('custom_components/huawei_solar/__init__.py', 87), timestamp=1694332197.133585, exception=Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 178, in create
    await huawei_solar._initialize()
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 107, in _initialize
    self.time_zone = (await self.get(rn.TIME_ZONE)).value
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 241, in get
    return (await self.get_multiple([name], slave))[0]
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 278, in get_multiple
    decoder = BinaryPayloadDecoder.fromRegisters(response.registers, byteorder=Endian.Big, wordorder=Endian.Big)
                                                                               ^^^^^^^^^^
  File "/usr/local/lib/python3.11/enum.py", line 784, in __getattr__
    raise AttributeError(name) from None
AttributeError: Big

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 387, in async_setup
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/huawei_solar/__init__.py", line 199, in async_setup_entry
    raise err
  File "/config/custom_components/huawei_solar/__init__.py", line 87, in async_setup_entry
    primary_bridge = await HuaweiSolarBridge.create(
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/bridge.py", line 66, in create
    client = await AsyncHuaweiSolar.create(host, port, slave_id)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 187, in create
    await client.close()
TypeError: object NoneType can't be used in 'await' expression
, count=1, first_occurred=1694332197.133585>
```

### Please confirm the following:

- [X] The problem is still present in the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [X] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [X] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

arva-zz, 2023-09-10:
Same here:
`Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 178, in create
    await huawei_solar._initialize()
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 107, in _initialize
    self.time_zone = (await self.get(rn.TIME_ZONE)).value
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 241, in get
    return (await self.get_multiple([name], slave))[0]
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 278, in get_multiple
    decoder = BinaryPayloadDecoder.fromRegisters(response.registers, byteorder=Endian.Big, wordorder=Endian.Big)
                                                                               ^^^^^^^^^^
  File "/usr/local/lib/python3.11/enum.py", line 784, in __getattr__
    raise AttributeError(name) from None
AttributeError: Big

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 387, in async_setup
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/huawei_solar/__init__.py", line 199, in async_setup_entry
    raise err
  File "/config/custom_components/huawei_solar/__init__.py", line 87, in async_setup_entry
    primary_bridge = await HuaweiSolarBridge.create(
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/bridge.py", line 66, in create
    client = await AsyncHuaweiSolar.create(host, port, slave_id)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 187, in create
    await client.close()
TypeError: object NoneType can't be used in 'await' expression
`

VaabenDK, 2023-09-10:
Ditto issue here

josepmendoza, 2023-09-10:
I’ve this isue on 1.2.6 and 1.2.7b1. Update to 1.2.7b2 works fine and fixes.

martinarva, 2023-09-10:
> I have this issue on 1.2.6 and 1.2.7b1. Update to 1.2.7b2 works fine and fixes.

Can confirm, 1.2.7b2 solves the issue

b3hni4, 2023-09-10:
I have the same issue also on 1.2.6. Update to 1.2.7b2 fixed it.

kpat69, 2023-09-11:
Can confirm too, 1.2.7b2 solves the issue.
thank you!

closed 2023-09-11 by kpat69