wlcrs/huawei_solar #467: [Bug]: error in huawei_solar.py after update to 2023.91

AnKloss, 2023-09-12:
### Describe the issue

Logger: huawei_solar.huawei_solar
Source: /usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py:184 
First occurred: 8:24:06 PM (1 occurrences) 
Last logged: 8:24:06 PM

Aborting client creation due to error.
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 178, in create
    await huawei_solar._initialize()
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 107, in _initialize
    self.time_zone = (await self.get(rn.TIME_ZONE)).value
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 241, in get
    return (await self.get_multiple([name], slave))[0]
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 278, in get_multiple
    decoder = BinaryPayloadDecoder.fromRegisters(response.registers, byteorder=Endian.Big, wordorder=Endian.Big)
                                                                               ^^^^^^^^^^
  File "/usr/local/lib/python3.11/enum.py", line 784, in __getattr__
    raise AttributeError(name) from None
AttributeError: Big

### Bescribe your Huawei Solar Setup

Inverter Type: SUN2000L-5KTL-CN-4G
Inverter Firmware version:
SDongle present: no
Power meter present: single phase
Battery:  None
Battery Firmware version:

### How do you connect to the inverter?

Over serial, with a serial-to-WiFi stick

### Upload your Diagnostics File

Drag & Drop your Diagnostics File here.

### Upload your relevant debug logs

```text
272829303132333435363738394041424344458910111213141516171819202122232425261234567
2023-09-12 20:36:42.263 DEBUG (SyncWorker_6) [pymodbus.logging] Factory Response[ReadHoldingRegistersResponse': 3]2023-09-12 20:36:42.263 DEBUG (SyncWorker_6) [pymodbus.logging] Adding transaction 12023-09-12 20:36:42.263 DEBUG (SyncWorker_6) [pymodbus.logging] Getting transaction 12023-09-12 20:36:42.263 DEBUG (SyncWorker_6) [pymodbus.logging] Changing transaction state from "PROCESSING REPLY" to "TRANSACTION_COMPLETE"2023-09-12 20:36:42.293 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 43006 with length 1 from slave 02023-09-12 20:36:42.293 DEBUG (MainThread) [pymodbus.logging] send: 0x0 0x1 0x0 0x0 0x0 0x6 0x0 0x3 0xa7 0xfe 0x0 0x12023-09-12 20:36:42.296 DEBUG (MainThread) [pymodbus.logging] Adding transaction 12023-09-12 20:36:42.324 DEBUG (MainThread) [pymodbus.logging] [50, 100, 100, 1, 0, 80, 0, 0, 0, 0, 0]2023-09-12 20:36:42.325 DEBUG (SyncWorker_1) [pymodbus.logging] Current transaction state - TRANSACTION_COMPLETE2023-09-12 20:36:42.325 DEBUG (SyncWorker_1) [pymodbus.logging] Running transaction 22023-09-12 20:36:42.325 DEBUG (SyncWorker_1) [pymodbus.logging] SEND: 0x0 0x2 0x0 0x0 0x0 0x6 0x64 0x3 0xb 0x54 0x0 0x42023-09-12 20:36:42.325 DEBUG (SyncWorker_1) [pymodbus.logging] New Transaction state "SENDING"2023-09-12 20:36:42.328 DEBUG (SyncWorker_1) [pymodbus.logging] Changing transaction state from "SENDING" to "WAITING FOR REPLY"2023-09-12 20:36:42.331 DEBUG (SyncWorker_1) [pymodbus.logging] Changing transaction state from "WAITING FOR REPLY" to "PROCESSING REPLY"2023-09-12 20:36:42.332 DEBUG (SyncWorker_1) [pymodbus.logging] RECV: 0x0 0x2 0x0 0x0 0x0 0xb 0x64 0x3 0x8 0x0 0x1 0x0 0x64 0x0 0x1 0x0 0x642023-09-12 20:36:42.332 DEBUG (SyncWorker_1) [pymodbus.logging] Processing: 0x0 0x2 0x0 0x0 0x0 0xb 0x64 0x3 0x8 0x0 0x1 0x0 0x64 0x0 0x1 0x0 0x642023-09-12 20:36:42.332 DEBUG (SyncWorker_1) [pymodbus.logging] Factory Response[ReadHoldingRegistersResponse': 3]2023-09-12 20:36:42.332 DEBUG (SyncWorker_1) [pymodbus.logging] Adding transaction 22023-09-12 20:36:42.332 DEBUG (SyncWorker_1) [pymodbus.logging] Getting transaction 22023-09-12 20:36:11.689 WARNING (SyncWorker_4) [homeassistant.loader] We found a custom integration simpleicons which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant2023-09-12 20:36:11.693 WARNING (SyncWorker_4) [homeassistant.loader] We found a custom integration huawei_solar which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant2023-09-12 20:36:11.697 WARNING (SyncWorker_4) [homeassistant.loader] We found a custom integration adaptive_lighting which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant2023-09-12 20:36:11.700 WARNING (SyncWorker_4) [homeassistant.loader] We found a custom integration smartir which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant2023-09-12 20:36:39.847 DEBUG (MainThread) [pymodbus.logging] Connecting to 192.168.222.121:502.2023-09-12 20:36:39.847 DEBUG (MainThread) [pymodbus.logging] Connecting comm2023-09-12 20:36:40.156 DEBUG (MainThread) [pymodbus.logging] Connected to comm2023-09-12 20:36:40.156 DEBUG (MainThread) [pymodbus.logging] callback_connected called2023-09-12 20:36:40.170 DEBUG (MainThread) [huawei_solar.modbus] Waiting for 1500 milliseconds after connection before performing operations2023-09-12 20:36:41.649 INFO (MainThread) [huawei_solar.huawei_solar] Waiting for connection ...2023-09-12 20:36:42.248 DEBUG (MainThread) [pymodbus.logging] Connection to Modbus server established. Socket ('192.168.222.106', 34710)2023-09-12 20:36:42.252 DEBUG (SyncWorker_6) [pymodbus.logging] Current transaction state - IDLE2023-09-12 20:36:42.254 DEBUG (SyncWorker_6) [pymodbus.logging] Running transaction 12023-09-12 20:36:42.254 DEBUG (SyncWorker_6) [pymodbus.logging] SEND: 0x0 0x1 0x0 0x0 0x0 0x6 0x64 0x3 0xa 0x8c 0x0 0xb2023-09-12 20:36:42.254 DEBUG (SyncWorker_6) [pymodbus.logging] New Transaction state "SENDING"2023-09-12 20:36:42.255 DEBUG (SyncWorker_6) [pymodbus.logging] Changing transaction state from "SENDING" to "WAITING FOR REPLY"2023-09-12 20:36:42.262 DEBUG (SyncWorker_6) [pymodbus.logging] Changing transaction state from "WAITING FOR REPLY" to "PROCESSING REPLY"2023-09-12 20:36:42.263 DEBUG (SyncWorker_6) [pymodbus.logging] RECV: 0x0 0x1 0x0 0x0 0x0 0x19 0x64 0x3 0x16 0x0 0x32 0x0 0x64 0x0 0x64 0x0 0x1 0x0 0x0 0x0 0x50 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x02023-09-12 20:36:42.263 DEBUG (SyncWorker_6) [pymodbus.logging] Processing: 0x0 0x1 0x0 0x0 0x0 0x19 0x64 0x3 0x16 0x0 0x32 0x0 0x64 0x0 0x64 0x0 0x1 0x0 0x0 0x0 0x50 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x02023-09-12 20:36:11.664 WARNING (SyncWorker_4) [homeassistant.loader] We found a custom integration victron which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant2023-09-12 20:36:11.668 WARNING (SyncWorker_4) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant2023-09-12 20:36:11.671 WARNING (SyncWorker_4) [homeassistant.loader] We found a custom integration alarmo which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant2023-09-12 20:36:11.674 WARNING (SyncWorker_4) [homeassistant.loader] We found a custom integration localtuya which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant2023-09-12 20:36:11.678 WARNING (SyncWorker_4) [homeassistant.loader] We found a custom integration alexa_media which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant2023-09-12 20:36:11.681 WARNING (SyncWorker_4) [homeassistant.loader] We found a custom integration openweathermap_all which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant2023-09-12 20:36:11.685 WARNING (SyncWorker_4) [homeassistant.loader] We found a custom integration battery_sim which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
```

### Please confirm the following:

- [X] The problem is still present in the latest release of this integration.
- [ ] I did not find an existing issue describing this problem.
- [ ] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [ ] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

DennisHedberg, 2023-09-12:
Hi! 
I have the same issue, thanks for investigating! :)

Error setting up entry SUN2000-8KTL-M101074300-002 for huawei_solar
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 178, in create
    await huawei_solar._initialize()
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 107, in _initialize
    self.time_zone = (await self.get(rn.TIME_ZONE)).value
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 241, in get
    return (await self.get_multiple([name], slave))[0]
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 278, in get_multiple
    decoder = BinaryPayloadDecoder.fromRegisters(response.registers, byteorder=Endian.Big, wordorder=Endian.Big)
                                                                               ^^^^^^^^^^
  File "/usr/local/lib/python3.11/enum.py", line 784, in __getattr__
    raise AttributeError(name) from None
AttributeError: Big

anthurta, 2023-09-12:
And same issue here:

2023-09-12 17:07:53.468 ERROR (MainThread) [huawei_solar.huawei_solar] Aborting client creation due to error.
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 178, in create
    await huawei_solar._initialize()
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 107, in _initialize
    self.time_zone = (await self.get(rn.TIME_ZONE)).value
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 241, in get
    return (await self.get_multiple([name], slave))[0]
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 278, in get_multiple
    decoder = BinaryPayloadDecoder.fromRegisters(response.registers, byteorder=Endian.Big, wordorder=Endian.Big)
                                                                               ^^^^^^^^^^
  File "/usr/local/lib/python3.11/enum.py", line 784, in __getattr__
    raise AttributeError(name) from None
AttributeError: Big
2023-09-12 17:07:53.471 ERROR (MainThread) [custom_components.huawei_solar.config_flow] object NoneType can't be used in 'await' expression
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 178, in create
    await huawei_solar._initialize()
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 107, in _initialize
    self.time_zone = (await self.get(rn.TIME_ZONE)).value
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 241, in get
    return (await self.get_multiple([name], slave))[0]
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 278, in get_multiple
    decoder = BinaryPayloadDecoder.fromRegisters(response.registers, byteorder=Endian.Big, wordorder=Endian.Big)
                                                                               ^^^^^^^^^^
  File "/usr/local/lib/python3.11/enum.py", line 784, in __getattr__
    raise AttributeError(name) from None
AttributeError: Big
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/config/custom_components/huawei_solar/config_flow.py", line 382, in async_step_setup_network
    info = await validate_network_setup(user_input)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/huawei_solar/config_flow.py", line 115, in validate_network_setup
    bridge = await HuaweiSolarBridge.create(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/bridge.py", line 66, in create
    client = await AsyncHuaweiSolar.create(host, port, slave_id)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 187, in create
    await client.close()
TypeError: object NoneType can't be used in 'await' expression

AnKloss, 2023-09-13:
Hello
Update to 2023.9.2,  but the same error.

lukasha1, 2023-09-13:
Hi,

I have the same problem (after update to 2023.9.1, problem still exist after update 2023.9.2)

Lukasha

wlcrs, 2023-09-13:
This was already fixed in the latest beta. I've just issued a new release.

closed 2023-09-13 by wlcrs

AnKloss, 2023-09-13:
Hello,

Thanks for your time to fix it. Now is working .

mfg
Andreas