wlcrs/huawei_solar #470: [Bug]:

nedievas, 2023-09-14:
### Describe the issue

Could not load after update to 2023.9.2 or *.*1. Works with 2023.9.0

### Bescribe your Huawei Solar Setup

Inverter Type: SUN2000-8KTL-M1
Inverter Firmware version:
SDongle present: no
Power meter present: three phase

### How do you connect to the inverter?

Over serial, with a serial-to-WiFi stick

### Upload your Diagnostics File

There is no download diagnostics.

### Upload your relevant debug logs

```text
Logger: homeassistant.setup
Source: loader.py:826
First occurred: 13:11:49 (1 occurrences)
Last logged: 13:11:49

Setup failed for custom integration huawei_solar: Unable to import component: Exception importing custom_components.huawei_solar
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/loader.py", line 818, in get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1147, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/config/custom_components/huawei_solar/__init__.py", line 25, in <module>
    from huawei_solar import HuaweiSolarBridge, HuaweiSolarException, InvalidCredentials
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/__init__.py", line 6, in <module>
    from .bridge import HuaweiSolarBridge  # noqa
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/bridge.py", line 16, in <module>
    from .huawei_solar import DEFAULT_BAUDRATE, DEFAULT_SLAVE, DEFAULT_TCP_PORT, AsyncHuaweiSolar, Result
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 17, in <module>
    from pymodbus.payload import BinaryPayloadBuilder, BinaryPayloadDecoder
  File "/usr/local/lib/python3.11/site-packages/pymodbus/payload.py", line 28, in <module>
    class BinaryPayloadBuilder:
  File "/usr/local/lib/python3.11/site-packages/pymodbus/payload.py", line 42, in BinaryPayloadBuilder
    self, payload=None, byteorder=Endian.Little, wordorder=Endian.Big, repack=False
                                  ^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/enum.py", line 784, in __getattr__
    raise AttributeError(name) from None
AttributeError: Little

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 215, in _async_setup_component
    component = integration.get_component()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 826, in get_component
    raise ImportError(f"Exception importing {self.pkg_path}") from err
ImportError: Exception importing custom_components.huawei_solar
****************************
This error originated from a custom integration.

Logger: homeassistant.loader
Source: custom_components/huawei_solar/__init__.py:25
Integration: Huawei Solar (documentation, issues)
First occurred: 13:11:49 (1 occurrences)
Last logged: 13:11:49

Unexpected exception importing component custom_components.huawei_solar
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/loader.py", line 818, in get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1147, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/config/custom_components/huawei_solar/__init__.py", line 25, in <module>
    from huawei_solar import HuaweiSolarBridge, HuaweiSolarException, InvalidCredentials
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/__init__.py", line 6, in <module>
    from .bridge import HuaweiSolarBridge  # noqa
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/bridge.py", line 16, in <module>
    from .huawei_solar import DEFAULT_BAUDRATE, DEFAULT_SLAVE, DEFAULT_TCP_PORT, AsyncHuaweiSolar, Result
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 17, in <module>
    from pymodbus.payload import BinaryPayloadBuilder, BinaryPayloadDecoder
  File "/usr/local/lib/python3.11/site-packages/pymodbus/payload.py", line 28, in <module>
    class BinaryPayloadBuilder:
  File "/usr/local/lib/python3.11/site-packages/pymodbus/payload.py", line 42, in BinaryPayloadBuilder
    self, payload=None, byteorder=Endian.Little, wordorder=Endian.Big, repack=False
                                  ^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/enum.py", line 784, in __getattr__
    raise AttributeError(name) from None
AttributeError: Little
```

### Please confirm the following:

- [X] The problem is still present in the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [ ] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [X] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

nicx, 2023-09-18:
same her with sDongle connection. Downgraded to HA 2023.9.0 and it works again.

wlcrs, 2023-09-18:
Please use the latest version of the integration.

If that doesn't solve it: please upload the diagnostics file

[image]

nedievas, 2023-09-18:
It is the same if I upgrade to 1.2.7, but then 2023.9.0 is also affected. Download diagnostics does not work when the integration does not load.

nedievas, 2023-09-18:
Here is diagnostics before upgrade:
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.9.0",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.11.5",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Vilnius",
    "os_name": "Linux",
    "os_version": "6.1.45",
    "supervisor": "2023.09.2",
    "host_os": "Home Assistant OS 10.5",
    "docker_version": "23.0.6",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {
    "lovelace_gen": {
      "version": "0.1.1",
      "requirements": [
        "jinja2"
      ]
    },
    "smartir": {
      "version": "1.17.6",
      "requirements": [
        "aiofiles==0.6.0"
      ]
    },
    "tesla_custom": {
      "version": "3.16.1",
      "requirements": [
        "teslajsonpy==3.9.3"
      ]
    },
    "xiaomi_cloud_map_extractor": {
      "version": "v2.2.0",
      "requirements": [
        "pillow",
        "pybase64",
        "python-miio",
        "requests",
        "pycryptodome"
      ]
    },
    "localtuya": {
      "version": "5.2.1",
      "requirements": []
    },
    "pyscript": {
      "version": "1.5.0",
      "requirements": [
        "croniter==1.3.8",
        "watchdog==2.3.1"
      ]
    },
    "huawei_solar": {
      "version": "1.2.6",
      "requirements": [
        "huawei-solar==2.2.8"
      ]
    },
    "hacs": {
      "version": "1.32.1",
      "requirements": [
        "aiogithubapi>=22.10.1"
      ]
    }
  },
  "integration_manifest": {
    "domain": "huawei_solar",
    "name": "Huawei Solar",
    "config_flow": true,
    "documentation": "https://github.com/wlcrs/huawei_solar/wiki",
    "issue_tracker": "https://github.com/wlcrs/huawei_solar/issues",
    "requirements": [
      "huawei-solar==2.2.8"
    ],
    "codeowners": [
      "@wlcrs"
    ],
    "iot_class": "local_polling",
    "version": "1.2.6",
    "loggers": [
      "huawei_solar",
      "pymodbus"
    ],
    "is_built_in": false
  },
  "data": {
    "config_entry_data": {
      "host": "192.168.1.224",
      "port": 8899,
      "slave_ids": [
        1
      ],
      "enable_parameter_configuration": true,
      "username": null,
      "password": null
    },
    "slave_0": {
      "model_name": "SUN2000-8KTL-M1\u000001074300-002",
      "pv_string_count": 2,
      "has_optimizers": 6,
      "battery_type": 0,
      "battery_1_type": 0,
      "battery_2_type": 0,
      "power_meter_type": null,
      "supports_capacity_control": false
    },
    "slave_0_data": {
      "input_power": [
        0,
        "W"
      ],
      "line_voltage_A_B": [
        0.0,
        "V"
      ],
      "line_voltage_B_C": [
        0.0,
        "V"
      ],
      "line_voltage_C_A": [
        0.0,
        "V"
      ],
      "phase_A_voltage": [
        0.0,
        "V"
      ],
      "phase_B_voltage": [
        0.0,
        "V"
      ],
      "phase_C_voltage": [
        0.0,
        "V"
      ],
      "phase_A_current": [
        0.0,
        "A"
      ],
      "phase_B_current": [
        0.0,
        "A"
      ],
      "phase_C_current": [
        0.0,
        "A"
      ],
      "day_active_power_peak": [
        6318,
        "W"
      ],
      "active_power": [
        0,
        "W"
      ],
      "reactive_power": [
        0,
        "VA"
      ],
      "power_factor": [
        0.0,
        null
      ],
      "grid_frequency": [
        0.0,
        "Hz"
      ],
      "efficiency": [
        0.0,
        "%"
      ],
      "internal_temperature": [
        0.0,
        "\u00b0C"
      ],
      "insulation_resistance": [
        30.0,
        "MOhm"
      ],
      "device_status": [
        "Standby: no irradiation",
        null
      ],
      "fault_code": [
        0,
        null
      ],
      "startup_time": [
        "2023-09-18T05:10:31+00:00",
        null
      ],
      "shutdown_time": [
        "2023-09-18T17:38:16+00:00",
        null
      ],
      "accumulated_yield_energy": [
        16600.86,
        "kWh"
      ],
      "daily_yield_energy": [
        29.21,
        "kWh"
      ],
      "state_1": [
        [
          "Standby"
        ],
        null
      ],
      "state_2": [
        [
          "Locked",
          "PV disconnected",
          "No DSP data collection"
        ],
        null
      ],
      "state_3": [
        [
          "On-grid",
          "Off-grid switch disabled"
        ],
        null
      ],
      "alarm_1": [
        [],
        null
      ],
      "alarm_2": [
        [],
        null
      ],
      "alarm_3": [
        [],
        null
      ],
      "pv_01_voltage": [
        0.0,
        "V"
      ],
      "pv_01_current": [
        0.0,
        "A"
      ],
      "pv_02_voltage": [
        0.0,
        "V"
      ],
      "pv_02_current": [
        0.0,
        "A"
      ],
      "nb_online_optimizers": [
        0,
        null
      ]
    },
    "slave_0_config_data": {},
    "slave_0_optimizer_data": {}
  }
}

marcgeurts, 2023-09-19:
I have the same issues, since i've upgraded to the latest version. the connection is lost once in a while, the DD-WRT router is still online and connected during these times.

pschraml, 2023-09-19:
I have the same issue here downgraded to 2023.9.0 and 2023.8.4 no luck so far.
I have also uninstalled and reinstalled it but same issues 

2023-09-19 15:34:56.324 WARNING (SyncWorker_3) [homeassistant.loader] We found a custom integration huawei_solar which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2023-09-19 15:34:56.325 WARNING (SyncWorker_3) [homeassistant.loader] We found a custom integration tapo which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2023-09-19 15:34:56.327 WARNING (SyncWorker_3) [homeassistant.loader] We found a custom integration solcast_solar which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2023-09-19 15:34:56.329 WARNING (SyncWorker_3) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2023-09-19 15:34:56.330 WARNING (SyncWorker_3) [homeassistant.loader] We found a custom integration octopus_energy which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2023-09-19 15:34:56.334 WARNING (SyncWorker_3) [homeassistant.loader] We found a custom integration edgeos which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2023-09-19 15:34:56.338 WARNING (SyncWorker_3) [homeassistant.loader] We found a custom integration hpprinter which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2023-09-19 15:34:59.048 WARNING (Recorder) [homeassistant.components.recorder.util] The system could not validate that the sqlite3 database at //config/home-assistant_v2.db was shutdown cleanly
2023-09-19 15:34:59.089 WARNING (Recorder) [homeassistant.components.recorder.util] Ended unfinished session (id=82 from 2023-09-01 06:02:23.942011)
2023-09-19 15:35:01.723 WARNING (MainThread) [homeassistant.components.http.ban] Login attempt or request with invalid authentication from 81.2.178.50 (81.2.178.50). Requested URL: '/auth/token'. (Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/117.0.0.0 Safari/537.36)
2023-09-19 15:35:01.832 WARNING (MainThread) [homeassistant.components.http.ban] Login attempt or request with invalid authentication from 81.2.178.50 (81.2.178.50). Requested URL: '/api/websocket'. (Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/117.0.0.0 Safari/537.36)
2023-09-19 15:35:02.173 WARNING (MainThread) [homeassistant.components.http.ban] Login attempt or request with invalid authentication from 81.2.178.50 (81.2.178.50). Requested URL: '/auth/token'. (Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/117.0.0.0 Safari/537.36)
2023-09-19 15:35:02.400 WARNING (MainThread) [homeassistant.components.http.ban] Login attempt or request with invalid authentication from 81.2.178.50 (81.2.178.50). Requested URL: '/auth/token'. (Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/117.0.0.0 Safari/537.36)
2023-09-19 15:35:36.078 ERROR (MainThread) [homeassistant.components.sensor] The huawei_solar platform for the sensor integration does not support platform setup. Please remove it from your config.
2023-09-19 15:35:48.859 INFO (MainThread) [backoff] Backing off _do_read(...) for 1.0s (TimeoutError)
2023-09-19 15:35:49.874 ERROR (MainThread) [huawei_solar.huawei_solar] Modbus client is not connected to the inverter.
NoneType: None
2023-09-19 15:35:49.875 INFO (MainThread) [backoff] Backing off _do_read(...) for 2.0s (huawei_solar.exceptions.ConnectionInterruptedException: Modbus client is not connected to the inverter.)
2023-09-19 15:35:51.876 ERROR (MainThread) [huawei_solar.huawei_solar] Modbus client is not connected to the inverter.
NoneType: None
2023-09-19 15:35:51.877 INFO (MainThread) [backoff] Backing off _do_read(...) for 4.0s (huawei_solar.exceptions.ConnectionInterruptedException: Modbus client is not connected to the inverter.)
2023-09-19 15:35:52.381 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 113] Connect call failed ('10.10.0.161', 6607)
2023-09-19 15:35:55.878 ERROR (MainThread) [huawei_solar.huawei_solar] Modbus client is not connected to the inverter.
NoneType: None
2023-09-19 15:35:55.879 INFO (MainThread) [backoff] Backing off _do_read(...) for 8.0s (huawei_solar.exceptions.ConnectionInterruptedException: Modbus client is not connected to the inverter.)
2023-09-19 15:36:00.266 WARNING (MainThread) [homeassistant.bootstrap] Waiting on integrations to complete setup: huawei_solar, raspberry_pi
2023-09-19 15:36:03.880 ERROR (MainThread) [huawei_solar.huawei_solar] Modbus client is not connected to the inverter.
NoneType: None
2023-09-19 15:36:03.882 INFO (MainThread) [backoff] Backing off _do_read(...) for 16.0s (huawei_solar.exceptions.ConnectionInterruptedException: Modbus client is not connected to the inverter.)
[... cut ...]
2023-09-19 15:41:36.313 DEBUG (MainThread) [pymodbus.logging] Connecting to 10.10.0.161:6607.
2023-09-19 15:41:36.313 DEBUG (MainThread) [pymodbus.logging] Connecting comm
2023-09-19 15:41:39.402 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 113] Connect call failed ('10.10.0.161', 6607)
2023-09-19 15:42:05.502 ERROR (MainThread) [custom_components.tapo.coordinators] Error fetching tapo data: Unexpected exception: 
2023-09-19 15:42:55.217 ERROR (MainThread) [custom_components.tapo.coordinators] Error fetching tapo data: Unexpected exception:

wlcrs, 2023-09-20:
The original error `AttributeError: Little` is caused by pyModbus changing the naming of `Endian.Little` to `Endian.LITTLE` (cfr. https://github.com/pymodbus-dev/pymodbus/commit/b1b560ab1e0b7482316f72e3e73110de802d904c#diff-c8ea7441e5976d849d2f3c89c966d3609df331a8c352d862e4238cbe1c5c0b7bL52), a change which was released in pyModbus v3.5.0

v1.2.6 of this integration has a dependency on a lower pyModbus version (cfr. https://github.com/wlcrs/huawei-solar-lib/blob/2.2.8/setup.cfg#L22) but apparently your HA installation has a more recent version installed nonetheless.

v1.2.7 of this integration has adopted this naming change and is compatible with pyModbus v3.5 (cfr. https://github.com/wlcrs/huawei-solar-lib/blob/2.2.9/setup.cfg#L22)

So you'll need to figure out what is causing these dependency conflicts which cause the wrong pyModbus version to be installed. Your diagnostics file doesn't readily show what other integration is causing the dependency conflict, so you'll need to figure this out yourself.

As this is not a bug in this integration, but a dependency conflict in your HA installation I'm closing this issue now. Keeping it open is causing people to dump unrelated errors in here.

closed 2023-09-20 by wlcrs