wlcrs/huawei_solar #475: [Bug]: Huawei extension has stopped working.

Krasto82, 2023-09-16:
### Describe the issue

For the past few days, the extension has not been working. It was working perfectly fine before without any issues. When I try to connect, I receive timeouts, and the log file says it's not connected. Nothing has changed in my network or settings. I have the latest version of Home Assistant. If I try to connect the heat pump to the dongle (Modbus IP) to receive data from the inverter, this function works correctly. Thank you.
[home-assistant.log](https://github.com/wlcrs/huawei_solar/files/12640666/home-assistant.log)

### Bescribe your Huawei Solar Setup

Inverter Type: SUN2000-10KTL-M1
Inverter Firmware version: V100R001C00SPC156
SDongle present: yes (SDongleA-05 FW: V100R001C00SPC130)
Power meter present: three phase 
Battery: LUNA2000 30kWh
Battery Firmware version: V100R002C00SPC121

### How do you connect to the inverter?

Via the SDongle, wired connection

### Upload your Diagnostics File

Drag & Drop your Diagnostics File here.
[home-assistant.log](https://github.com/wlcrs/huawei_solar/files/12640675/home-assistant.log)

### Upload your relevant debug logs

```text
2023-09-16 14:06:11.229 ERROR (MainThread) [huawei_solar.huawei_solar] Could not read register value, has another device interrupted the connection?
2023-09-16 14:06:11.237 INFO (MainThread) [backoff] Backing off _do_read(...) for 1.0s (huawei_solar.exceptions.ConnectionInterruptedException: Could not read register value, has another device interrupted the connection?)
2023-09-16 14:06:12.240 ERROR (MainThread) [huawei_solar.huawei_solar] Modbus client is not connected to the inverter.
NoneType: None
2023-09-16 14:06:12.241 INFO (MainThread) [backoff] Backing off _do_read(...) for 2.0s (huawei_solar.exceptions.ConnectionInterruptedException: Modbus client is not connected to the inverter.)
2023-09-16 14:06:14.242 ERROR (MainThread) [huawei_solar.huawei_solar] Modbus client is not connected to the inverter.
NoneType: None
2023-09-16 14:06:14.243 INFO (MainThread) [backoff] Backing off _do_read(...) for 4.0s (huawei_solar.exceptions.ConnectionInterruptedException: Modbus client is not connected to the inverter.)
2023-09-16 14:06:18.244 ERROR (MainThread) [huawei_solar.huawei_solar] Modbus client is not connected to the inverter.
NoneType: None
2023-09-16 14:06:18.245 INFO (MainThread) [backoff] Backing off _do_read(...) for 8.0s (huawei_solar.exceptions.ConnectionInterruptedException: Modbus client is not connected to the inverter.)
2023-09-16 14:06:26.246 ERROR (MainThread) [huawei_solar.huawei_solar] Modbus client is not connected to the inverter.
NoneType: None
2023-09-16 14:06:26.248 INFO (MainThread) [backoff] Backing off _do_read(...) for 16.0s (huawei_solar.exceptions.ConnectionInterruptedException: Modbus client is not connected to the inverter.)
2023-09-16 14:06:42.250 ERROR (MainThread) [huawei_solar.huawei_solar] Modbus client is not connected to the inverter.
NoneType: None
2023-09-16 14:06:42.251 ERROR (MainThread) [backoff] Giving up _do_read(...) after 6 tries (huawei_solar.exceptions.ConnectionInterruptedException: Modbus client is not connected to the inverter.)
2023-09-16 14:06:42.256 ERROR (MainThread) [huawei_solar.huawei_solar] Aborting client creation due to error.
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/backoff/_async.py", line 151, in retry
    ret = await target(*args, **kwargs)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/backoff/_async.py", line 151, in retry
    ret = await target(*args, **kwargs)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 349, in _do_read
    raise ConnectionInterruptedException(message)
huawei_solar.exceptions.ConnectionInterruptedException: Modbus client is not connected to the inverter.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 178, in create
    await huawei_solar._initialize()
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 107, in _initialize
    self.time_zone = (await self.get(rn.TIME_ZONE)).value
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 241, in get
    return (await self.get_multiple([name], slave))[0]
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 276, in get_multiple
    response = await self._read_registers(registers[0].register, total_length, slave)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 398, in _read_registers
    return await _do_read()
           ^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/backoff/_async.py", line 159, in retry
    await _call_handlers(on_giveup, **details, exception=e)
  File "/usr/local/lib/python3.11/site-packages/backoff/_async.py", line 37, in _call_handlers
    await handler(details)
  File "/usr/local/lib/python3.11/site-packages/backoff/_async.py", line 16, in f
    return coro_or_func(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 327, in backoff_giveup
    raise ReadException(f"Failed to read register {register} after {details['tries']} tries")
huawei_solar.exceptions.ReadException: Failed to read register 43006 after 6 tries
```

### Please confirm the following:

- [X] The problem is still present in the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [ ] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [ ] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

xm4rcell0x, 2023-09-16:
Hi, i'm on the same boat and i really don't know how to fix this. I'm the OP of #473 and one day, without any reason the integration stop working. Nothing has been changed in my network, i only rebooted my hypervisor for an update, that's it. 
I'm trying to revert to ha core 8.4 and huawei solar 1.2.6 , the last version that worked for me, but nothing to do...

wlcrs, 2023-09-18:
You provide me with an absolute minimum of logs, which only show that the inverter did not respond in time to any requests. During the backoff sequence, the integration has already tried reconnecting to the inverter (it does that after 3 consecutive failures). This clearly didn't resolve the issue.

Please check how stable the network connection to your inverter is.

As I don't see a bug in this integration, I'm closing this issue as I do not have anything I can do here.

For future readers:  **your must extensive debug logs AND the diagnostics file of the integration if you want me to be able to do a proper diagnosis of your problem**

[image]

closed 2023-09-18 by wlcrs

Krasto82, 2023-09-19:
I understand, but the lack of more detailed logs is frustrating. I removed the integration when it stopped working, and now it's not possible to reconnect. My network is functioning perfectly, as is the LAN connection to the Huawei dongle. Based on the feedback I've received, it doesn't seem like I'm the only one in this situation.

GrueneNeun, 2023-11-11:
I had the exact same Problem today - for more than 6 hours nor restarting the integration, home assistant or the whole VM helped with the same error that you had. Before today, the integration worked flawless for weeks.
I restarted the whole inverter and instantly the integration started working as expected. I guess their modbus implementation might have some issues? The inverter communicated with the fusion solar cloud without problems during that time...

Krasto82, 2023-11-12:
> I had the exact same Problem today - for more than 6 hours nor restarting the integration, home assistant or the whole VM helped with the same error that you had. Before today, the integration worked flawless for weeks. I restarted the whole inverter and instantly the integration started working as expected. I guess the Huawei modbus implementation might have some issues? The inverter communicated with the fusion solar cloud without problems during that time... I also use the dongle with a wired connection.

Same, the modbus works because my heat pump gets data from the inverter, but the integration of ha no longer works. I enter the ip but it always returns error in connecting. Fusion solar is set up correctly and I also have installer privileges on my account....