wlcrs/huawei_solar #476: [Bug]: Integration stopped working

taert79, 2023-09-18:
### Describe the issue

My integration has stopped working as of last Saturday. Has been working perfectly before that for months. No changes to my setup besides the usual upgrades. 

### Bescribe your Huawei Solar Setup

Inverter Type: SUN2000-2KTL_L1
Inverter Firmware version: Unknown
SDongle present: yes
Power meter present: single phase 
Battery: None
Battery Firmware version: N/A

### How do you connect to the inverter?

Via the SDongle, wireless connection

### Upload your Diagnostics File

Drag & Drop your Diagnostics File here.

### Upload your relevant debug logs

```text
Logger: huawei_solar.huawei_solar
Source: /usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py:184
First occurred: 12:05:35 (71 occurrences)
Last logged: 13:45:53

Aborting client creation due to error.
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/asyncio/tasks.py", line 490, in wait_for
    return fut.result()
           ^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/asyncio/locks.py", line 213, in wait
    await fut
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 178, in create
    await huawei_solar._initialize()
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 107, in _initialize
    self.time_zone = (await self.get(rn.TIME_ZONE)).value
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 241, in get
    return (await self.get_multiple([name], slave))[0]
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 276, in get_multiple
    response = await self._read_registers(registers[0].register, total_length, slave)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 396, in _read_registers
    async with self._communication_lock():
  File "/usr/local/lib/python3.11/contextlib.py", line 204, in __aenter__
    return await anext(self.gen)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 145, in _communication_lock
    raise err
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 141, in _communication_lock
    await asyncio.wait_for(self._client.connected_event.wait(), WAIT_FOR_CONNECTION_TIMEOUT)
  File "/usr/local/lib/python3.11/asyncio/tasks.py", line 492, in wait_for
    raise exceptions.TimeoutError() from exc
TimeoutError

Logger: huawei_solar.huawei_solar
Source: /usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py:143
First occurred: 12:05:35 (71 occurrences)
Last logged: 13:45:53

Timeout while waiting for connection. Reconnecting...
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/asyncio/tasks.py", line 490, in wait_for
    return fut.result()
           ^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/asyncio/locks.py", line 213, in wait
    await fut
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 141, in _communication_lock
    await asyncio.wait_for(self._client.connected_event.wait(), WAIT_FOR_CONNECTION_TIMEOUT)
  File "/usr/local/lib/python3.11/asyncio/tasks.py", line 492, in wait_for
    raise exceptions.TimeoutError() from exc
TimeoutError
```

### Please confirm the following:

- [X] The problem is still present in the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [X] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [X] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

wlcrs, 2023-09-18:
Please upload the diagnostics file and a debug log. Without it I cannot diagnose the issue.

What is the version of your inverter firmware? Did it maybe update itself in the last few days?

taert79, 2023-09-18:
Hi @wlcrs,
FW version is v200r001c00spc117. If needed I can upgrade it to a higher revision.
Log is attached.
[home-assistant_2023-09-18T13-57-00.589Z.log](https://github.com/wlcrs/huawei_solar/files/12648583/home-assistant_2023-09-18T13-57-00.589Z.log)