wlcrs/huawei_solar #479: [Bug]: Integration does not load when upgraded to 1.2.7 on 2023.9.0

nedievas, 2023-09-21:
### Describe the issue

The same problem.

### Bescribe your Huawei Solar Setup

Inverter Type: SUN2000-8KTL-M1
Inverter Firmware version:
SDongle present: no
Power meter present: no
Battery:  None

### How do you connect to the inverter?

Please select your connection method

### Upload your Diagnostics File

Drag & Drop your Diagnostics File here.
[image]

### Upload your relevant debug logs

```text
This error originated from a custom integration.

Logger: homeassistant.loader
Source: custom_components/huawei_solar/__init__.py:25
Integration: Huawei Solar (documentation, issues)
First occurred: 16:18:50 (1 occurrences)
Last logged: 16:18:50

Unexpected exception importing component custom_components.huawei_solar
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/loader.py", line 818, in get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1147, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/config/custom_components/huawei_solar/__init__.py", line 25, in <module>
    from huawei_solar import HuaweiSolarBridge, HuaweiSolarException, InvalidCredentials
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/__init__.py", line 6, in <module>
    from .bridge import HuaweiSolarBridge  # noqa
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/bridge.py", line 16, in <module>
    from .huawei_solar import DEFAULT_BAUDRATE, DEFAULT_SLAVE, DEFAULT_TCP_PORT, AsyncHuaweiSolar, Result
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 17, in <module>
    from pymodbus.payload import BinaryPayloadBuilder, BinaryPayloadDecoder
  File "/usr/local/lib/python3.11/site-packages/pymodbus/payload.py", line 28, in <module>
    class BinaryPayloadBuilder:
  File "/usr/local/lib/python3.11/site-packages/pymodbus/payload.py", line 42, in BinaryPayloadBuilder
    self, payload=None, byteorder=Endian.LITTLE, wordorder=Endian.BIG, repack=False
                                  ^^^^^^^^^^^^^
AttributeError: type object 'Endian' has no attribute 'LITTLE'
****
Logger: homeassistant.setup
Source: loader.py:826
First occurred: 16:18:50 (1 occurrences)
Last logged: 16:18:50

Setup failed for custom integration huawei_solar: Unable to import component: Exception importing custom_components.huawei_solar
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/loader.py", line 818, in get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1147, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/config/custom_components/huawei_solar/__init__.py", line 25, in <module>
    from huawei_solar import HuaweiSolarBridge, HuaweiSolarException, InvalidCredentials
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/__init__.py", line 6, in <module>
    from .bridge import HuaweiSolarBridge  # noqa
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/bridge.py", line 16, in <module>
    from .huawei_solar import DEFAULT_BAUDRATE, DEFAULT_SLAVE, DEFAULT_TCP_PORT, AsyncHuaweiSolar, Result
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 17, in <module>
    from pymodbus.payload import BinaryPayloadBuilder, BinaryPayloadDecoder
  File "/usr/local/lib/python3.11/site-packages/pymodbus/payload.py", line 28, in <module>
    class BinaryPayloadBuilder:
  File "/usr/local/lib/python3.11/site-packages/pymodbus/payload.py", line 42, in BinaryPayloadBuilder
    self, payload=None, byteorder=Endian.LITTLE, wordorder=Endian.BIG, repack=False
                                  ^^^^^^^^^^^^^
AttributeError: type object 'Endian' has no attribute 'LITTLE'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 215, in _async_setup_component
    component = integration.get_component()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 826, in get_component
    raise ImportError(f"Exception importing {self.pkg_path}") from err
ImportError: Exception importing custom_components.huawei_solar
```

### Please confirm the following:

- [X] The problem is still present in the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [ ] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [ ] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

nedievas, 2023-09-21:
Downgraded to 1.2.6 on 2023.9.0. That way integration loads and I am able to download diagnostics. Here it is:
'''{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.9.0",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.11.5",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Vilnius",
    "os_name": "Linux",
    "os_version": "6.1.45",
    "supervisor": "2023.09.2",
    "host_os": "Home Assistant OS 10.5",
    "docker_version": "23.0.6",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {
    "smartir": {
      "version": "1.17.6",
      "requirements": [
        "aiofiles==0.6.0"
      ]
    },
    "tesla_custom": {
      "version": "3.16.1",
      "requirements": [
        "teslajsonpy==3.9.3"
      ]
    },
    "xiaomi_cloud_map_extractor": {
      "version": "v2.2.0",
      "requirements": [
        "pillow",
        "pybase64",
        "python-miio",
        "requests",
        "pycryptodome"
      ]
    },
    "localtuya": {
      "version": "5.2.1",
      "requirements": []
    },
    "huawei_solar": {
      "version": "1.2.6",
      "requirements": [
        "huawei-solar==2.2.8"
      ]
    },
    "hacs": {
      "version": "1.32.1",
      "requirements": [
        "aiogithubapi>=22.10.1"
      ]
    }
  },
  "integration_manifest": {
    "domain": "huawei_solar",
    "name": "Huawei Solar",
    "config_flow": true,
    "documentation": "https://github.com/wlcrs/huawei_solar/wiki",
    "issue_tracker": "https://github.com/wlcrs/huawei_solar/issues",
    "requirements": [
      "huawei-solar==2.2.8"
    ],
    "codeowners": [
[... 187 lines cut ...]
        "repr": "OptimizerRealTimeData(optimizer_address=1, output_power=155.9, voltage_to_ground=1.9, alarm=[], output_voltage=32.9, output_current=4.73, input_voltage=33.3, input_current=4.73, temperature=34.4, running_status=<OptimizerRunningStatus.RUNNING: 4>, accumulated_energy_yield=806.201)"
      },
      "2": {
        "__type": "<class 'huawei_solar.files.OptimizerRealTimeData'>",
        "repr": "OptimizerRealTimeData(optimizer_address=2, output_power=58.0, voltage_to_ground=2.2, alarm=[], output_voltage=12.2, output_current=4.74, input_voltage=22.2, input_current=2.82, temperature=37.8, running_status=<OptimizerRunningStatus.RUNNING: 4>, accumulated_energy_yield=733.78)"
      },
      "3": {
        "__type": "<class 'huawei_solar.files.OptimizerRealTimeData'>",
        "repr": "OptimizerRealTimeData(optimizer_address=3, output_power=160.1, voltage_to_ground=2.1, alarm=[], output_voltage=33.4, output_current=4.78, input_voltage=33.8, input_current=4.78, temperature=33.4, running_status=<OptimizerRunningStatus.RUNNING: 4>, accumulated_energy_yield=785.981)"
      },
      "4": {
        "__type": "<class 'huawei_solar.files.OptimizerRealTimeData'>",
        "repr": "OptimizerRealTimeData(optimizer_address=4, output_power=151.3, voltage_to_ground=2.1, alarm=[], output_voltage=31.4, output_current=4.81, input_voltage=31.8, input_current=4.81, temperature=32.8, running_status=<OptimizerRunningStatus.RUNNING: 4>, accumulated_energy_yield=787.588)"
      },
      "5": {
        "__type": "<class 'huawei_solar.files.OptimizerRealTimeData'>",
        "repr": "OptimizerRealTimeData(optimizer_address=5, output_power=155.4, voltage_to_ground=1.8, alarm=[], output_voltage=33.1, output_current=4.69, input_voltage=33.4, input_current=4.69, temperature=34.3, running_status=<OptimizerRunningStatus.RUNNING: 4>, accumulated_energy_yield=787.904)"
      },
      "6": {
        "__type": "<class 'huawei_solar.files.OptimizerRealTimeData'>",
        "repr": "OptimizerRealTimeData(optimizer_address=6, output_power=161.9, voltage_to_ground=2.1, alarm=[], output_voltage=33.2, output_current=4.87, input_voltage=33.6, input_current=4.83, temperature=35.0, running_status=<OptimizerRunningStatus.RUNNING: 4>, accumulated_energy_yield=825.287)"
      }
    }
  }
}'''

wlcrs, 2023-09-21:
I've described the source of your problem and the path to a solution at length here: https://github.com/wlcrs/huawei_solar/issues/470#issuecomment-1727258019

This is not a bug in this integration, nor a general problem. It is caused by a dependency conflict with another integration in your HA instance. This is not something that I can provide you any further assistance with. You'll need to figure out which other integration is causing the conflict and find a path forward yourself.

Please stop abusing this issue tracker. I will block you from interacting with this repository if you do it again.

closed 2023-09-21 by wlcrs

nedievas, 2023-09-21:
I would like to tell how I solved this issue, so maybe that will be interesting for someone else. I tried to update integration to 1.2.7 as on 2023.9.0 with no luck, as I believe I have no conflicting integrations. No luck, but then I restored to 1.2.6 version and firstly updated HA to 2023.9.2, Version 1.2.6 did not load, but then I did upgrade to integration to 1.2.7 and it loaded successfully. So the moral is that one should firstly update HA, then the integration.

myrkoo, 2023-09-23:
> I would like to tell how I solved this issue, so maybe that will be interesting for someone else. I tried to update integration to 1.2.7 as on 2023.9.0 with no luck, as I believe I have no [...]

Hi, same issue I solved in this way. Thank you