wlcrs/huawei_solar #482: [Bug]: Connection unstable after update to 1.2.7

lucio-modrono, 2023-09-22:
### Describe the issue

After update to 1.2.7 and restart HA, integration reports problem on UI and does not update sensors values.

Trying reset component using this trigger (wich helped in the past)...

```
action:
  - service: homeassistant.reload_config_entry
    data: {}
    target:
      entity_id:
        - sensor.inverter_total_yield
```

... but no luck, shows this error message on screen: 
```
The config entry SUN2000-3KTL-L1 (huawei_solar) with entry_id ... cannot be unloaded because it is not in a recoverable state (ConfigEntryState.FAILED_UNLOAD)
```
HA logs on starting: 
```
2023-09-22 16:47:38.952 ERROR (MainThread) [homeassistant.config_entries] Error unloading entry SUN2000-3KTL-L1 for huawei_solar
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 546, in async_unload
    result = await component.async_unload_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/huawei_solar/__init__.py", line 218, in async_unload_entry
    await async_cleanup_services(hass)
  File "/config/custom_components/huawei_solar/services.py", line 680, in async_cleanup_services
    if len(hass.data[DOMAIN] == 1):
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^
```
**[SOLVED (patched locally)]:** 
Changing code on "/config/custom_components/huawei_solar/services.py", line 680 from
```
    if len(hass.data[DOMAIN] == 1):
```
to
```
    if len(hass.data[DOMAIN]) == 1:
```
seems to solve the problem and the integration is stable again.

### Bescribe your Huawei Solar Setup

Inverter Type: SUN2000-3KTL
Inverter Firmware version: ¿?
SDongle present: no
Power meter present: single phase
Battery: None
Battery Firmware version:

### How do you connect to the inverter?

Over serial, with a serial-to-WiFi stick

### Upload your Diagnostics File

Drag & Drop your Diagnostics File here.

### Upload your relevant debug logs

```text
2023-09-22 16:47:38.952 ERROR (MainThread) [homeassistant.config_entries] Error unloading entry SUN2000-3KTL-L1 for huawei_solar
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 546, in async_unload
    result = await component.async_unload_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/huawei_solar/__init__.py", line 218, in async_unload_entry
    await async_cleanup_services(hass)
  File "/config/custom_components/huawei_solar/services.py", line 680, in async_cleanup_services
    if len(hass.data[DOMAIN] == 1):
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^
```

### Please confirm the following:

- [X] The problem is still present in the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [ ] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [X] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

closed 2023-09-22 by wlcrs

lucio-modrono, 2023-09-22:
BTW, thanks for this great work and sorry for my english XD

wlcrs, 2023-09-22:
Thank you for this clear bug report! 👍 I've already committed your bugfix

commit 43548072 (2023-09-22): Fix unload in services.py

Fixes #482